warpme/miniarch #1: I seem to have missed a step - no console

jfabernathy, 2023-01-14:
I', trying to test this on a Raspbery Pi 4 4GB. I downloaded MiniArch-20230106-6.1.5-board-sm1.tanix_tx5_plus-board-rpi34.mainline64-SD-Image.img.xz and flashed it to a microSD card using the Raspberry Pi Imager 1.7.2.

I can see the Uboot test on my Monitor but after the message Starting Kernel, the Monitor goes blank and I have not idea what to do next.

warpme, 2023-01-14:
Ah - 4G RPI :-(
On the BOOT partion there is file boot-debug.scr
May you pls rename it it boot.scr
This will give us kernel debug verbosity on screen - and maybe we will see where kernel traps...

jfabernathy, 2023-01-14:
[image]

jfabernathy, 2023-01-14:
Seems to be looping

jfabernathy, 2023-01-14:
[image]

warpme, 2023-01-14:
it looks to me like tfpt interfering with local boot.
May you pls try with disconnected Eth cable?

jfabernathy, 2023-01-14:
[image]

warpme, 2023-01-14:
give me some time - i need to ask rpi guys...

jfabernathy, 2023-01-14:
BTW, I put the SD card into a RPI 3B+ and got the same behavior with and without the Ethernet connected.

jfabernathy, 2023-01-14:
Not critical, just playing with it until my NFL football game comes on.

warpme, 2023-01-14:
huh - i just tested on mine rpi3 model B v1.2.
boots ok
this is really frustrating :-(

let me ask rpi guys where issue might be...
(btw: do you know what eeprom ver you have in yours rpi3/4)?

jfabernathy, 2023-01-14:
On the rpi4 
[image]

jfabernathy, 2023-01-14:
The rpi3 B+ is a rev 1.3

jfabernathy, 2023-01-15:
I when back to what I usually run on the RPI4. It's EndeavourOS ARM.

Linux rpi4-mythtv 5.15.84-4-rpi-ARCH #1 SMP PREEMPT Wed Jan 11 14:06:42 MST 2023 aarch64 GNU/Linux

NOTE the Ethernet port is not eth0, it's end0. Not sure when that changes

2: end0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether e4:5f:01:79:34:8a brd ff:ff:ff:ff:ff:ff
    inet 169.254.197.102/16 brd 169.254.255.255 scope link noprefixroute end0
       valid_lft forever preferred_lft forever
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether e4:5f:01:79:34:8b brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.115/24 brd 192.168.0.255 scope global dynamic noprefixroute wlan0
       valid_lft 6857sec preferred_lft 6857sec
    inet6 fe80::2208:402d:45c7:e23e/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever

model name      : ARMv8 Processor rev 3 (v8l)
BogoMIPS        : 108.00
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

Hardware        : BCM2835
Revision        : c03114
Serial          : 100000004ba90648
Model           : Raspberry Pi 4 Model B Rev 1.4

warpme, 2023-01-15:
If you still have energy to play with boot issue, may you try new versions i just uploaded?
There are:

- universal rpi34 image with latest uboot
- dedicated rpi3 image with latest uboot
- dedicated rpi4 image with latest uboot

jfabernathy, 2023-01-15:
I tried the dedicated rpi4. No luck.
[image]

jfabernathy, 2023-01-15:
Just to try something I setup a microSD card the way they say to on archlinux ARM site for the RPI4. That wouldn't boot either.  I posted that in the Archlinuxarm.org forum.
[https://archlinuxarm.org/forum/viewtopic.php?f=65&t=16325](url)

I have had good luck with EOS ARM and Ubuntu 22.10 ARM. both work great with GFX acceleration.

warpme, 2023-01-15:
Most probable theory is incompatibility of SD card you are using & u-boot sd card driver.
Your's screenshots showing issue is very early - at tpl (third program loader which is u-boot). This happens before control is passed to kernel.
This limits suspected components just to: sd card; rpi firmware and u-boot.  
Is it possible to try with different vendor sd card?

warpme, 2023-01-15:
I will also try to build image with most recent rpi firmware

jfabernathy, 2023-01-15:
The cards from yesterday were SanDisk. Today I used Samsung. Both cards with RPi OS.
[image]

jfabernathy, 2023-01-15:
There seems to be similar problems reported over on the archlinuxarm.org forum.
[https://archlinuxarm.org/forum/viewtopic.php?f=65&t=16310](url)

[https://archlinuxarm.org/forum/viewtopic.php?f=65&t=16243](url)

warpme, 2023-01-16:
Ah - this is very good info!
As for test: may you pls burn latest rpi34 image, replace boot.scr to attached one and try?
[boot.scr.zip](https://github.com/warpme/miniarch/files/10424410/boot.scr.zip)

jfabernathy, 2023-01-16:
yes that boots to kernel 6.1.5 with Ethernet connected with IP address from local router.

closed 2023-01-16 by jfabernathy