warpme/miniarch #2: SD card only or is SSD supported?

jfabernathy, 2023-01-16:
on a Raspberry Pi 4 can Miniarch be used with the boot drive being a USB3 to SATA adapter with SATA SSD?

jfabernathy, 2023-01-16:
I know I was messing with ArchlinuxARM on a SSD a while ago. I'll post what I used. It's not critical for me as I'm just playing with this. Back when I found EndeaovourOS I quit messing with ArchlinuxARM.  Here was my post on that.
[https://archlinuxarm.org/forum/viewtopic.php?f=65&t=16078](url)

Back in 2021 I did this to make it work from an SSD. 
[https://archlinuxarm.org/forum/viewtopic.php?f=67&t=15697](url)

jfabernathy, 2023-01-18:
I corrected the links above.

warpme, 2023-01-19:
well - i have this module as build-in in miniach kernel.
So i think method you described should work....

closed 2023-05-25 by warpme