warpme/miniarch #3: MPV no hardware decode Allwinner H616

paulsykes, 2023-01-17:
This project looks really interesting. I was manually picking patches from your minimyth2 project as I was trying to create my own custom image that supports hardware decode using mpv.

I've managed to successfully do this for the H6, however the H616 is proving much more difficult. Everything looks to be working okay, except the video is a glitchy green mess. When you released miniarch I was hoping it would provide a shortcut or some proof that mpv and hardware decode was possible on the H616.

Unfortunately installing MPV in miniarch using pacman gets me a version with FFmpeg 5.1.2 instead of your patched 4.4.1 version which does not support h264-drm

Is it possible to somehow ship your patched version of FFmpeg in miniarch?

Thank you for all your work, it has been invaluable in helping me get closer to my goal of working hardware decode using mpv on the H616.

warpme, 2023-01-19:
Paul,

_Is it possible to somehow ship your patched version of FFmpeg in miniarch?_
It is possible but....will potentially create hell of support on me :-(
Only reason is: dependencies
Generating MiniArch image (from downloading rootfs to customised miniarch image) is fully automated and takes something 20sec + 1..2min to packing by xz
This is possible because i'm reusing mm2 build artefacts in MiniArch.
Theoretically I can do the same for ffmpeg - but any single case where mm2 has diff in dependency compared to current archlinux arm - will cause such ffmpeg fail on archlinux arm.
we have 4 options then:
1. cross-mm2-build and keep all ffmpeg dependencies on mm2 on par with archlinux arm
2. cross-mm2-build of ffmpeg and build miniarch specific ffmpeg dependencies (so still close track atrclinux arm state)
3. cross-mm2-build of ffmpeg with mm2 dependencies then do MiniArch "bundle" (to remove any dependencies from archlinux arm)
4. natively patch + build ffmpeg on MiniArch

p.1 - i don't have time/motivation
p.2 - see p.1
p.3 - interesting. we need to look how to do this?
p.4 - seems best / most clean /standard approach

If you have spare time to do i.e. dedicated, patched natively build ffmpeg - here are video decode/drm_prime patches:
ffmpeg4: https://github.com/warpme/minimyth2/tree/master/script/lib/ffmpeg4/files
ffmpeg5: https://github.com/warpme/minimyth2/tree/master/script/lib/ffmpeg5/files

in such case - pls create pull request in https://github.com/warpme/archlinuxarm for benefits of community.

Above ffmpeg5 i tested under mm2 with current kernel and Kodi20: it works well on H6 but fails on H616.

This is result of hdmi code of h616 display driver - it does not yet properly support YUV planes.
(H616 has newer display engine (DE3.3) than H6 (DE3) and changes are big enough so it's not trivial to support
and I think that there are additional corner cases which are not problematic on DE3 but they are on DE3.3. If I'm right, whole DRM driver needs to be reworked).

As I have perfectly working H616 in MythTV - i would say: MyhtTV has sort of better implemented video pipeline than KODI as myth seems to more universally exploiting/dealing with HW limitations....

paulsykes, 2023-01-23:
Agree, p.4 sounds like a good approach. I'll give this a go.

Briefly looking at the MythTV source, it looks like the video pipeline relies on the GPU a little more after the video engine.

warpme, 2023-01-23:
Indeed!
Mark done incredible good work with myth video pipeline.
btw: Are you using mythtv?

paulsykes, 2023-01-25:
@warpme Please can you release an SD card image for Orange Pi 3 (H6)? I've tested my v4l2_request patched ffmpeg5 PKGBUILD on h616, it builds okay, yet fails to output video, which is expected. I would like to try the PKGBUILD I've created on a working device, however the only h6 device I have is the Orange Pi 3, and currently no miniarch SD card image exists for this.

warpme, 2023-01-25:
Paul, 
I uploaded OPi3 image - but i give low chances it will work....
I remember in past something was not ok with my mm2 opi3 patches.
Unfortunately I don't have opi3 HW to fix this :-(

paulsykes, 2023-01-26:
@warpme Opi3 image worked okay, except for no Ethernet. I cheated and used a USB to Ethernet adaptor. PKGBUILD for patched FFmpeg with v4l2_request support seemed to work okay. I will open a pull request once I've tidied things up a little.

warpme, 2023-01-26:
may you pls give me dmesg from opi3?

paulsykes, 2023-01-27:
Sure, no problem. 

Bootlog and dmesg from Opi3. For clarity this is for the Opi3, not the Opi3_lts.

[bootlog_and_dmesg_Opi3.txt](https://github.com/warpme/miniarch/files/10515093/bootlog_and_dmesg_Opi3.txt)

warpme, 2023-02-26:
Paul,
It looks to me - without hw at hand - i can't move forward...

warpme, 2023-05-05:
Due lack of HW at my hands for tests/devel i'm closing with status: no HW req. for devel.
Thx!

closed 2023-05-05 by warpme