warpme/miniarch #4: xr819 support

dchauran, 2023-01-19:
The link to the supported devices implies that an H6 with xr819 should work, however I only see the wired connection. I see the frmware but no xradio modules. Is there a different module I should be loading?

dchauran, 2023-01-19:
Hello - hardware is a "t95 max" stb, visual inspection of the hardware shows an xr819, 32gb toshiba storage (which I am able to boot the OS from!). Mali T-720, 4gb ram.

I understand this is a random cheap amazon device, but it very nearly works perfectly out of the box.

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.1.5 (piotro@minimyth2-aarch64) (aarch64-minimyth-linux-gnu-gcc (GCC) 12.2.0, GNU ld (GNU Binutils) 2.38) #1 SMP PREEMPT Thu Jan 12 18:28:43 CET 2023
[    0.000000] Machine model: Tanix TX6
[    0.000000] efi: UEFI not found.
[    0.000000] earlycon: uart0 at MMIO32 0x0000000005000000 (options '115200n8')
[    0.000000] printk: bootconsole [uart0] enabled
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000040000000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x00000000ffffffff]
[    0.000000] cma: Reserved 384 MiB at 0x00000000e4800000
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 29 pages/cpu s80040 r8192 d30552 u118784
[    0.000000] pcpu-alloc: s80040 r8192 d30552 u118784 alloc=29*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 774144
[    0.000000] Kernel command line: root=PARTUUID=346d1a6a-02 rw rootwait earlycon console=ttyS0,115200n8 logo.nologo vt.cur_default=1 cma=384M deferred_probe_timeout=0
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 2651604K/3145728K available (14272K kernel code, 3514K rwdata, 5972K rodata, 6592K init, 641K bss, 100908K reserved, 393216K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 50439 entries in 198 pages
[    0.000000] ftrace: allocated 198 pages with 4 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=4.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 30 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000000] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.000170] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 79635851949 ns
[    0.000809] Console: colour dummy device 80x25
[    0.000858] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=80000)
[    0.000873] pid_max: default: 32768 minimum: 301
[    0.001252] Mount-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.001287] Mountpoint-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.002693] cacheinfo: Unable to detect cache hierarchy for CPU 0
[    0.003650] cblist_init_generic: Setting adjustable number of callback queues.
[    0.003664] cblist_init_generic: Setting shift to 2 and lim to 1.
[    0.003778] cblist_init_generic: Setting shift to 2 and lim to 1.
[    0.003891] cblist_init_generic: Setting shift to 2 and lim to 1.
[    0.004169] rcu: Hierarchical SRCU implementation.
[    0.004175] rcu:     Max phase no-delay instances is 1000.
[    0.006530] EFI services will not be available.
[    0.007193] smp: Bringing up secondary CPUs ...
[    0.008432] Detected VIPT I-cache on CPU1
[    0.008541] cacheinfo: Unable to detect cache hierarchy for CPU 1
[... 394 lines cut ...]
[    8.693859] usbcore: registered new interface driver uas
[    9.654700] scsi 0:0:0:0: Direct-Access              USB DISK 3.0     PMAP PQ: 0 ANSI: 6
[    9.655691] sd 0:0:0:0: Attached scsi generic sg0 type 0
[    9.657432] sd 0:0:0:0: [sda] 60623680 512-byte logical blocks: (31.0 GB/28.9 GiB)
[    9.658627] sd 0:0:0:0: [sda] Write Protect is off
[    9.658647] sd 0:0:0:0: [sda] Mode Sense: 2b 00 00 08
[    9.659765] sd 0:0:0:0: [sda] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
[    9.915505]  sda: sda1
[    9.916021] sd 0:0:0:0: [sda] Attached SCSI removable disk
[... cut ...]
[   11.935698] audit: type=1300 audit(1674152220.278:49): arch=c00000b7 syscall=64 success=yes exit=4 a0=3 a1=fffff5e70be0 a2=4 a3=ffff9648f9c0 items=0 ppid=342 pid=349 auid=1001 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=1 comm="sshd" exe="/usr/bin/sshd" key=(null)
[   11.935721] audit: type=1327 audit(1674152220.278:49): proctitle=737368643A20646572656B205B707269765D
[   12.051779] dwmac-sun8i 5020000.ethernet end0: Too many address, switching to promiscuous
[   12.688821] systemd-journald[262]: Failed to set ACL on /var/log/journal/a20f346ca7114c18808af150ca23caa1/user-1001.journal, ignoring: Operation not supported
[derek@archlinux ~]$

warpme, 2023-01-20:
Hmm,
I just flashed SD card for TanixTX6s, inserted into my box i have and wlan0 is present.
in dmesg i  have:
```
[   13.826915] Firmware download completed.
[   13.831433] CW1200 WSM init done.
                  Input buffers: 30 x 1632 bytes
                  Hardware: 7.9
                  WSM firmware [XR_C01.08.52.65 Jul 16 2019 11:09:57], ver: 8, build: 5264,   api: 1060, cap: 0x0003
[   13.884821] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
[   13.893122] Registered as 'phy0'
```

You have t95 box so most probable reasons of no wifi:
1.different power-on/reset seq is needed (probable)
2.different reset/power_on gpio wiring (image i published is for tanix-tx6s, you have t95)

To test - i prepared dtb with different approach to p.2
pls copy it to /boot part named as `h616_dtb` 

If this will not help: is it possible to get android gpio info from running android on your's device?
(boot android, install terminal, cat sysfs related to gpio)
[sun50i-h616-t95.dtb.zip](https://github.com/warpme/miniarch/files/10464230/sun50i-h616-t95.dtb.zip)

dchauran, 2023-01-20:
This is an h6 based t95, not h616. I haven't been able to get this on my h616 based t95 at all (it doesn't want to boot from SD). I can reload android if necesary but wanted to clarify this first.

warpme, 2023-01-21:
Ah - so you have h6 based t95? It is surprise for me that there are t95 devices with h6 soc...
When you are saying "_it doesn't want to boot from SD_"- is this mean:
1. not even start to boot from SD or
2. it starts to boot but fails?

Have you capability to use uart and see output telling us at what stage boot fails?

It is probable that you have case 1 and i.e sd card is not recognised due wrong cd-gpios declared in dt vs. wiring you have on your board. 

If you still want to play with this box - best will be to boot android and see on terminal what gpio are used for peripherals.
Ideally: retrieve android dt by adb

If you be able to determine correct gpio for sd card detection - you can sue tx6 dt and modify accordingly https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/arch/arm64/boot/dts/allwinner/sun50i-h6-tanix.dtsi?h=v6.1.7#n126

btw: don't be surprised by this issues: supporting cheap chinese tvboxes is pure hell: we have 10 variants of the same named tvbox with different soc, dram, wifi, gpio wiring etc.

dchauran, 2023-01-21:
Hey, the tx6-mini dtb worked to get wi-fi on the h6! I learned a lot in this process, and I greatly appreciate your patience.

Re: the 616, it fails to even start to boot. If I cannot figure that one out I will open another issue. Closing this issue as it is resolved.

closed 2023-01-21 by dchauran