warpme/miniarch #5: Initial update fails following 'Quick Start' instructions

paulsykes, 2023-01-23:
Following the 'Quick Start' instructions results in an error when doing initial update.

```
(83/83) checking keys in keyring                   [######################] 100%
warning: Public keyring not found; have you run 'pacman-key --init'?
downloading required keys...
error: keyring is not writable
error: required key missing from keyring
error: failed to commit transaction (unexpected error)
Errors occurred, no packages were upgraded.
Packages to keep:
  All locally installed packages

Cache directory: /var/cache/pacman/pkg/
:: Do you want to remove all other packages from cache? [Y/n] 
```

Step to reproduce
1. [Download](https://github.com/warpme/miniarch/blob/master) SD card Image
2. burn image on SD card
insert SD Card into SBC/TVbox
3. power-on
4. login as root:root
5. type 'start'
6. select (1) to do full update

Maybe reverse the order of steps 1 and 2 in point 5 (do update) of the Quick Start?

warpme, 2023-01-23:
Paul,
Oh - it looks like you are right!
I don't remember where i learn (in past) that: firs we need -Syu, then init keyring.
I just looked on archlinuxarm install desc. and indeed currently it looks like first required step is: keyring init. 
I'll fix that!
Thx

warpme, 2023-01-24:
fixed in c0d270537f2826c4655b0e47c62acdfb39d5f5c3
thx for pointing this!

closed 2023-01-24 by warpme