warpme/miniarch #6: How do I build the kernel?

dchauran, 2023-01-26:
I need to enable ACM support, however I am unable to get the kernel to build. What am I doing wrong?

```
[derek@archradio linux-aarch64]$ sudo make KCONFIG_CONFIG=~/running.config
  SYNC    include/config/auto.conf
  HOSTCC  scripts/kconfig/conf.o
  HOSTLD  scripts/kconfig/conf
drivers/ras/Kconfig:34: can't open file "arch/x86/ras/Kconfig"
make[2]: *** [scripts/kconfig/Makefile:77: syncconfig] Error 1
make[1]: *** [Makefile:697: syncconfig] Error 2
make: *** [Makefile:798: include/config/auto.conf] Error 2
make: *** [include/config/auto.conf] Deleting file 'include/generated/rustc_cfg'
make: *** [include/config/auto.conf] Deleting file 'include/generated/autoconf.h'
```

warpme, 2023-01-26:
Well - you may look https://github.com/warpme/miniarch/wiki/Building-MiniArch
(btw: any improvements in wiki are highly appreciated!)

warpme, 2023-01-26:
"build on box far more efficiently" - well - in my case it is disputable. Building on my x86 is 9..11x faster than on box...
For native build You may try: https://github.com/warpme/archlinuxarm/tree/master/linux

dchauran, 2023-01-27:
Thanks, I tried it and hit some issues. First, the patches to src/linux-6.1/drivers/media/v4l2-core/v4l2-ioctl.c fail to apply. I removed those patched as they appeared related to video drivers I don't care about for my application, however the build is failing:

```
  CC      drivers/iommu/sun50i-iommu.o
drivers/iommu/sun50i-iommu.c: In function ‘sun50i_iommu_map’:
drivers/iommu/sun50i-iommu.c:529:24: error: ‘SPAGE_SIZE’ undeclared (first use in this function); did you mean ‘HPAGE_SIZE’?
  529 |         pages = size / SPAGE_SIZE;
      |                        ^~~~~~~~~~
      |                        HPAGE_SIZE
drivers/iommu/sun50i-iommu.c:529:24: note: each undeclared identifier is reported only once for each function it appears in
drivers/iommu/sun50i-iommu.c: In function ‘sun50i_iommu_unmap’:
drivers/iommu/sun50i-iommu.c:566:24: error: ‘SPAGE_SIZE’ undeclared (first use in this function); did you mean ‘HPAGE_SIZE’?
  566 |         pages = size / SPAGE_SIZE;
      |                        ^~~~~~~~~~
      |                        HPAGE_SIZE
make[3]: *** [scripts/Makefile.build:250: drivers/iommu/sun50i-iommu.o] Error 1
make[2]: *** [scripts/Makefile.build:500: drivers/iommu] Error 2
make[1]: *** [scripts/Makefile.build:500: drivers] Error 2
make: *** [Makefile:1992: .] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
```

dchauran, 2023-01-27:
I suspect this patch is missing as it defines spage_size, but it's odd that it has a later date than patch 0531 in the build which introduced the reference: 
https://lore.kernel.org/stable/20221228144348.982291413@linuxfoundation.org/

warpme, 2023-01-28:
Derek,
Im a bit lost with this...
I have building no problem mainline 6.1.8 kernel on my minimyth2/miniarch build env - so i suspect root cause of your's issue is related not to patches we are using - but rather to somehow build env you are using.
How exactly are you building kernel?
(asking as week or so ago i build natively kernel from https://github.com/warpme/archlinuxarm/tree/master/linux with no problems)

dchauran, 2023-01-30:
I sent a PR to the kernel readme that contains the exact steps I followed.

After removing the 2 patch sections that gave me trouble, and adding the definition for spage_size manually I was able to build the kernel, and got my ACM device running. There was a small conflict during install with the linux-api-headers as well, but I let it overwrite those:

```
linux-aarch64-api-headers: /usr/include/asm/hwcap.h exists in filesystem (owned by linux-api-headers)
linux-aarch64-api-headers: /usr/include/asm/sve_context.h exists in filesystem (owned by linux-api-headers)
```