warpme/miniarch #7: Support for H96 MAX Allwinner H616 Ethernet

1gordingoxtozo, 2023-02-22:
Hello,

I have a H96 MAX TV Box with Allwinner H616 SoC, i could boot using the OrangePi Zero 2 image but i have the problem like all others OS's that i tried (Armbian, OrangePi Zero 2) the ethernet is not recognized. I have been searching for help in many forums but couldn't find nothing that could help.

This is the photos that i took from the TV Box PCB with all chips, if you please could add drivers for this ethernet chip or tell me how to do that, i would really appreciate that.
[image]
[image]

warpme, 2023-02-24:
May you pls try with sun50i-h616-t95.dtb ?

Procedure: pls look https://github.com/warpme/minimyth2/wiki/Running-on-ARM-boards-or-TVboxes#quick-start-guide

it will be useful to have dmesg output from each try…

warpme, 2023-05-05:
Closed due no replay from issue reporter.

closed 2023-05-05 by warpme