warpme/miniarch #8: Tanix Tx6s does not boot with the SD image

Astaldo2, 2023-03-27:
I burn the image of the tanix tx6s and when I insert the sd card, it does nothing, it stays black and the display boot does not jump, however, if I remove the sd card and turn on it starts fine with android

warpme, 2023-03-27:
Oh - there is multiple version of tx6s with different types of ram chips (ddr3, lpddr3, etc)
To move forward - output of debug uart will be needed...
Are you capable to connect uart and get output from it?
If not - i rather can't help without logs...

LarsLinux93, 2023-03-27:
Hi @warpme,
I just received my TX6S in the mail and it seems I got a model with LPDDR3 which won't boot.
Before I open it and install UART, I will try to troubleshoot a little bit myself with different firmwares. 
Thanks for you work, I'll send a little gratitude via PayPal.

warpme, 2023-03-28:
Lars,
I can try build image with experimental lpddr3 support. give me sec...

warpme, 2023-03-28:
Lars, Astaldo2

May you pls try https://github.com/warpme/minimyth2/releases/download/v12.13.0-v33-Pre-1175-g91e3da68e6/MiniMyth2-armv8-master-12.13.1.r59-board-h616.tanix_tx6s_lpddr3-SD-Image.img.xz

It will be good to see uart output (also from vendor android)

LarsLinux93, 2023-03-28:
Hi @warpme, thanks for the quick reply & rebuild. However, same results as @Astaldo2 
I set up a build environment with Ubuntu 22.04 meanwhile and will make myself familiar with the structure a bit. 
Until the weekend I will try to open the box without breaking it and report my findings. I hope Astaldo2 and I have the same hardware revision. For UART, I'd have to go to the local hackerspace, don't know when that's going to happen. 
Have a great day

LarsLinux93, 2023-03-29:
Just opened the box to check for markings and information about revision: CS_H616_TX6S_B4_V4.1 20210410
PMIC is now APX313A
Storage moved to the other side of the PCB compared to what's on sunxi-linux and is marked SAMSUNG 540 KLMCG8GEND-B031 S3DPR020
RAM chips are marked as follows:
Soc-Side:
4NE77 D9PQL ZNC6
4IE77 D9PQL PL92
4HE77 D9PQL P5NF
4IE77 D9PQL PVV2
Back-Side:
4GE77 D9PQL N6R9
4HE77 D9PQL P4JK
4HE77 D9PQL P4PP
4IE77 D9PQL PL9F

[image]

[image]

LarsLinux93, 2023-03-29:
I just checked the RAM chips using[ this tool ](https://www.micron.com/support/tools-and-utilities/fbga?fbga=D9PQL#pnlFBGA) and they seem to be the same chips as on the board in[ sunxi-linux wiki ](https://linux-sunxi.org/H616) i.e. MT41K1G4RH-125:E DDR3L 800Mhz. The Samsung part is a [EMMC 5.0](https://semiconductor.samsung.com/estorage/emmc/emmc-5-0/klmcg8gend-b031/). I'm glad that the WiFi chip remained the same(AP6330-X), so that will probably work. 
For the PMIC AXP313a some recent patches are found in the[ sunxi-linux mailing list](https://lore.kernel.org/linux-sunxi/?q=AXP313a)
I hope this helps until we get more UART information.
Kind regards

warpme, 2023-03-29:
For PMIC ware a bit prepared already...https://github.com/warpme/minimyth2/blob/master/script/kernel/linux-6.2/files/0637-drivers-mfd-axp20x-add-support-for-AXP313a-PMIC.patch

LarsLinux93, 2023-04-06:
I picked up a USB UART converter and sent you some money on PayPal. The pins are smaller pitch than the cables I got, so I'll try to solder some wires directly instead.

LarsLinux93, 2023-04-07:
Hi @warpme
I just started gathering the UART output from the Tanix T6S V4.1.
Here is the output of [MiniArch-20230212-6.2.6-board-h616.tanix_tx6s-SD-Image](https://github.com/warpme/miniarch/releases/download/20230226-6.2-6b34b2ce/MiniArch-20230212-6.2.6-board-h616.tanix_tx6s-SD-Image.img.xz):
`
U-Boot SPL 2021.04 (Mar 16 2023 - 19:23:57 +0100)
DRAM: 8192 MiB
Failed to set core voltage! Can't set CPU frequency
`
Here the output of [MiniMyth2-armv8-master-12.13.0.r1175-board-h616.tanix_tx6s-SD-Image](https://github.com/warpme/minimyth2/releases/download/v12.13.0-v33-Pre-1175-g91e3da68e6/MiniMyth2-armv8-master-12.13.0.r1175-board-h616.tanix_tx6s-SD-Image.img.xz), guess it is the same u-boot as in MiniArch:
`
U-Boot SPL 2021.04 (Feb 25 2023 - 15:19:32 +0100)
DRAM: 8192 MiB
Failed to set core voltage! Can't set CPU frequency
`

Here from the [MiniMyth2-armv8-master-12.13.1.r59-board-h616.tanix_tx6s_lpddr3-SD-Image](https://github.com/warpme/minimyth2/releases/download/v12.13.0-v33-Pre-1175-g91e3da68e6/MiniMyth2-armv8-master-12.13.1.r59-board-h616.tanix_tx6s_lpddr3-SD-Image.img.xz) you suggested earlier for completeness, the error is ofc due to my board having DDR3L RAM:
`
U-Boot SPL 2023.01-rc3 (Mar 28 2023 - 10:35:02 +0200)
Error, wrong i2c adapter 1 max 1 possible
DRAM:This DRAM setup is currently not supported.

resetting ...
`

As reference I also uploaded the Stock Android bootlog here: https://gist.github.com/LarsLinux93/de09c7bcbf2472b5fd9354de08083374

I'll try to work myself through the sunxi-linux wiki regarding mainline u-boot to be able to test and give feedback better. 
Enjoy the weekend and many thanks.
±Lars

warpme, 2023-04-07:
Lars,

Thx for logs.
H313/H616 TV boxes are in multiple ver: 

- dram: ddr3 or lpddr3
- Int Eth phy: AC200 or AC300
- PMIC: axp305 or axp313a (aka axp1530)

To address this i developed uboot (and ATF) which should auto-detect all combinations (see https://github.com/warpme/minimyth2/commit/cc455b8f28ff4b306bceaee82c9371e99e2f966b)

As MiniMyth2 and MiniArch are sharing exactly the same: SD card layout, kernel and bootloader - for testing pls use just MiniMyth2. When we will get well booting/working MiniMyth2 on you box- i will publish also MiniArch images.

On picture of bottom side of PCB I can read what PMIC you have - but chip phy size suggests axp313a. 

I prepared image with new bootloader which should support axp313a: https://github.com/warpme/minimyth2/releases/download/v12.13.1-v34-Pre-89-g0363c0dd1b/MiniMyth2-armv8-master-12.13.1.r89-board-h616.tanix_tx6s-SD-Image.img.xz

It should give us correct working u-boot but probably kernel will fail due device-tree not ready yet for your box...
May you try and give me back uart logs?

btw: sd card has boot script with turn-off kernel verbose logging. pls overwrite boot.scr by boot-debug.scr

LarsLinux93, 2023-04-07:
Hi @warpme,

Thanks for the quick reply.
I just quickly wrote the new build to an SD and got the following on UART:

`U-Boot SPL 2021.04 (Apr 07 2023 - 18:28:28 +0200)
DRAM: 16384 MiB
Failed to set core voltage!   set CPU 792000000hz frequency`

After looking through your MiniMyth repo a little, I understood how MiniArch is more of a byproduct .
Should I open or add to an[ issue](https://github.com/warpme/minimyth2/issues/49) there instead?  I will have more time to test with verbose logging tomorrow.

Kind regards,
~Lars

warpme, 2023-04-08:
Lars,

I forget switch uboot to use axp313 pmic.
I prepared SD image with added axp313 in uboot/kernel: https://github.com/warpme/minimyth2/releases/download/v12.13.1-v34-Pre-89-g0363c0dd1b/MiniMyth2-armv8-master-12.13.1.r89-board-h616.tanix_tx6s_axp313-SD-Image.img.xz
pls try....

Astaldo2, 2023-04-08:
No boot for me..
Imagen muy box tx6s:
[image]
[image]

Astaldo2, 2023-04-08:
[image]
[image]
[image]
[image]
[image]

Astaldo2, 2023-04-08:
no boot for me
Imagen muy tanix tx6s
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

warpme, 2023-04-08:
[Astaldo2](https://github.com/Astaldo2),

of course you are using this image https://github.com/warpme/minimyth2/releases/download/v12.13.1-v34-Pre-89-g0363c0dd1b/MiniMyth2-armv8-master-12.13.1.r89-board-h616.tanix_tx6s-SD-Image.img.xz right?
If yes - without uart log i can't say where issue is....

LarsLinux93, 2023-04-08:
The latest build booted! Thank you so much!
[image]
UART remains silent though and I didn't figure out how to drop to command line for more info yet.
May I ask for a build of MiniArch with the U-Boot&Kernel?
Kind regards
~Lars

warpme, 2023-04-08:
Lars,
Perfect!
Many thx for your's patience :-)

MiniArch: here it is: https://github.com/warpme/miniarch/releases/download/20230226-6.2-6b34b2ce/MiniArch-20230212-6.2.10-board-h616.tanix_tx6s_axp313-SD-Image.img.xz

For logging:  sd card has boot script with turn-off kernel verbose logging. pls overwrite boot.scr by boot-debug.scr

Re to your Q "I understood how MiniArch is more of a byproduct " - yes :-).
MiniArch reuses all work i'm doing for MiniMyth2. With [Endeavour OS](https://arm.endeavouros.com) you can have 8 different DE on your box...

LarsLinux93, 2023-04-08:
Hi,
I tried finding the error log in /var/*dontrememberfullpath* on the SD to upload it, but there does not seem to be a /var/ directory on ROOT partition. Instead I switched to boot-debug.scr and was able to capture the boot log. Uploaded here: https://gist.github.com/LarsLinux93/e63182c6c67de5488ffa4270561267ab
Kind regards,
~Lars

Edit: Started typing, then saw your comment. Moving to MiniArch now :)

warpme, 2023-04-08:
Ah - I see no Eth.
It looks Your board uses Eth MDIO as addr=1 not 0

May you pls update sd card bootloader with this: 
[u-boot-sunxi-with-spl.bin.zip](https://github.com/warpme/miniarch/files/11183600/u-boot-sunxi-with-spl.bin.zip)
and give me uart logs?

How to update bootloader:
-insert sd card
-run `sudo dd if=u-boot-sunxi-with-spl.bin of=/dev/sd[x] bs=1k seek=8`
where /dev/sd[x] is block device with Yours sdcard
(pls be careful: using wrong block device may destroy your data / make machine non-bootable)

LarsLinux93, 2023-04-08:
@warpme 
I tethered my phone for internet connectivity for now and started the EndeavourOS installation for fun and giggles.
Do you have tips for backing up the original contents of the EMMC to external disk? I am familiar with dd and would just write to an image file, but I think there should be a way to compress this image file while creating it. 
Will update the bootloader later and try to find an ethernet cable to test. If everything goes smooth, I might install to EMMC in the evening.

Wifi also isn't up yet, seems to be related to this: https://github.com/warpme/minimyth2/issues/46

Thank you for you swift replies and support.
FOSS is really fun like this :)
~Lars

LarsLinux93, 2023-04-11:
Hi @warpme,

I tested a little more and can report that the visual artifacts are only happening on Xorg, running Gnome or KDE Wayland is pretty usable and seems to suffer mostly from running from the SDcard.

After creating a backup of the internal storage, I tried to boot from EMMC as detailed here, without success: 
https://u-boot.readthedocs.io/en/latest/board/allwinner/sunxi.html
And here:
https://linux-sunxi.org/Bootable_eMMC
However,  I didn't gather UART logs yet.

Ethernet, WiFi, Bluetooth and audio codec are not working yet, I have uploaded the output of dmesg here: https://gist.github.com/LarsLinux93/df1eaf2b25b57431209b12d6b018a345

I will probably not have time to investigate much until the end of the month, but will be happy to test new builds and provide logs. 

Kind regards,
~Lars

rayzpham, 2023-04-12:
Any can help me use "start" of miniarch
I use img: [MiniArch-20230212-6.2.10-board-h616.tanix_tx6s_axp313-SD-Image.img.xz](https://github.com/warpme/miniarch/releases/download/20230226-6.2-6b34b2ce/MiniArch-20230212-6.2.10-board-h616.tanix_tx6s_axp313-SD-Image.img.xz)
I tried setup as Quick Start and I error when use step 1 as image ( I tried use a network cable but it's still the same )
Thanks

[image]

LarsLinux93, 2023-04-12:
Hey @rayzpham,
I followed the instructions here to tether from my Android phone: https://wiki.archlinux.org/title/android_tethering
Neither WiFi nor Ethernet work on my device yet using the same image. 
Kind regards,
~Lars

LarsLinux93, 2023-04-19:
Hi,
Latest build for tanix tx6s with AXP313 doesn't boot up, but the screen is detecting some input and flickers. I am still using the HDMI cable that came in the box, which is rather low quality. I will see if a new cable gives better results and otherwise collect UART logs in the evening.
Where can I track the mainline status for the H616? 
I tried to build a fedora kernel with the defconfig from sunxi Linux for this device, but only the H96 Mate is available.
Kind regards,
Lars

LarsLinux93, 2023-04-19:
Hi,
I just captured the bootlog until the point where it says "cut here".
Uploaded them here: https://gist.github.com/LarsLinux93/0105fb0be057dc1c1407e8cc2401645d
Display is showing some artifacts and that there's active input with correct resolution.
This error is repeated over and over:
`[    4.000866] ------------[ cut here ]------------
[    4.005480] WARNING: CPU: 1 PID: 38 at drivers/iommu/sun50i-iommu.c:540 sun50i_dte_get_page_table+0x27c/0x2c0
[    4.015383] Modules linked in:
[    4.018432] CPU: 1 PID: 38 Comm: kworker/u8:1 Tainted: G        W          6.2.11 #1
[    4.026163] Hardware name: Tanix TX6s (axp313) (DT)
[    4.031032] Workqueue: events_unbound deferred_probe_work_func
[    4.036860] pstate: 80000005 (Nzcv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[    4.043811] pc : sun50i_dte_get_page_table+0x27c/0x2c0
[    4.048943] lr : sun50i_dte_get_page_table+0x94/0x2c0
[    4.053988] sp : ffff80000a21b510
[    4.057296] x29: ffff80000a21b510 x28: 0000000000100000 x27: ffff800009364720
[    4.064425] x26: 0000000000e10000 x25: ffff0000c0464810 x24: 00000000c0493800
[    4.071554] x23: 0001000000000000 x22: ffff0000c2900f80 x21: ffff0000c2b2a900
[    4.078682] x20: ffff0000c2cd7fd0 x19: ffff0000c0493800 x18: 0000000000000014
[    4.085810] x17: 000000005472def9 x16: 000000000000007c x15: 0000000000000000
[    4.092938] x14: 0000000000000000 x13: 0000000000000000 x12: 0000000000000000
[    4.100065] x11: 0000000000000000 x10: 0000000000000000 x9 : ffff800008135060
[    4.107194] x8 : 0000000000000000 x7 : 0000000000000000 x6 : 0000000000000000
[    4.114321] x5 : ffff0000b9002400 x4 : ffff0000c0493c00 x3 : 0000000000000000
[    4.121449] x2 : ffff800009400000 x1 : 0000000100493800 x0 : 00000000f9002000
[    4.128577] Call trace:
[    4.131017]  sun50i_dte_get_page_table+0x27c/0x2c0
[    4.135803]  sun50i_iommu_map+0x44/0x140
[    4.139722]  __iommu_map+0x154/0x2cc
[    4.143295]  __iommu_map_sg+0xa4/0x160
[    4.147040]  iommu_map_sg_atomic+0x18/0x20
[    4.151130]  __iommu_dma_alloc_noncontiguous.constprop.0+0x230/0x3e0
[    4.157474]  iommu_dma_alloc+0x168/0x224
[    4.161391]  dma_alloc_attrs+0xe8/0xf0
[    4.165136]  drm_gem_dma_create+0xa4/0x150
[    4.169228]  drm_gem_dma_create_with_handle+0x2c/0xd0
[    4.174274]  drm_gem_dma_dumb_create_internal+0x60/0x70
[    4.179492]  drm_sun4i_gem_dumb_create+0x30/0x40
[    4.184105]  drm_mode_create_dumb+0x94/0xb0
[    4.188282]  drm_client_framebuffer_create+0xa0/0x20c
[    4.193327]  drm_fbdev_fb_probe+0x6c/0x1c0
[    4.197419]  drm_fb_helper_single_fb_probe+0x2f0/0x454
[    4.202551]  __drm_fb_helper_initial_config_and_unlock+0x44/0x150
[    4.208637]  drm_fb_helper_initial_config+0x4c/0x60
[    4.213509]  drm_fbdev_client_hotplug+0xac/0x104
[    4.218121]  drm_fbdev_generic_setup+0x94/0x15c
[    4.222646]  sun4i_drv_bind+0x154/0x1b0
[    4.226477]  try_to_bring_up_aggregate_device+0x68/0x154
[    4.231784]  __component_add+0xc0/0x1b4
[    4.235616]  component_add+0x18/0x20
[    4.239188]  sun8i_dw_hdmi_probe+0x20/0x30
[    4.243278]  platform_probe+0x6c/0xc0
[    4.246934]  really_probe+0xc4/0x3dc
[    4.250506]  __driver_probe_device+0x80/0x190
[    4.254858]  driver_probe_device+0x40/0x110
[    4.259036]  __device_attach_driver+0xb8/0x150
[    4.263473]  bus_for_each_drv+0x7c/0xd0
[    4.267303]  __device_attach+0xa0/0x1c0
[    4.271135]  device_initial_probe+0x18/0x20
[    4.275314]  bus_probe_device+0x9c/0xac
[    4.279144]  deferred_probe_work_func+0xb8/0x110
[    4.283756]  process_one_work+0x1c8/0x434
[    4.287761]  worker_thread+0x1fc/0x40c
[    4.291505]  kthread+0xe8/0xf4
[    4.294555]  ret_from_fork+0x10/0x20
[    4.298127] ---[ end trace 0000000000000000 ]---
`
After some 10k+ lines, the log contains mostly page faults at sun50i-iommu:
`sun50i-iommu 30f0000.iommu: Page fault for 0x00000000ff000000 (master 0, dir rd)`

Kind regards,
Lars

warpme, 2023-04-19:
Lars,

Ah ok. You have 4G system.
Recently I added https://github.com/warpme/minimyth2/commit/d7448a475968f69e5d78c5d8f840ccf563459521
Unfortunately it breaks 4G systems.
We are working on fix.
Sorry for this!
Stay tuned....

LarsLinux93, 2023-05-08:
Heya, 
any update for the AXP313 version of the TX6S? 
I see some work related to H616 in Minimyth, but don't have any bootable image since the last version. Also no changes in upstream UBoot or Kernel that seem to be relevant. 
The Minimyth build is too big for me to try myself. 
Kind regards,
~Lars

warpme, 2023-05-25:
Lars,
It looks we fixed (i hope) iommu issue on 4G systems (https://github.com/warpme/minimyth2/commit/2348373fd98ce33eaba53caa5579f51d37d54890)
I uploaded new h313/h616 images with this fix.
May you try and see how it goes?

LarsLinux93, 2023-05-26:
Piotr,
Thanks for notifying!
I'll try to test it Sunday evening, otherwise during next week.

Have a good weekend,
~Lars

QualFoiKK, 2023-07-02:
Can someone help me,
how do I make the TX6s boot from the SD card? I've tried several images, but it always boots Android.

Thanks.

closed 2023-07-16 by warpme