warpme/miniarch #10: SSV6256p / SV6256P wifi driver

mercs759, 2023-05-03:
Hi,
SDIO bus correctly detects my wifi chip , and i can find it under : 

/sys/bus/mmc/devices/mmc3:0001/uevent 
MMC_TYPE=SDIO
SDIO_ID=3030:3030
SDIO_REVISION=8.0
SDIO_INFO1=SouthSiliconValley.Inc.
SDIO_INFO2=Turismo
SDIO_INFO3=0
SDIO_INFO4=00

(I don't know why it's put under "mmc3" , there's no mmc3 reference in miniarch dtb or in my filesystem device tree, but the hardware address is the right one, so.... it's that)

now,
SV6256P chip on my board is made by Icomm, but in SDIO_INFO1 there is still the original company : South Sil Vall; moreover SDIO_ID is the same of the old SSV6256P

So, it seems (as reported by others on armbian forums) that SSV and SV are the same wifi chip.

The original SSV6256P has a driver (ssv6x5x) under armbian 

from 12656-csc-armbian-for-rk322x-tv-boxes/page/18/ :
"the ssv6x5x is only for ssv6256p chip. People at SSV company are not very well acquainted with names and IDs, in fact they used the very same ID 3030:3030 for both ssv6051 and ssv6256p wifi chips and the driver name is also confusing (looking at ssv6x5x you may think it works for ssv6051 too, but it doesn't  )"

ssv6x5x is here:
https://github.com/armbian/firmware/blob/master/ssv6x5x-wifi.cfg

 Icomm SV6256P datasheet is here :

http://www.icomm-semi.com/Uploads/Temp/files/2022-08-19/SV6256P%20Datasheet_V1.0.pdf

on chinese sites there are other resources , but there is a paywall for downloading
https://download.csdn.net/download/lihuan680680/20670850

warpme, 2023-05-04:
Well,
1. There is no 6.3 code for this chip.
2. support 2 different silicons acting with the same ID is NIGHTMARE

By above I tend to close this ticket with status: closed due no support planned for this hw.
Will it be ok?

mercs759, 2023-05-05:
as I wrote above "i'd say not to spend time trying to port the driver to some new kernel." ;) 
yep I would close the ticket.

warpme, 2023-05-05:
closed due no support planned for this hw

closed 2023-05-05 by warpme

slckgms, 2026-09-04:
For anyone finding this later: the "no 6.x code for this chip" part is out of date. https://github.com/cdhigh/armbian_sv6256p builds on modern kernels, and I have it running as a module on an H616 box against a stock 6.12 Armbian kernel.

Two build gotchas on arm64, both needed: drop `-fpatchable-function-entry=0` from the driver's `config_common.mak` (it overrides the kernel's `=4,2` and the module then won't load), and make sure `CONFIG_DEBUG_INFO_BTF_MODULES` matches between your headers' `.config` and the running kernel, or `struct module` comes out the wrong size.

Not asking to reopen anything, just leaving the pointer: https://github.com/slckgms/h616-sv6256p-wifi-ethernet