warpme/miniarch #12: H313 , Only 2GB seen by system

mercs759, 2023-05-05:
Anwol Q86x (H313) has ram 8 chips , they are marked SEC XXX hyh9 k4b2g0446c , where XXX changes
They should be by Samsung.
According to Samsung docs they are
DDR3L-1333(667mhz) SDRAM sodimm
8 of these should be 4GB total. 

Please verify XD
from https://www.datasheets360.com/pdf/2129653661416128875 , page 5:

Samsung 2Gb DDR3L C-die
[ Table 2 ] 
2Gb DDR3 C-die Speed bins
Organization -DDR3L-1066 (7-7-7) ---DDR3L-1333 (9-9-9)3 -----
512Mx4 ------K4B2G0446C-HYF8 ------**K4B2G0446C-HYH9**

but my system says
[root@alarm sys]# cat /proc/meminfo
MemTotal:        2030724 kB

possible causes : 
1) one or more memory chips are "weak" , they can't work at their nominal frequency/timing,  bringing down the whole 4 chips bank (2gb) . (Can I check the working frequency of my ram?)

2) fake Samsung chips (Can I check how many ram modules does the system see?)

3) ram limit somewhere , for compatibility/stability reasons 
(according to armbian forums something similar happened with Alwinner H6, which had problems with more than 3gb ram. 
H6 and H616/H313 are very similar, but I didn't read anywhere that H313 has the same 3gb problem. Some armbian user found a workaround for this, I will see)

warpme, 2023-05-07:
Well,
As Anwol Q86x (H313) has 8 ram chips and uses 512Mx4 chips.
It means you have 2GB device...
(512Mx4 chip is 256MB chip. 8x256MB chips seems 2GB to me...)

mercs759, 2023-05-07:
[image]

From this table I understand that I can get 2Gb with 4 **446** chips, or with 8 **846**.
mine are 446... 
(i'm confused)

warpme, 2023-05-07:
Well. Both in above table are 2Gbit chips (pls see table title). 
8 chips, 2Gbit each giving 16GBit total.
As each byte is 8 bits, 16GBit=2Gbytes
Hope this help you!

ps: false labeling (2GB box sold you as 4GB) is almost normal practice in this market. 
You know - it is bussines style by Chin***...

mercs759, 2023-05-07:
ok, I was misleaded by "cpu-z" and "device info HW" under stock's android ... both of them reported 4GB, but probably just took the info from TV box  producer's "BIOS". 

i'll try do play with ramdisks under android and see what happens. 🤣

mercs759, 2023-05-08:
You are right.  
Each "hyh9 k4b2g0446c" chip is 2Gbits ,or "512Mx4" (M= Mbits) ... 8 chips are 16Gbits = 2GBytes. 
Problem solved.
My fault. XD

warpme, 2023-05-08:
Thx.
So lets close this ticket :-)

closed 2023-05-08 by warpme