warpme/miniarch #13: FUSE Module Missing? Can't mount NTFS!

adyatan, 2023-05-06:
I tried mounting external HDD. First installed ntfs-3g, then used the mount command which gave the following error.
`modprobe: FATAL: Module fuse not found in directory /lib/modules/6.3.1`
`fuse: device not found, try 'modprobe fuse' first`

Doing `modprobe fuse` also gave the same message.
`modprobe: FATAL: Module fuse not found in directory /lib/modules/6.3.1`

mercs759, 2023-05-06:
if you just have to mount an hard disk, I think doing it as root could help.

adyatan, 2023-05-06:
Same thing with root.

mercs759, 2023-05-07:
from arch Linux wiki, May 2023
voice "NTFS"

"Note:
All officially supported kernels with versions 5.15 or newer are built with CONFIG_NTFS3_FS=m and thus support it. Before 5.15, NTFS read and write support is provided by the NTFS-3G FUSE file system. Or you can use backported NTFS3 via ntfs3-dkmsAUR.
Paragon Software, the author of the kernel module, has not yet released userspace utilities for NTFS3. You can use NTFS-3G userspace utilities without NTFS-3G driver via ntfsprogs-ntfs3AUR"

adyatan, 2023-05-07:
@mercs759 thanks! How couldn't I see that?
Using ntfs3 instead of ntfs-3g successfully mounted the HDD.

closed 2023-05-07 by adyatan