warpme/miniarch #14: Cannot enable ufw. Modules missing once again?

adyatan, 2023-05-08:
Installed ufw through pacman. Running `sudo ufw enable` throws the following.
```
Command may disrupt existing ssh connections. Proceed with operation (y|n)? y
ERROR: problem running ufw-init
Warning: Extension conntrack revision 0 not supported, missing kernel module?
Warning: Extension conntrack is not supported, missing kernel module?
iptables-restore: line 4 failed
Warning: Extension conntrack revision 0 not supported, missing kernel module?
Warning: Extension conntrack is not supported, missing kernel module?
Warning: Extension conntrack is not supported, missing kernel module?
Warning: Extension conntrack is not supported, missing kernel module?
Warning: Extension conntrack is not supported, missing kernel module?
Warning: Extension addrtype revision 0 not supported, missing kernel module?
Warning: Extension addrtype is not supported, missing kernel module?
Warning: Extension addrtype is not supported, missing kernel module?
Warning: Extension limit revision 0 not supported, missing kernel module?
iptables-restore: line 75 failed
Warning: Extension addrtype revision 0 not supported, missing kernel module?
iptables-restore: line 30 failed
Warning: Extension LOG revision 0 not supported, missing kernel module?
Warning: Extension limit revision 0 not supported, missing kernel module?
Warning: Extension conntrack revision 0 not supported, missing kernel module?
iptables-restore: line 36 failed

Problem running '/etc/ufw/before.rules'
Problem running '/etc/ufw/after.rules'
Problem running '/etc/ufw/user.rules'
```

Afterwards can't connect via SSH. Needs a restart. SSH is already allowed in ufw rules.

adyatan, 2023-05-08:
Hi @warpme, it says 
```
:: Synchronizing package databases...
 miniarch is up to date
:: Starting full system upgrade...
 there is nothing to do
```

warpme, 2023-05-08:
ah - forgot bump pkg rel.
maybe now will be better?

adyatan, 2023-05-08:
Firewall is active and enabled on system startup.
Thanks @warpme. I will report if there are some other issues.

warpme, 2023-05-08:
thx!
closing as solved by https://github.com/warpme/minimyth2/commit/e9dddf5e4556503a67b2800e36b76dfabade3f71

closed 2023-05-08 by warpme