warpme/miniarch #16: Andowl Q96x, (AIP1618 display driver) no signs of life while ON

mercs759, 2023-05-15:
system described in previous issue,

One can't tell if the box is on by looking at it. 
Box is completely dark, no lights at all. 

I already made a mess once (unplugged AC  while miniarch was running)

My system has 2 possible light sources : VFD or Optical SPDIF

Enabling one of them (even if only partially, even if Spdif audio not working ) could help a lot.

Some references for AIP1628 (VFD driver-chip)

AIP1628 clones : TM1628 | CS1628 | HT1628| LK1628 | SM1628

https://github.com/LibreELEC/linux_openvfd

https://lwn.net/Articles/886379/

warpme, 2023-05-15:
Well,

No need to compile vfd kernel driver as miniarch already has it: https://github.com/warpme/minimyth2/blob/master/script/kernel/linux-6.3/files/0105-drivers-led-add-openvfd-1.4.2.patch

What you need is user space daemon.
To test yours led display you may use vfd user space script i'm including in MiniMyth2: https://raw.githubusercontent.com/warpme/minimyth2/master/script/utils/openvfd-service/files/openvfd and attached daemon:
[OpenVFDService.zip](https://github.com/warpme/miniarch/files/11480418/OpenVFDService.zip)

To test vfd device tree config for your device - you may try MiniMyth2 image. If led display appears then you have correct vfd config in yours box device tree - so no any custom vfd.conf is required.

If led display will not work with MiniMyth2 image - you need to develop vfd.conf for your device. 
For this pls look at vfd documentation & exemplary conf files...   

good luck

btw: i understand this discussion is not about miniarch bug....

mercs759, 2023-05-16:
Well , yes, it doesen't seem a bug. Maybe U can close the issue.
 
Ok, you wrote that kernel already has [0105-drivers-led-add-openvfd-1.4.2.patch] in it.
So , i tried the script

[root@alarm ~]# ./openvfd
starting OpenVFDService daemon
OpenVFDService daemon started. Exiting with rc=0
[root@alarm ~]# Open device failed.
: No such file or directory

I don't know what's missing. 
But in my device tree file (generated via dtc and filesystem) I can't see any reference to either "openvfd" or "vfd" or similar. 
The only node which could have something to do with leds is 
 leds {
                compatible = "gpio-leds";

                led-red {
                        color = <0x01>;
                        function = "disk-activity";
                        gpios = <0x15 0x07 0x06 0x00>;
                        linux,default-trigger = "mmc0";

Should I add, search for something? 

I also tried to create a vdf.conf file , 
i started from here :
https://github.com/arthur-liberman/vfd-configurations/tree/master
but there is nothing for H313/H616 SoCs. Just one .conf for H6.

I modified that to include my vfd-controller etc, but nothing
Is there some other "repository" for .conf?

mercs759, 2023-05-20:
I think I'm using the correct values, 
they come from my working android device tree and and /sys/kernel/debug/gpio : 

Only difference is that in miniArch gpiochip1 becomes gpiochip0

so,
clk 0,230 , dat=0,231 and stb=0,233 ; display type 03 , display controller  0 (FD628 compatible)

[root@alarm alarm]# modprobe -r openvfd
[root@alarm alarm]# /sbin/modprobe openvfd vfd_gpio_clk='0,230,0' vfd_gpio_dat='0,231,0' vfd_gpio_stb='0,233,0' vfd_gpio_chip_name='300b000.pinctrl' vfd_chars='2,4,3,2,1' vfd_dot_bits='0,1,2,3,4,5,6' vfd_display_type='0x03,0x00,0x00,0x00'

But I have no way of understanding if openvfd.ko is taking those values or not.
gpioinfo says nothing changes on pins 230,231,233. 
They remain "unused, input, active-high"
(Is there some way for tracing modprobe?)

and when I give 

[root@alarm alarm]# /usr/bin/OpenVFDService ${clock_12h_format:+-12h}
Open device failed.
: No such file or directory

What is OpenVFDService looking for?

mercs759, 2023-05-22:
now I can see errors on journalctl.  
I can't believe I didn't see those before.

I don't know if adding openvfd node to device tree helped, but that was one of the last things I did.

Now service works.

closed 2023-05-22 by mercs759