warpme/miniarch #18: Tanix Tx6s can not connect network

rayzpham, 2023-06-27:
Currently, I use new version of [MiniArch-20230212-6.3.5-board-h616.tanix_tx6s_axp313-SD-Image.img.xz](https://github.com/warpme/miniarch/releases/download/20230531-6.3.5-g66a40f5ba/MiniArch-20230212-6.3.5-board-h616.tanix_tx6s_axp313-SD-Image.img.xz) and I can not use network by **cable smartphone**, ethernet and wifi but before that I use [MiniArch-20230212-6.2.10-board-h616.tanix_tx6s_axp313-SD-Image.img.xz](https://github.com/warpme/miniarch/releases/download/20230226-6.2-6b34b2ce/MiniArch-20230212-6.2.10-board-h616.tanix_tx6s_axp313-SD-Image.img.xz) (Link removed ) and I still can use **cable smartphone** to connect network
Anybody help me create file backup of img MiniArch-20230212-6.3.5-board-h616.tanix_tx6s_axp313-SD-Image.img.xz. Thanks so much :((

Sorry, my English may be bad for writing.

Null-Kelvin, 2023-07-15:
Same here. Board model - CS_H616_TX6S_B4_V4.1

```
[root@alarm alarm]# rmmod dwmac-sun8i
[root@alarm alarm]# dmesg --clear
[root@alarm alarm]# modprobe dwmac-sun8i
[root@alarm alarm]# dmesg
[ 1203.974373] sun50i-h616-pinctrl 300b000.pinctrl: supply vcc-pa not found, using dummy regulator
[ 1203.983722] dwmac-sun8i 5030000.ethernet: IRQ eth_wake_irq not found
[ 1203.983753] dwmac-sun8i 5030000.ethernet: IRQ eth_lpi not found
[ 1203.984819] dwmac-sun8i 5030000.ethernet: supply phy-io not found, using dummy regulator
[ 1203.993590] dwmac-sun8i 5030000.ethernet: PTP uses main clock
[ 1203.994076] dwmac-sun8i 5030000.ethernet: No HW DMA feature register supported
[ 1203.994104] dwmac-sun8i 5030000.ethernet: RX Checksum Offload Engine supported
[ 1203.994112] dwmac-sun8i 5030000.ethernet: COE Type 2
[ 1203.994123] dwmac-sun8i 5030000.ethernet: TX Checksum insertion supported
[ 1203.994131] dwmac-sun8i 5030000.ethernet: Normal descriptors
[ 1203.994138] dwmac-sun8i 5030000.ethernet: Chain mode enabled
[ 1204.050329] dwmac-sun8i 5030000.ethernet end0: renamed from eth0
[ 1204.114890] dwmac-sun8i 5030000.ethernet: EMAC reset timeout
[ 1204.120830] dwmac-sun8i 5030000.ethernet end0: stmmac_dvr_remove: removing driver
[ 1204.163959] dwmac-sun8i: probe of 5030000.ethernet failed with error -110
```

warpme, 2023-07-16:
Hi,
It looks you have different hardware in your's tx6s :-(

To move forward may you:
1.provide me pictures of pcb from your box
2.extract android dt from box factory android ( i.e. using this procedure: https://github.com/warpme/minimyth2/wiki/Getting-device-tree-from-TVbox-vendor-Android)

Null-Kelvin, 2023-07-16:
It's seems so. Sunxi metioned CS_H616_TX6S_B4_V3.0 as reference board, and here is 4.1
[image]
[dtb and dts.zip](https://github.com/warpme/miniarch/files/12064023/dtb.and.dts.zip)

warpme, 2023-07-16:
Fast Eth on h616 is tricky thing (as it uses companion chip ac200 or ac300) and currently is not supported by linux kernel.
To get it working i developed u-boot code which auto-detects acX00 variant and initiates it. Then hw is passed in warm state to linux kernel.

To see is this process working ok in your case we need uart output from uboot.
(more info: https://linux-sunxi.org/Tanix_TX6s)

Without this info I can't say where is root cause of dead eth in your case....

Null-Kelvin, 2023-07-16:
[UART Log.txt](https://github.com/warpme/miniarch/files/12064310/UART.Log.txt)

warpme, 2023-07-16:
@Null-Kelvin
I prepared test build with some changes in uboot: https://github.com/warpme/minimyth2/releases/download/v12.15.2/MiniMyth2-armv8-master-12.16.0.r378-board-h616.tanix_tx6s_axp313-SD-Image.img.xz

may you pls flash and provide me uart output from this build?

Null-Kelvin, 2023-07-16:
[UART Log 2.txt](https://github.com/warpme/miniarch/files/12064435/UART.Log.2.txt)

warpme, 2023-07-16:
thx. in this build i changed mdio (eth phy) addr from 1 to 0.
I see:
`Could not get PHY for ethernet@5030000: addr 0 `
It looks still nogo.

I prepared another build with mdio @ addr16
May you download again, flash and provide uart out from uboot?

(all those plays are because android dt done have mdio entry at all so we trying to guess addr)

Null-Kelvin, 2023-07-16:
Same link as before?
Should it be recompile every time, or I can just replace one file on memory card instead of full reflashing?

warpme, 2023-07-16:
yes. the same link. i re-uploaded image
with this image uart out should refer to `addr 16` instead of [addr 0](url) in
`Could not get PHY for ethernet@5030000: addr 0 `

Null-Kelvin, 2023-07-16:
[UART Log 3.txt](https://github.com/warpme/miniarch/files/12064542/UART.Log.3.txt)

warpme, 2023-07-16:
this looks much better :-)
May you on sd card overwrite boot.scr file by boot-debug.scr  (so boot-debug.scr becomes boot.scr)?

Null-Kelvin, 2023-07-16:
[UART Log 4.txt](https://github.com/warpme/miniarch/files/12064598/UART.Log.4.txt)
It works, received IP address from DHCP and boot MiniMyth. 
[DMESG.txt](https://github.com/warpme/miniarch/files/12064601/DMESG.txt)

warpme, 2023-07-16:
https://github.com/warpme/miniarch/releases/download/20230531-6.3.5-g66a40f5ba/MiniArch-20230212-6.4.3-board-h616.tanix_tx6s_axp313-SD-Image.img.xz

Null-Kelvin, 2023-07-16:
Works perfectly! Thank you very much!

closed 2023-07-16 by warpme

rayzpham, 2024-02-29:
Hi @Null-Kelvin 
I need your help to back up a file named 'MiniArch-20230212-6.4.3-board-h616.tanix_tx6s_axp313-SD-Image.img.xz' because it can be used with Ethernet. If you can assist me, please send the file to [rayzpham@gmail.com](mailto:rayzpham@gmail.com). Thank you for your support.