warpme/miniarch #19: tanix tx6 install or migrate to emmc

SAM6o9, 2023-07-14:
Hi mate , 
First of all thank you for this awesome project 🙏 I just found this and it works pretty ok on my tx6 , but I want to know if there is a way to install or migrate it from sd to internal emmc ? maybe a script like what armbian uses to install to sata-nand-emmc storages ?? 🤔
thanks .

warpme, 2023-07-14:
Thx for (good) feedback :-)
I'm happy you found it useful.

re: migration to emmc - honestly i'm fully loaded with work duties and have not much time to look how to develop emmc migration in convenient way....
Maybe you can look how armbian to doing this and report here findings?

warpme, 2023-09-15:
Closed as in project plans there is no plans for emmc boot support.

If required - any user contribution to provide such feature is very welcomed!

closed 2023-09-15 by warpme