warpme/miniarch #20: RTL8211F on X96 x6 not working properly

yannick, 2023-07-15:
Hi! 

the ethernet controller on the x96 doesnt really work. interestingly enough he gets a ipv6 address that doesnt work. very weird.
works fine on android.

any hints?

warpme, 2023-07-16:
Hi,
This type of subtle issues with Eth usually are meaning yours x96-x6 has pcb with different traces layout than mine (and to which DT i'm providing).

Different traces layout means different propagation delays and these delays requiring different delay compensation setup at phy init (example for my x96-x6 box :https://github.com/warpme/minimyth2/blob/82de74168df68f4ba9205803138f05a5838b80ef/script/kernel/linux-6.4/files/0836-arm64-dts-rockchip-add-dts-for-x96-x6.patch#L256)

All this is result x96 company approach to hw standardization: there is no such thing like standard hw set (used chips, pcb ver., etc) for box called "x96-x6"

From support perspective: dealing with this is pure nightmare...

To move forward may you:
1.provide me pictures of pcb from your box
2.extract android dt from box factory android  ( i.e. using this procedure: https://github.com/warpme/minimyth2/wiki/Getting-device-tree-from-TVbox-vendor-Android)

yannick, 2023-07-16:
Hi Piotr, 
thanks, those are very helpful hints.
here is the dts: https://gist.github.com/yannick/c97a09fc05eba44fa50181fee5709d0a 

pic to follow. 

do you have further hints on how i can use this to build my own dtb? 
are dtb's kernel specific or could i just copy the one from android?

further, is it possible to install alarm onto the mmc and if yet is there a way back to flash the original firmware. ie. dont really know where the debug usb port is to see uboot.

yannick, 2023-07-16:
[image]
[image]

sidenote: have you seen anyone getting the LCD via the AiP1628 chip to work?

warpme, 2023-07-16:
Yannick,
In your's dts is see (for gmac1)
```
tx_delay = <0x4b>;
rx_delay = <0x2b>;
```
in miniarch dt i'm using
```
tx_delay = <0x4f>;
rx_delay = <0x2d>;
```
Diff seems to be minimal - but worth to try...
May you try with miniarch dt (attached) - but with tx/rx delay values from yours dt?
(dt is kern.ver specific - so you need to compile for your's kern.ver.) 
[rk3566-x96_x6.dts.zip](https://github.com/warpme/miniarch/files/12063892/rk3566-x96_x6.dts.zip)

yannick, 2023-07-16:
@warpme awesome, thank you! how can i compile this for my kernel? (sorry i'm new to this)
```
dtc -I dtb -O dts -o out.dtb rk3566-x96_x6.dts
FATAL ERROR: Blob has incorrect magic number
```

warpme, 2023-07-18:
@yannick 
I uploaded update miniarch with current kernel: https://github.com/warpme/miniarch/releases/download/20230531-6.3.5-g66a40f5ba/MiniArch-20230718-6.4.3-board-rk3566.x96_x6-SD-Image.img.xz
Lets try this. If Eth will be still non-working - i will provide you x96-x6 dtb with tx/rx delays taken from your's box....

yannick, 2023-07-18:
@warpme thank you so much!  unfortunately same result, i get a non working ipv6 address and thats it. 

do you have scripts available so i could do my own image builds and try them? i unfortunately failed to compile my own kernel with the DT using makepkg (i made a kernel in the end but it would not boot)
i'd be interested to learn how to do this

warpme, 2023-07-18:
@yannick
Whole process is almost the same like for minimyth2 https://github.com/warpme/minimyth2/wiki/Build-Instructions
Only diff is that you need to go to `miniarch` in `cd ~/minimyth2/script/meta/minimyth/ `not to `minimyth` then simple run `prepare-miniarch-image.sh` and select your board
(be prepared for long first run - as described in wiki - as minimyth2 builds ALL bits (even cross-compiler)    

I'm attaching dtb for your box with tx/rx delays from yours dts. pls try...
[rk3566-x96_x6.dtb.zip](https://github.com/warpme/miniarch/files/12088582/rk3566-x96_x6.dtb.zip)

yannick, 2023-07-18:
@warpme  z pozdrowieniami z Chin, dtb doesnt work :/   

or is this supposed to work with the above image that you sent, i was using the normal build?
interestingly enough with the latest kernel from your repo ( Linux x96 6.3.12 ) the wifi doesnt show up anymore

```
[yannick@x96 ~]$ md5sum /boot/dtbs/rockchip/rk3566-x96_x6.dtb
fb8a8f433a4ae6daabde64d8370c2538  /boot/dtbs/rockchip/rk3566-x96_x6.dtb
Linux x96 6.3.12 #1 SMP PREEMPT Fri Jul  7 11:14:39 CEST 2023 aarch64 GNU/Linux
[root@x96 yannick]# uname -a
Linux x96 6.3.12 #1 SMP PREEMPT Fri Jul  7 11:14:39 CEST 2023 aarch64 GNU/Linux

[root@x96 yannick]# ethtool -i end0
driver: st_gmac
version: 6.3.12

```

yannick, 2023-07-18:
fwiw, given the dropped packets could be really a timing issue. 
is there a faster way to iterate over those than replacing the DT?
```
end0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet6 fe80::4c7:e9ff:fe73:c2f1  prefixlen 64  scopeid 0x20<link>
        inet6 2a02:168:4a4a:2:4c7:e9ff:fe73:c2f1  prefixlen 64  scopeid 0x0<global>
        inet6 2a02:168:4a4a:2:b8f5:c41c:5500:9409  prefixlen 64  scopeid 0x0<global>
        ether 06:c7:e9:73:c2:f1  txqueuelen 1000  (Ethernet)
        RX packets 3474  bytes 1044766 (1020.2 KiB)
        RX errors 0  dropped 19  overruns 0  frame 0
        TX packets 315  bytes 37252 (36.3 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 32
        ```

warpme, 2023-07-19:
@yannick 
re: issue with wlan: indeed there was missing nvram firmware file. I fixed this in https://github.com/warpme/minimyth2/commit/e625239e4a4fc9c26390f3a851eba1d5bd2564f9 and produced new firmware pkg. You may update your install (`start` then press `4`)  

For eth issue: as issue seems to be on rx, you can play with rx-delay value in dts maybe?
Each time you need to compile dts to dtb - you need this for exact kernel you are using (6.4.3 here)
So maybe you can try setup native kernel pkg building with 6.4.3 kernel and use this just for dtb compiling?

yannick, 2023-07-20:
thank you for the new build! 

>So maybe you can try setup native kernel pkg building with 6.4.3 kernel and use this just for dtb compiling? 
thats what i tried, but i dont really understand how i can build only the dtb and not the full kernel

yannick, 2023-07-20:
so are the PKGBUILDS in https://github.com/warpme/archlinuxarm basically an outdated extraction from whats in the minimyth2 repo?
i was under the impression i could just build that one which did not work.

so whats unclear to me is which kernel source tree i should use/start from. i'm now naivly trying to run make within the `minimyth2/script/kernel/linux-6.4` tree 

is there an easy way to extract working standalone PKGBUILDs that allow native compilation?

warpme, 2023-07-24:
Ah, indeed some explanation  is needed:
archlinuxarm repo is old (and not maintained) repo before i created miniarch...
miniarch uses build system (and build artifacts) from minimyth2
so effectively - to get any miniarch binaries you need build part of minimyth2
build system is clever enough to build _only_ miniarch required things - when you are building only miniarch bins/image

yannick, 2023-07-25:
cool, i assumed that.  its running now in a proxmox VM since a few hours, lets see :)

yannick, 2023-08-03:
so i was successfully able to compile 
```
linux-aarch64-6.4.4-1-any.pkg.tar.gz              linux-aarch64-headers-6.4.4-1-any.pkg.tar.gz  MiniArch-20230718-6.4.4-board-rk3566.x96_x6-SD-Image.img.xz
linux-aarch64-api-headers-6.4.4-1-any.pkg.tar.gz  linux-firmware-20230707-1-any.pkg.tar.gz      miniarch-meta-1.0-2-any.pkg.tar.gz
```
by running `make install` in `/home/minimyth/minimyth2/script/meta/miniarch` 
however when i change dts values in `/home/minimyth/minimyth2/script/kernel/linux-6.4/work/main.d/linux-6.4.4/arch/arm64/boot/dts/rockchip/rk3566-x96_x6.dts`  and run `make update-kernel`  i dont see md5sum of `/home/minimyth/minimyth2/script/miniarch/linux-pkg/work/main.d/pkg/linux-aarch64/boot/dtbs/rockchip/rk3566-x96_x6.dtb` (or any of the other `rk3566-x96_x6.dtb` files) changing which i'd guess it should as this is packed into the arch package?

so 

q1: am i editing the right file? 
q2: how do i upgrade kernel, do i need to do complete rebuild everything after git pull?

yannick, 2023-08-06:
@warpme
some progress: apparently the device tree works and its the initialisation of some of the registers. 
if i boot first into android and then reboot into linux the adapter works just fine and gives me around 950mbit in iperf3 .

fwiw: diff of the registers as seen by ethtool https://gist.github.com/yannick/497ed1f488d4d7b436805350b3cbdfd6 
i just don't really know how to understand/set them

warpme, 2023-09-15:
Closing due lack of HW to develop support for this variant of x96_x6 tv box

closed 2023-09-15 by warpme

TiBismarthome, 2024-07-11:
Please help me, I need debian/armbian on X96 X6, to use it for the Home Assistant server

TiBismarthome, 2024-07-26:
> @yannick I uploaded update miniarch with current kernel: https://github.com/warpme/miniarch/releases/download/20230531-6.3.5-g66a40f5ba/MiniArch-20230718-6.4.3-board-rk3566.x96_x6-SD-Image.img.xz [...]

Currently I really need the dedian 12 or armbian version to install home assistant OS on the x96 x6 rk3566 TV box, I hope you can help me, I have searched everywhere for *.img but it cannot boot.