warpme/miniarch #21: about TV box H313-EMCP-V1.0

jmsvigno, 2023-08-23:
my tv box uses the same chip,Ik316=h313, but whe I use 
MiniArch-20230212-6.3.5-board-h313.x96_q_lpddr3-SD-Image.img.xz  on boot  says this error: 

U-Boot SPL 2021.04 (May 25 2023 - 13:06:08 +0200)
DRAM:This DRAM setup is currently not supported.

 Attached I send the android original  boot log and the images of my tvbox 

[image]
[image]
[bootik316.txt](https://github.com/warpme/miniarch/files/12422956/bootik316.txt)

warpme, 2023-08-24:
Hi,
In boot log i see axp806 pmic....
But can't verify this on picture.
What PMIC your's board uses?
Also - may you get android DT from this box? (instructions howo are in minimyth2 wiki)

jmsvigno, 2023-08-24:
this chip is axp305 accordingly  this picture  above

I don't know  get Dt android , but I'll try

Also - may you get android DT from this box? (instructions howo are in minimyth2 wiki) ( what is the link for me to try?)

warpme, 2023-08-25:
Ah this explains a bit issue.
MiniArch-20230212-6.3.5-board-h313.x96_q_lpddr3-SD-Image.img.xz is for lpddr3 and axp1530
Your box has lpddr3 (albeit with different timmings) - but has axp305 pmic
To have running MiniAarch on your box i need:
1. build different bootloader 
2. update lpddr3 dram timings
For p.2 i need access to real hardware (tvbox you are using)
If anybody provide me hardware - i can work on support...

jmsvigno, 2023-08-25:
> this chip is axp305 accordingly this picture above
> 
> I don't know get Dt android , but I'll try
> 
> Also - may you get android DT from this box? (instructions howo are in minimyth2 wiki) ( what is the link for me to try?)

jmsvigno, 2023-08-25:
> > this chip is axp305 accordingly this picture above
> > I don't know get Dt android , but I'll try
> > Also - may you get android DT from this box? (instructions howo are in minimyth2 wiki) ( what is the link for me to try?)

I was able to extract the dtb files along with the boot.img and bootloader.img. I can't send you a hardware, but I'm sending the files that I dumped. The link is below because I couldn't attach it here.

[https://we.tl/t-xp8WGoP80f](https://we.tl/t-xp8WGoP80f)

jmsvigno, 2023-08-28:
I was able to extract the dtb files along with the boot.img and bootloader.img. I can't send you a hardware, but I'm sending the files that I dumped. The link is below because I couldn't attach it here.

https://we.tl/t-xp8WGoP80f

warpme, 2023-09-15:
Closing due lack of HW to develop support for this variant of h313 tv box

closed 2023-09-15 by warpme

jmsvigno, 2023-10-07:
@warpme 
I managed to get the TVbox to work, I had to edit the u-boot defconfig with the emcp RAM parameters, but now when using miniarch, it gets stuck according to the error below:

 platform regulatory.0: Falling back to sysfs fallback for: regulatory.db
FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
OK [0m] Mounted /boot [0m.
OK [0m] Started Network Configuration
 OK Reached target  Local File Systems [0m.
         Starting Rebuild Dynamic Linker Cache [0m...
         Starting Create Volatile Files and Directories [0m...
 How can I fix it?