warpme/miniarch #22: Ethernet interface not available

PBugalho, 2023-09-04:
Hi,

I have a X96Q box and I was able to run the "MiniArch-20230212-6.3.5-board-h313.x96_q_lpddr3-SD-Image.img.xz" whitout any initial problem. It boots correctly but when I check the available network interfaces (ifconfig), I only have the loopback  and the wifi interfaces available. The ethernet interface is not available. 

Any ideia why? Am I missing something?
[image]

Thank you very much.

warpme, 2023-09-05:
Hi,
Without output from uart i can't say where issue is.
Is is possible for you to provide uart output from booting process?

PBugalho, 2023-09-05:
Hi,

Thank you for your help.
Please find attached the android and the miniarch UART boot outputs.

When listing the available interfaces I noticed that the loopback interface is the number one and the wlan0 is the number three. So it should exist a number two in between them that would be the eth0.
[alarm@alarm ~]$ ip link
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
3: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/ether 02:80:e1:0a:96:56 brd ff:ff:ff:ff:ff:ff
[alarm@alarm ~]$

[AndroidBoxBoot.txt](https://github.com/warpme/miniarch/files/12529042/AndroidBoxBoot.txt)
[MiniarchBoot.txt](https://github.com/warpme/miniarch/files/12529043/MiniarchBoot.txt)

Thank you once again.

warpme, 2023-09-11:
Well - it looks your board (v1.3) has mac phy on different address:
Boot log says:
`Could not get PHY for ethernet@5030000: addr 0`

I just verified all boots ok on my x96-q v1.2 box.

Unfortunately android bootloader also not enables Eth:
`[02.057]No ethernet found.`
so i can't deduce what wiring v1.3 pcb has for Eth phy...

If anybody will provide me HW - i can try to find correct addr....

warpme, 2023-09-15:
Closing due lack of HW to develop support for this variant of h313 tv box

closed 2023-09-15 by warpme

kerta1n, 2024-10-10:
Image at the time of writing confirmed working ethernet on boards with the RAM/MMC combo listed (2G-16 instead of HX-JA)