warpme/miniarch #23: Building external kernel module for USB to I2C adapter

mumbozver, 2023-09-08:
Hi,

Thank for your work, miniarch on Vontar H618 semms to work OK.

I want to compile kernel driver for USB to I2C adapter based on ch-341 chip:
https://github.com/frank-zago/ch341-i2c-spi-gpio
So I need only linux kernel headers, not whole source three.
And it looks like pacman offers only `linux-aarch64-headers-6.2.10-1`

Is there any way to obtain linux headers for current kernel? 
`linux alarm 6.4.4 #1 SMP PREEMPT Sat Jul 22 15:43:59 CEST 2023 aarch64 GNU/Linux`

P.S. I want to build on-box because my main x86 bos is Fedora based and I really want to avoid cross-build cross-arch and cross-distros.

mumbozver, 2023-09-14:
I've managed to compile kernel module, after building whole miniarch.

Spotted issue - some supplimentary binaries in source tree are wrong arch:
```
$ file /home/minimyth/kernel-src/usr/lib/modules/6.5.2/build/scripts/basic/fixdep 
/home/minimyth/kernel-src/usr/lib/modules/6.5.2/build/scripts/basic/fixdep: ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 3.2.0, not stripped
```

Workarounded by download binaries from main ARM ArchLinux repo's kernel-headers packages (not ideal and hackish, but worked in my case).

P.S. Found typo in wiki page: 
Step 2 in https://github.com/warpme/miniarch/wiki/Building-MiniArch should probably be `make makesums-all clean`

warpme, 2023-09-15:
Hi,

Thx for finding typo. Corrected :-)
Re fixdep binary arch missmatch: it is expected as when you are cross-compiling kernel, kernel build process uses it at build time (so on x86_64) - not runtime (aarch64)
I think we may close this ticket....

closed 2023-09-15 by warpme