warpme/miniarch #24: Helping with install to eMMC

kerta1n, 2023-10-01:
Hi @warpme,
Insane project you have here. I got two X96Q LPDDR3 boxes booted off of your image, and also tagged you on your [minimyth2 repo](https://github.com/warpme/minimyth2/issues/48), but wanted to know if I could help with figuring out how to install to the internal eMMC storage. I ended up using fdisk on one of the boxes, cloning the disk layout, and getting the H313 to recognize the internal storage as data, HOWEVER, I was unable to migrate the boot partition. Any way I could help here? Would be great to use these as IO-less RPi's (IK they are coming back in stock, but FOSS on a $10 is crazy)

warpme, 2023-10-02:
Hi,

May you pls look at https://linux-sunxi.org/Bootable_eMMC
There is section about install u-boot from Linux....

kerta1n, 2023-10-03:
If I executed those commands, would this allow me to flash mainline U-Boot (with HDMI support)? And if so, what build would I need, or what config file would I need so that I can build it myself? I have a R620 server with 16C and 128GB of DDR3, so if something needs to be compiled, I can do that.

kerta1n, 2023-10-03:
I have access to U-Boot now, but when trying to run `mmc bootbus 2 1 0 0` (mmc device is recognized as 2), it just spits out the help menu. Attached is the output of `version`, `setenv`, and `bdinfo`.

[x96q-h313-uboot-info.txt](https://github.com/warpme/miniarch/files/12797285/x96q-h313-uboot-info.txt)

kerta1n, 2023-10-03:
I see that you've already built a successful U-Boot image, so I'll stop worrying about that for now. I just can't figure out why the commands that linux-sunxi has on their website don't work/how you got the board's U-Boot to execute your U-Boot.

warpme, 2023-10-04:
Oh - commands that linux-sunxi has on their website may not work probably because sunxi page is about older uboot ver?
I'm using mainline uboot 2021.07.
I don't have much time to look on emmc boot on allwinner.
Probably best will be to ask at irc.
pls see https://linux-sunxi.org/Category:Community

Also - when you manage successful boot from emmc - maybe you can:
-write wiki about this
-devel. small script to automate this (i.e. started from sd card booted box which will dd rootfs/boot image + write bootloader)

kerta1n, 2023-10-04:
> Oh - commands that linux-sunxi has on their website may not work probably because sunxi page is about older uboot ver?
> I'm using mainline uboot 2021.07.

Yeah, and are you referring to mainline on the image?

> I don't have much time to look on emmc boot on allwinner.

That's perfectly fine, but I figured since I have devices to mess around with I could do most of the lifting

warpme, 2023-10-04:
> Yeah, and are you referring to mainline on the image?

Yes.
It is 2021.07 mainline + patches:
https://github.com/warpme/minimyth2/tree/master/script/bootloaders/u-boot-h616/files

kerta1n, 2023-10-04:
If I can compile it and install it onto a partition on the eMMC's storage, I will. One more thing though, what partition does your U-Boot recognize the eMMC as?

warpme, 2023-10-04:
On my lpddr3 h313 box i have:
```
[    1.631687] sunxi-mmc 4022000.mmc: initialized, max. request size: 2048 KB, uses new timings mode
[    1.633280] sunxi-mmc 4020000.mmc: initialized, max. request size: 16384 KB, uses new timings mode
[    1.668451] mmc0: host does not support reading read-only switch, assuming write-enable
[    1.670454] mmc0: new high speed SDXC card at address b368
[    1.671749] mmcblk0: mmc0:b368 NCard 58.2 GiB
[    1.675577]  mmcblk0: p1 p2 p3
[    1.706795] mmc2: new DDR MMC card at address 0001
[    1.708045] mmcblk2: mmc2:0001 P1J95K 7.30 GiB
[    1.711944]  mmcblk2: p1 p2 p3 p4 p5 p6 p7 p8 p9 p10 p11 p12 p13 p14 p15 p16 p17
[    1.716431] mmcblk2boot0: mmc2:0001 P1J95K 2.00 MiB
[    1.719355] mmcblk2boot1: mmc2:0001 P1J95K 2.00 MiB
[    1.914516] sunxi-mmc 4021000.mmc: allocated mmc-pwrseq
[    1.937543] sunxi-mmc 4021000.mmc: initialized, max. request size: 16384 KB, uses new timings mode
[    1.953645] mmc3: new high speed SDIO card at address 0001
```
it looks like:
mmc0 is sd card
mmc2 is emmc

kerta1n, 2023-10-04:
Alright, I'll look into this later today

warpme, 2023-10-04:
h313 lpddr3 uboot from my build:
[u-boot-sunxi-with-spl.bin.zip](https://github.com/warpme/miniarch/files/12803003/u-boot-sunxi-with-spl.bin.zip)

kerta1n, 2023-10-05:
Is it possible for me to include HDMI support for U-Boot so I can interface with it over just the monitor and NOT UART?

warpme, 2023-10-05:
Oh - hdmi support in uboot is rather big think (in sense of implementation). In allwinner i would rather not expect support soon...

kerta1n, 2023-10-05:
Alright. I tried just booting U-Boot on an SD card, and it seemed to have flashed it to the Boot0 partition. I rebooted your miniarch image, and am following the steps from linux-sunxi to try and reflash the new U-Boot image to Boot0.

kerta1n, 2023-10-05:
Seems to have crashed it, as there is no output whatsoever from UART. Any ideas?

warpme, 2023-10-05:
hmm - this might be issue with emmc support in uboot.
I think better approach will be to deal with writing to emmc under booted linux - as under linux we have very recent emmc support code (fixes, improvements) while under uboot emmc code is from 2021 (2y old)....

kerta1n, 2023-10-10:
Let me boot the thing back to factory, I'm going to re-try DD'ing the updated U-Boot build once I can get miniarch up and running again

kerta1n, 2023-10-15:
I just got a S905W2 box off of AliX. I'm having the exact same issue that others are having, and its the -110 error.
https://github.com/devmfc/debian-on-amlogic

warpme, 2023-12-08:
Usually boot from eMMC is SoC specific so direct repeating procedure for AMlogic not works for Allwinner.
I don't have plans (and time/energy - as i'm doing this project as hobby in spare time) to work on eMMC booting.
By above - If you mind - i'm closing with status "not planned"

xAbdaziz, 2024-02-16:
@kerta1n Hi, were you able to install MiniArch on X96Q LPDDR3?

I have the same box and I want to install MiniArch too on eMMC

kerta1n, 2024-02-16:
Nope, it's been a while now but there were just too many errors

aldolo69, 2024-03-21:
> Nope, it's been a while now but there were just too many errors

did you succede? i've got a full install now on sd which sometime boot and sometime not. is your boot 100% reliable?

kerta1n, 2024-04-08:
> did you succede? i've got a full install now on sd which sometime boot and sometime not. is your boot 100% reliable?

Nope, couldn't find the internal emmc under /dev

jalbrethsen, 2024-06-03:
I was able to install to emmc with X96Q LPDDR3 without too much difficulty. I first installed via sd card, then booted from that install I did the following. 

1. Use fdisk to delete all the partitions on /dev/mmcblk2 (not sure if this part was necessary). 
2. Download the miniarch image and used the dd command to write it to /dev/mmcblk2. 
3. Mounted the newly created emmc boot partition (/dev/mmcblk2p1) and edit the extlinux.conf and change the root device to /dev/mmcblk2p2. 
4. Shutdown and remove the SD card and it booted fine from emmc.

pdouby, 2024-06-04:
> I was able to install to emmc with X96Q LPDDR3 without too much difficulty. I first installed via sd card, then booted from that install I did the following.
>
> 1. Use fdisk to delete all the [...]

Works like a charm! Thanks!

Proguard007, 2024-06-04:
> I was able to install to emmc with X96Q LPDDR3 without too much difficulty. I first installed via sd card, then booted from that install I did the following.
>
> 1. Use fdisk to delete all the [...]

Having not much experience with linux, this even worked for me! Awesome project here.

kerta1n, 2024-06-05:
> Works like a charm! Thanks!

I will give this a shot, thanks for the post

warpme, 2024-06-05:
@jalbrethsen 
May you pls give me more details (seps, log) when install on emmc by using `install-on-emmc.sh` failed for you?
`install-on-emmc.sh` is in fact doing steps you done (+ some more)

xAbdaziz, 2024-06-05:
> @jalbrethsen May you pls give me more details (steps, log) when install on emmc by using `install-on-emmc.sh` failed for you? `install-on-emmc.sh` is in fact doing steps you done (+ some more)

It failed for me too, in my case it says "Couldn't calculate usage of mmcblk2, Exiting now"
Where can I find logs for the command?

Proguard007, 2024-06-05:
I think I still had to do "sed -e 's/warped.inet2.org/miniarch.mooo.com/g' -i /etc/pacman-miniarch.conf" from another issue before

bonifaido, 2024-06-05:
> I think I still had to do "sed -e 's/warped.inet2.org/miniarch.mooo.com/g' -i /etc/pacman-miniarch.conf" from another issue before

Thanks! This was a life saver for me.

jalbrethsen, 2024-06-06:
> @jalbrethsen May you pls give me more details (steps, log) when install on emmc by using `install-on-emmc.sh` failed for you? `install-on-emmc.sh` is in fact doing steps you done (+ some more)

I just tried this again I get the error "Couldn't calculate usage of mmcblk2p, Exiting now". I appended "| sed 's/p//g' " in line 259 to remove the 'p', after that your script ran fine and installed correctly to emmc.

warpme, 2024-06-06:
May you pls try attached script and provide me log?
[install-on-emmc.zip](https://github.com/user-attachments/files/15608142/install-on-emmc.zip)

warpme, 2024-06-07:
I found miniarch related bug...
You may try this script: 
[install-on-emmc.zip](https://github.com/user-attachments/files/15741590/install-on-emmc.zip)

Blue-Mink, 2024-06-07:
You can refer to this issues.
https://github.com/warpme/miniarch/issues/37

jalbrethsen, 2024-06-08:
> I found miniarch related bug... You may try this script: [install-on-emmc.zip](https://github.com/user-attachments/files/15741590/install-on-emmc.zip)

I just did a fresh install and tried this script to install to emmc. It worked perfectly, thanks!

warpme, 2024-06-08:
thx for testing!

closed 2024-06-08 by warpme