warpme/miniarch #26: recompile u-boot miniarch with dram parametres

jmsvigno, 2023-11-08:
how to recompile u-boot with defconfig dram parameters like: 

CONFIG_DRAM_SUN50I_H616_DX_ODT=0x08080808
CONFIG_DRAM_SUN50I_H616_DX_DRI=0x0e0e0e0e
CONFIG_DRAM_SUN50I_H616_CA_DRI=0x0e0e
CONFIG_DRAM_SUN50I_H616_TPR10=0xf83438
CONFIG_MACH_SUN50I_H616=y
CONFIG_SUNXI_DRAM_LPDDR3=y
CONFIG_SUNXI_DRAM_H616_LPDDR3=y

warpme, 2023-11-08:
Hi,
May you pls give me: 
-model of tvbox using above para
-picture of PCB
-ver of PCB
I want to make sure i don't have already added support for it....

jmsvigno, 2023-11-09:
[image]
[box.dts1.txt](https://github.com/warpme/miniarch/files/13312070/box.dts1.txt)

warpme, 2023-11-10:
oh - i can't read exact type of dram...

jmsvigno, 2023-11-10:
> oh - i can't read exact type of dram...

[image]

jmsvigno, 2023-11-10:
original log of android boot
[148]HELLO! BOOT0 is starting!
[151]BOOT0 commit : 904a4b2
[154]set pll start
[156]periph0 has been enabled
[160]set pll end
[161]unknow PMU
[163]PMU: AXP806
[170]vaild para:8  select dram para2
[174]board init ok
[176]DRAM BOOT DRIVE INFO: V0.60
[179]the chip id is 0x5d00
[182]chip id check OK
[184]DRAM_VCC set to 1200 mv
[189][AUTO DEBUG]32bit,2 ranks training success!
[199]DRAM CLK =720 MHZ
[201]DRAM Type =7 (3:DDR3,4:DDR4,7:LPDDR3,8:LPDDR4)
[209]Actual DRAM SIZE =1024 M
[211]DRAM SIZE =1024 MBytes, para1 = 30ea, para2 = 4001000, dram_tpr13 = 6061
[225]DRAM simple test OK.
[227]rtc standby flag is 0x0, super standby flag is 0x0
[233]dram size =1024
[236]card no is 2
[237]sdcard 2 line count 8
[240][mmc]: mmc driver ver 2019-12-19 10:41
[244][mmc]: set f_max to 50M, set f_max_ddr to 50M
[249][mmc]: mmc 2 bias 4
[257][mmc]: ***Try MMC card 2***
[274][mmc]: MMC 5.0
[276][mmc]: HSDDR52/DDR50 8 bit
[279][mmc]: 50000000 Hz
[281][mmc]: 7296 MB
[283][mmc]: ***SD/MMC 2 init OK!!!***
[345]Loading boot-pkg Succeed(index=0).
[349]Entry_name        = u-boot
[358]Entry_name        = monitor
[362]Entry_name        = dtbo
[365]Entry_name        = dtb
[368]tunning data addr:0x4a0003e8
[372]Jump to second Boot.
NOTICE:  BL3-1: v1.0(debug):cfcc355
NOTICE:  BL3-1: Built : 19:08:45, 2020-11-17
NOTICE:  BL3-1 commit: 8
ERROR:   Error initializing runtime service tspd_fast
NOTICE:  BL3-1: Preparing for EL3 exit to normal world
NOTICE:  BL3-1: Next image address = 0x4a000000
NOTICE:  BL3-1: Next image spsr = 0x1d3

U-Boot 2018.05 (May 14 2021 - 10:23:10 +0800) Allwinner Technology

[00.446]CPU:   Allwinner Family
[00.449]Model: sun50iw9
I2C:   ready
[00.453]DRAM:  1 GiB
[00.456]Relocation Offset is: 35ec5000
[00.494]secure enable bit: 0
[00.496]pmu_axp152_probe pmic_bus_read fail
[00.500]PMU: AXP806
[00.505]CPU=1008 MHz,PLL6=600 Mhz,AHB=200 Mhz, APB1=100Mhz  MBus=400Mhz
[00.692]sunxi overlay merged okqv
[00.695]drv_disp_init
[00.725]__clk_enable: clk is null.
[00.731]drv_disp_init finish
[00.734]gic: sec monitor mode
[00.756]flash init start
[00.758]workmode = 0,storage type = 2
[00.761]MMC:	 2
[00.763]get mem for descripter OK !
[00.772]get sdc2 sdc_boot0_sup_1v8 fail.
[00.776]io is 1.8V
[... 79 lines cut ...]
[    1.008791] sunxi-mmc sdc1: smc 2 p1 err, cmd 5, RTO !!
[    1.014697] sunxi-mmc sdc1: smc 2 p1 err, cmd 5, RTO !!
[    1.034256] failed get gpio-spdif gpio from dts,spdif_gpio:-2
[    1.042710] [sunxi_internal_codec_probe]:get audio avcc failed
[    1.049377] [sunxi_internal_codec_probe]:get audio vcc3v3-audio failed
[    1.057096] [audio-codec]dachpf_cfg configurations missing or invalid.
[    1.064465] lineout_vol:26, linein_gain:3, fmin_gain:3, digital_vol:0, adcdrc_cfg:0, adchpf_cfg:0, dacdrc_cfg:0, dachpf_cfg:0, ramp_func_used:1, pa_msleep_time:160, pa_ctl_level:0, gpio-spk:0
[    1.088515] sndhdmi sndhdmi: ASoC: CPU DAI (null) not registered
[    1.095346] sndhdmi sndhdmi: snd_soc_register_card() failed: -517
[    1.108882] sunxi-ahub-cpudai 5097000.cpudai3-controller: ahub cpudai id invalid
[    1.142277] ERROR: pinctrl_get for HDMI2.0 DDC fail
[    1.153849] tv_probe()1435 - of_property_read_string tv_power failed!
[    1.162134] tv_probe()1435 - of_property_read_string tv_power failed!
[    1.245993] cpu cpu1: opp_list_debug_create_link: Failed to create link
[    1.253527] cpu cpu1: _add_opp_dev: Failed to register opp debugfs (-12)
[    1.261152] cpu cpu2: opp_list_debug_create_link: Failed to create link
[    1.268607] cpu cpu2: _add_opp_dev: Failed to register opp debugfs (-12)
[    1.276244] cpu cpu3: opp_list_debug_create_link: Failed to create link
[    1.283710] cpu cpu3: _add_opp_dev: Failed to register opp debugfs (-12)
[    1.339686] init: [liblp]Logical partition metadata has invalid geometry magic signature.
[    1.349838] init: [liblp]Logical partition metadata has invalid geometry magic signature.
[    1.359538] init: Could not read logical partition metadata from /dev/block/platform/soc/sdc2/by-name/super
[    1.370830] init: Failed to mount required partitions early ...
[    1.408138] init: #00 pc 000a4068  /init (match+18784)
[    1.414006] init: #01 pc 00000daa  <anonymous:0015d000>

jmsvigno, 2023-11-10:
> oh - i can't read exact type of dram...

In the photos above I put the original Android DTS (box.dts1.txt)

warpme, 2023-11-10:
Oh picture shows Emma

jmsvigno, 2023-11-10:
> Oh picture shows Emmc memory you should have 4 or 8 other chips

This type is an emcp memória, ddr+flash memory in The same chip

warpme, 2023-11-11:
here is miniarch uboot with dram params you provided :)
[tanix_tx6s_kingston_lpddr3_defconfig.zip](https://github.com/warpme/miniarch/files/13325946/tanix_tx6s_kingston_lpddr3_defconfig.zip)

jmsvigno, 2023-11-12:
> here is miniarch uboot with dram params you provided :) [tanix_tx6s_kingston_lpddr3_defconfig.zip](https://github.com/warpme/miniarch/files/13325946/tanix_tx6s_kingston_lpddr3_defconfig.zip)

you can recompile with the parameters below, I saw that I made a mistake when passing the parameters below, as this works but not the network card. Another thing you can recompile with support in your other tag: "Fast Eth on h616 is tricky thing (as it uses companion chip ac200 or ac300) and currently is not supported by linux kernel.
To get it working i developed u-boot code which auto-detects acX00 variant and initiates it. Then hw is passed in warm state to linux kernel."

[config.txt](https://github.com/warpme/miniarch/files/13329046/config.txt)

jmsvigno, 2023-11-16:
> here is miniarch uboot with dram params you provided :) [tanix_tx6s_kingston_lpddr3_defconfig.zip](https://github.com/warpme/miniarch/files
/13325946/tanix_tx6s_kingston_lpddr3_defconfig.zip)

@warpme Thank you very much for your patches on u-boot 2021.07 I downloaded minimyth2 and u-boot and applied all your patches to it and recompiled u-boot-sunxi-with-spl.bin, but I still have problems with ethernet it now recognizes it but an error as shown below and ic2 does not recognize

U-Boot SPL 2021.07 (Nov 16 2023 - 10:46:51 -0300)
*Error, wrong i2c adapter 1 max 1 possible
DRAM: 1024 MiB
*Failed to set core voltage!   set CPU 792000000hz frequency
Trying to boot from MMC1
NOTICE:  BL31: v2.8(debug):v2.8
NOTICE:  BL31: Built : 11:38:29, Oct  6 2023
NOTICE:  BL31: Detected Allwinner H616 SoC (1823)
NOTICE:  BL31: Found U-Boot DTB at 0x4a08d958, model: tanix t6s
INFO:    ARM GICv2 driver initialized
INFO:    Configuring SPC Controller
INFO:    PMIC: Probing AXP305 on RSB
WARNING: PMIC: No regulators DT node, skipping setup
INFO:    BL31: Platform setup done
INFO:    BL31: Initializing runtime services
INFO:    BL31: cortex_a53: CPU workaround for 855873 was applied
INFO:    BL31: cortex_a53: CPU workaround for 1530924 was applied
INFO:    PSCI: Suspend is unavailable
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x4a000000
INFO:    SPSR = 0x3c9
INFO:    Changed devicetree.

U-Boot 2021.07 (Nov 16 2023 - 10:46:51 -0300) Allwinner Technology

CPU:   Allwinner H616 (SUN50I)
Model: tanix-t6s
I2C:   ready
DRAM:  1 GiB
MMC:   mmc@4020000: 0, mmc@4022000: 2
Loading Environment from FAT... In:    serial@5000000
Out:   serial@5000000
Err:   serial@5000000
Net:   phy interface6
Could not get PHY for ethernet@5030000: addr 1
No ethernet found.

starting USB...
Bus usb@5101000: USB EHCI 1.00
Bus usb@5101400: USB OHCI 1.0
Bus usb@5310000: USB EHCI 1.00
Bus usb@5310400: USB OHCI 1.0

warpme, 2023-12-08:
Well - I'm afraid without access to real hardware i can't help you...

closed 2023-12-08 by warpme