warpme/miniarch #27: several issues on orange pi zero 3

fizban99, 2023-12-09:
I tried the orange pi zero 3 image on my opizero3 4GB and I got several issues:

1.- First boot seemed to only work through the serial port. I was able to follow the Quick start and at some point I got the login prompt also on my monitor.
2. The screen resolution on the monitor seems to be too high. The font is extremely small. Maybe it is picking up the native resolution of the monitor, which is not the best one for a console...
3. Endeavoros was not working after installing it through option 5 of the menu. I had to install lightdm and enable it, then I installed lightdm-gtk-greeter and xfce4-goodies. I edited /usr/share/xsessions/default.desktop with Exec=/etc/lightdm/Xsession
 and I managed to get the greeter, but after entering the password It did not start anything else. It just showed the greeter again.

fizban99, 2023-12-09:
I managed to install KDE Plasma with sddm as display manager, although I have to install them explicitely with pacman and enable sddm

warpme, 2024-02-12:
Closed due no hardware to work on.

closed 2024-02-12 by warpme