warpme/miniarch #29: Ethernet port on x96q LPDDR3

hcr2020, 2024-01-05:
Hello
I must thank you , your work is the only savior that let me run linux on my android box x96q.
The downside is the ethernet port  isn't detected neither wifi.
I am using an usb2ethernet adapter.
Any chances or guide to add support for built-in network adapter ?
Thank you very much.

aldolo69, 2024-02-01:
> Hello I must thank you , your work is the only savior that let me run linux on my android box x96q. The downside is the ethernet port isn't detected neither wifi. I am using an usb2ethernet adapter. Any chances or guide to add support for built-in network adapter ? Thank you very much.

hi. have you some advise on how to install? used both x96q images with no image on screen. don't know if something happen. only black screen. remote does nothing. only the red led. used balenaetcher to burn the image.

hcr2020, 2024-02-02:
> hi. have you some advise on how to install? used both x96q images with no image on screen. don't know if something happen. only black screen. remote does nothing. only the red led. used balenaetcher [...]

You sould use a tooth pick to press the button inside the AV slot while powering the x96q until you have a terminal in your tv

aldolo69, 2024-02-02:
> You sould use a tooth pick to press the button inside the AV slot while powering the x96q until you have a terminal in your tv

no lucky. maybe my model is different. with the sd card in place it never boot at all. the stick alone without the sd card only provides a couple of seconds of hdmi signal without a cursor and then hang on no-signal.

hcr2020, 2024-02-09:
Thank you very much
I am from a third world country where access to online international payment isn't allowed ( Tunisia) , But Your work definetely is worth any donation.
I thank you very much and if by chance, I travel to another country or my country will allow that, I will definetely donate.

warpme, 2024-02-12:
Closed due no hardware to work on.

closed 2024-02-12 by warpme

SPZel99, 2024-02-22:
Found a solution to use USB tethering on my phone and connecting via USB i can continue installation