warpme/miniarch #30: Won't boot on my X96Q

pdouby, 2024-01-06:
Hello, i have X96Q TV box with H313, 2GB RAM and 16GB ROM. I'am not able to boot any of yours H313/H616 images. Image that should work - MiniArch-20231112-6.6.8-board-h313.x96_q-SD-Image.img.xz - ends with this written to the serial line:

U-Boot SPL 2024.01-rc3 (Dec 14 2023 - 21:40:58 +0100)
DRAM: 64 MiB
Failed to set core voltage!   set CPU 792000000hz frequency

RAM chips seems to be Micron D9PFK:
[image]
Thank you!

pdouby, 2024-01-08:
UPDATE:
I managed to boot MiniArch-20231112-6.6.8-board-h616.tanix_tx6s_axp313-SD-Image.img.xz. After few times pluging-unpluging power adapter, it sometimes starts up. It somehow works, but ethernet is very unstable.

aldolo69, 2024-02-04:
> UPDATE: I managed to boot MiniArch-20231112-6.6.8-board-h616.tanix_tx6s_axp313-SD-Image.img.xz. After few times pluging-unpluging power adapter, it sometimes starts up. It somehow works, but ethernet is very unstable.

hi. did you use the stick-in-avplug to force the boot from sd?

pdouby, 2024-02-04:
> > UPDATE: I managed to boot MiniArch-20231112-6.6.8-board-h616.tanix_tx6s_axp313-SD-Image.img.xz. After few times pluging-unpluging power adapter, it sometimes starts up. It somehow works, but ethernet is very unstable.
> 
> hi. did you use the stick-in-avplug to force the boot from sd?

Hi, nope. I just plug power adapter in. Sometimes it boots on the first try, sometimes not even on the tenth try. I don't know where is the problem...
But when i finally manage to boot it up, everything (probably except remote control) somehow works. Wi-Fi, HDMI sound works well, ethernet is let's say usable after installing LXQt with NetworkManager (still sometimes dropping packets...). Media playback in web browsers are unusable but Kodi playback works fine in 720p or lower resolution.

aldolo69, 2024-02-04:
> > > UPDATE: I managed to boot MiniArch-20231112-6.6.8-board-h616.tanix_tx6s_axp313-SD-Image.img.xz. After few times pluging-unpluging power adapter, it sometimes starts up. It somehow works, but ethernet is very unstable.
> > 
> > 
> > hi. did you use the stick-in-avplug to force the boot from sd?
> 
> Hi, nope. I just plug power adapter in. Sometimes it boots on the first try, sometimes not even on the tenth try. I don't know where is the problem... But when i finally manage to boot it up, everything (probably except remote control) somehow works. Wi-Fi, HDMI sound works well, ethernet is let's say usable after installing LXQt with NetworkManager (still sometimes dropping packets...). Media playback in web browser is unusable but Kodi playback works fine in 720p or lower resolution.

at last i managed to boot it. sometime it boot in a few attempts, sometime more. when it boot it boot fast, and the screen show 4 linux mascottes. cheched a couple of sd card. maybe i will find an sd better then other. the noname sd was a nogo, the samsung is the one i'm testing now. i've go another verbatim. who know... anyway for the record: i used rufus to burn the image on a 16gb card. power led does not change to blue. alwasys red. wired lan works if connected on boot, but not always. all in all not a good situation.

pdouby, 2024-02-04:
Ye, I'am using Samsung EVO 64 GB SDXC and RPi imager for writing the image.  Looks like we facing the same problem with booting. Also power led doesn't change to blue when it finally boots up.

aldolo69, 2024-02-04:
> Ye, I'am using Samsung EVO 64 GB SDXC and RPi imager for writing the image. Looks like we facing the same problem with booting. Also power led doesn't change to blue when it finally boots up. Good point! Looks like it boots in less attempts when button in AV plug is pressed :D

and what about the desktop install? start-5 script always fails because of network problems. now i'm trying the manual approach with "pacman -S xorg-server" but seems a long journey

pdouby, 2024-02-04:
> and what about the desktop install? start-5 script always fails because of network problems. now i'm trying the manual approach with "pacman -S xorg-server" but seems a long journey

Yeah as I wrote before, ethernet is very unstable, sometimes it won't obtain an IP address from network, sometimes it does and connection works fine, sometimes it tends to have huge packet loss... Restarting systemd-networkd service sometimes help... Long journey...
I managed to complete all steps from "Quick start" and then i installed LXQt desktop using this guide:
https://www.debugpoint.com/lxqt-arch-linux-install/

aldolo69, 2024-02-04:
> > > Ye, I'am using Samsung EVO 64 GB SDXC and RPi imager for writing the image. Looks like we facing the same problem with booting. Also power led doesn't change to blue when it finally boots up. Good point! Looks like it boots in less attempts when button in AV plug is pressed :D
> > 
> > 
> > and what about the desktop install? start-5 script always fails because of network problems. now i'm trying the manual approach with "pacman -S xorg-server" but seems a long journey
> 
> Yeah as I wrote before, ethernet is very unstable, sometimes it won't obtain an IP address from network, sometimes it does and connection works fine, sometimes it tends to have huge packet loss... Restarting systemd-networkd service sometimes help... Long journey... I managed to complete all steps from "Quick start" and then i installed LXQt desktop using part 2 of this guide: https://www.debugpoint.com/lxqt-arch-linux-install/

thankyou for the suggestions. i managed to install the desktop. i miss some icons. from the hardware point of view everything works fine with the exception of the initial boot and the unaccelerated display of course. sure there is a trick to get better result on the initial boot

warpme, 2024-02-12:
Closed due no hardware to work on.

closed 2024-02-12 by warpme

aldolo69, 2024-02-12:
> bout the de

in case you are still interested: have found that boot always fails with "sun50i-cpufreq-nvmem: will use speed0 CPU OPPs" in the log. i'm preparing a self-reset with arduino in case of such event, just for fun. i know arduino better then i know linux on arm...