warpme/miniarch #33: Need package tx6s h616s axp313 of version 20230531

rayzpham, 2024-02-08:
Hi
Currently, I use ArchLinux v20231112 but I can not use network and it can not fast run archlinux. Sometimes it accepts sometimes it doesn't. So I hope anybody send me a package of miniarch with version 20230531. I can not download it because the link is not exist. Thanks

Link old package: https://github.com/warpme/miniarch/releases/download/20230531-6.3.5-g66a40f5ba/MiniArch-20230212-6.4.3-board-h616.tanix_tx6s_axp313-SD-Image.img.xz
Link old issues: https://github.com/warpme/miniarch/issues/18

@warpme please help me :(

warpme, 2024-02-08:
Sorry for silence last time - i'm highly occupied by professional life...
To fix Eth issue i need access to real hardware :-(
Last years i founded by semi-lab with 15+ various TVboxes (including x96-q) from my private money and decided to stop investing my private money - effectively just (and only) to increase sells of tv box vendors.
If you want me to work on Eth issue - pls consider donation (on main project page) to help me get hardware to work on. Please note that this will also increase my satisfaction from - currently free of charge - work on minimyth2/miniarch.

warpme, 2024-02-12:
Closed due no hardware to work on.

closed 2024-02-12 by warpme

rayzpham, 2024-02-22:
@warpme I need your help to back up a file named 'MiniArch-20230212-6.4.3-board-h616.tanix_tx6s_axp313-SD-Image.img.xz' because it can be used with Ethernet. Thank you for your support