warpme/miniarch #34: Miniarch.db from warped.inet2.org failing to connect

Luigimash, 2024-03-19:
During installation of Miniarch, when typing command `start` and then (4) to update Miniarch artefacts, I get the following error:
`error: failed retrieving file 'miniarch.db' from warped.inet2.org : Failed to connect to warped.inet2.org port 80 after 10001 ms: Timeout was reached
error: failed to synchronize all databases (download library error)`

This is very similar to the issue here https://github.com/warpme/miniarch/issues/31 , which makes me think your update server is down once again.

Would appreciate any support on this!

warpme, 2024-05-21:
Yes.
inet2.org domain just disappeared :-(
pls use: miniarch.mooo.com

Luigimash, 2024-05-21:
How do I change the url to miniarch.mooo.com during the miniarch installation?

warpme, 2024-05-21:
Well - probably easiest will be:
-before step (4) "update MiniArch artefacts" issue at command prompt: `sed -e 's/warped.inet2.org/miniarch.mooo.com/g' -i /etc/pacman-miniarch.conf`

Luigimash, 2024-05-22:
I'm getting the same timeout error, except from miniarch.mooo.com instead of warped.inet2.org.

warpme, 2024-05-22:
hmm - thats stange.
What happens when you:
-go in web browser to `http://miniarch.mooo.com`
-do `ping miniarch.mooo.com`

Jean-DrEaD, 2024-05-22:
> Well - probably easiest will be: -before step (4) "update MiniArch artefacts" issue at command prompt: `sed -e 's/warped.inet2.org/miniarch.mooo.com/g' -i /etc/pacman-miniarch.conf`

Worked here, thank you so much!
Next challenge, ethernet/wifi, but without priority in that

Luigimash, 2024-05-26:
Works for me now, not sure why it wasn't working before - thanks!

warpme, 2024-05-26:
Probably DNS refresh after adding  mooo.com to my IP :-)

closed 2024-05-26 by warpme