warpme/miniarch #35: X96-Q with AXP313A chip

0x4A4D00, 2024-05-09:
I have an X96Q TV box, but I'm unable to boot either the h313.x96_q-SD-Image.img.xz or the h313.x96_q_lpddr3-SD-Image.img.xz on it. Upon monitoring the serial output, both images become stuck during boot startup while recognizing the DRam. I've managed to boot the h616.tanix_tx6s_axp313-SD-Image.img.xz, but it encounters a crash at:

[1.358424] Loading compiled-in X.509 certificates
[1.400410] gpio gpiochip0: Static allocation of GPIO base is deprecated, use dynamic allocation.
[1.426773] sun50i-h616-pinctrl 300b000.pinctrl: supply vcc-ph not found, using dummy regulator
[1.440861] sun50i-h616-pinctrl 300b000.pinctrl: supply vcc-pg not found, using dummy regulator
[1.465410] usb_phy_generic usb_phy_generic.1.auto: dummy supplies not allowed for exclusive requests
[1.874901] sun50i-cpufreq-nvmem: will use speed0 CPU OPPs

Would it be possible for you to create an image for h313 with axp313 instead of axp305?

0x4A4D00, 2024-05-16:
Hi,
Sorry for my late response, I was too busy with my Uni projects.
Yes sure, here you are:

[image]

[image]

[image]

and also [dts](https://ufile.io/lq6m3sdb)

warpme, 2024-05-16:
thx for pictures!
It will be good also to somehow identify ver. of PCB/box. Are there any stickers / labels on PCB telling something like ver or something?   
Unfortunately i can't get dram params from mmc.bin.
when you run "dd" - was this on device?
I mean e.g.: `dd if=/dev/mmcblk2 of=/mmc.bin bs=1M count=16` not `dd if=/dev/mmcblk2p2 of=/mmc.bin bs=1M count=16`
Or maybe under vendor android, emmc device is not `/dev/mmcblk2` but something different?
On mine x96-b boxes, view of mmc.bin beginning gives something like this:
```
.......................................................................................................................................................................
.......................................................................................................................................................................
.......................................................................................................................................................................
.........U.EFI PART....\...Ρ....................... ...............8o..V&I.h....@....................4.................................................................
.......................................................................................................................................................................
.......................................................................................................................................................................
...........................3D..h..&..FU..fAJt.S..'+.E. ......................b.o.o.t.l.o.a.d.e.r...........................................................3D..h..&..FU
..fAJt.S..'+.F. ......................e.n.v.........................................................................3D..h..&..FU..fAJt.S..'+.G........................b
.o.o.t.......................................................................3D..h..&..FU..fAJt.S..'+.H..........C.............s.u.p.e.r...............................
......................................3D..h..&..FU..fAJt.S..'+.I..C.......D.............m.i.s.c.......................................................................3
D..h..&..FU..fAJt.S..'+.J. D.......E.............r.e.c.o.v.e.r.y...............................................................3D..h..&..FU..fAJt.S..'+.K. E.......Y...
..........c.a.c.h.e.....................................................................3D..h..&..FU..fAJt.S..'+.L. Y.......Y.............v.b.m.e.t.a..................
.................................................3D..h..&..FU..fAJt.S..'+.M..Y.......Z.............v.b.m.e.t.a._.s.y.s.t.e.m...........................................
..........3D..h..&..FU..fAJt.S..'+.N. Z.......Z.............v.b.m.e.t.a._.v.e.n.d.o.r.....................................................3D..h..&..FU..fAJt.S..'+.O..Z
.......[.............m.e.t.a.d.a.t.a...............................................................3D..h..&..FU..fAJt.S..'+.P. [.......[.............p.r.i.v.a.t.e.....
............................................................3D..h..&..FU..fAJt.S..'+.Q..[.......[.............f.r.p....................................................
.....................3D..h..&..FU..fAJt.S..'+.R..[.......\.............e.m.p.t.y.....................................................................3D..h..&..FU..fAJt
.S..'+.S. \.......\.............m.e.d.i.a._.d.a.t.a...........................................................3D..h..&..FU..fAJt.S..'+.T..\.......].............R.e.s.e
.r.v.e.0...............................................................3D..h..&..FU..fAJt.S..'+.U. ].....................U.D.I.S.K.....................................
.......................................................................................................................................................................
.......................................................................................................................................................................
.......................................................................................................................................................................
.......................................................................................................................................................................
```

0x4A4D00, 2024-05-16:
Yes, there is a sticker on it:

[image]

and yes, I had taken dd from wrong Device. Here is the [correct one](https://ufile.io/7y8x22f4).

warpme, 2024-05-16:
perfect,
this one gives nice info:
```
@   0: mbr: DOS MBR
	protective MBR, GPT used
	GPT version 00010000
	usable disk size: 14991 MB
	number of partition entries: 17
@  16: boot0: Allwinner boot0
	size: 65536 bytes
	eGON checksum matches: 0x26465ebd
	DRAM parameters:	A64		H616
		DRAM clock  :       0x258       0x258           -
		DRAM type   :         0x3         0x3           -
		ZQ value    :   0x3030303           -           -
		ODT enabled :   0xe0e0e0e         0x1           -
		DX ODT      :           -   0x3030303           -
		DX DRI      :           -   0xe0e0e0e           -
		CA DRI      :           -      0x1c12           -
		PARA1       :      0x1c12      0x30fb           -
		PARA2       :         0x1   0x8000000           -
		MR0         :      0x30fb      0x1f14           -
		MR1         :   0x8000000         0x4           -
		MR2         :      0x1f14        0x20           -
		MR3         :         0x4           0           -
		TPR0        :        0x20  0xc0000a05           -
		TPR1        :           0  0x80000000           -
		TPR6        :           0  0x33808080           -
		TRP10       :           0    0x2f0006           -
		TRP11       :           0  0xddddcccc           -
		TRP12       :  0xc0000a05  0xeddc7564           -
		TRP13       :  0x80000000      0x6041           -
@ 512: boot0: Allwinner boot0
	size: 65536 bytes
	eGON checksum matches: 0x26465ebd
	DRAM parameters:	A64		H616
		DRAM clock  :       0x258       0x258           -
		DRAM type   :         0x3         0x3           -
		ZQ value    :   0x3030303           -           -
		ODT enabled :   0xe0e0e0e         0x1           -
		DX ODT      :           -   0x3030303           -
		DX DRI      :           -   0xe0e0e0e           -
		CA DRI      :           -      0x1c12           -
		PARA1       :      0x1c12      0x30fb           -
		PARA2       :         0x1   0x8000000           -
		MR0         :      0x30fb      0x1f14           -
		MR1         :   0x8000000         0x4           -
		MR2         :      0x1f14        0x20           -
		MR3         :         0x4           0           -
		TPR0        :        0x20  0xc0000a05           -
		TPR1        :           0  0x80000000           -
		TPR6        :           0  0x33808080           -
		TRP10       :           0    0x2f0006           -
		TRP11       :           0  0xddddcccc           -
		TRP12       :  0xc0000a05  0xeddc7564           -
		TRP13       :  0x80000000      0x6041           -
```

next days i'll prepare test image so pls stay tuned...

warpme, 2024-05-16:
may you pls try: http://warped.inet2.org/MiniMyth2-armv8-master-13.2.0.r130-board-h313.x96_q_v5.1-SD-Image.img.xz ?
This is minimyth2 - i just want to check are bootloader + kernel ok.
I'm doing this blindly (no hw to test) - is in case of any problems - uart log will be crucial....

warpme, 2024-05-17:
Ah - issue with my dyndns provider :-(
pls try with IP: 84.10.100.139

warpme, 2024-05-17:
pls try MiniMyth2-armv8-master-13.3.0.r130-board-h313.x96_q_v5.1-SD-Image.img.xz

0x4A4D00, 2024-05-18:
Sure! Thank you, Here is the [log](https://ufile.io/ppcniskj)

warpme, 2024-05-18:
ah silly me!
on sd card boot partition there is typo in extlinux.conf file:
line `FDT /dtbs/allwinner/sun50i-h313-x96-q-v5.1 `should be `FDT /dtbs/allwinner/sun50i-h313-x96-q-v5.1.dtb`

0x4A4D00, 2024-05-18:
Thank you, I changed that. After seven reboots, it seems to be stuck at:
mv64xxx_i2c 7081400.i2c: can't get pinctrl, bus recovery not supported
[log](https://ufile.io/omtsbe36)

warpme, 2024-05-19:
i rebuild with some changes related to PMIC.
May you pls download MiniMyth2-armv8-master-13.2.0.r130-board-h313.x96_q_v5.1-SD-Image.img.xz
and see how it goes?

0x4A4D00, 2024-05-20:
Thank you, I tested it. Here is the [log](https://ufile.io/4r7pcyxg).

warpme, 2024-05-20:
It looks like granularity of DCDC3 is only 20mV for "higher" voltages (>1.2mV), so we have to specify 1.36V - not 1.35V like i provided you.
May you pls use attached dtb ?
[sun50i-h313-x96-q-v5.1.dtb.zip](https://github.com/warpme/miniarch/files/15377678/sun50i-h313-x96-q-v5.1.dtb.zip)

0x4A4D00, 2024-05-20:
Thank you! It worked perfectly. [(log)](https://ufile.io/iv4dxmou)
I also changed this in [miniarch](https://github.com/warpme/miniarch/releases/download/v20240209/MiniArch-20231112-6.7.4-board-h313.x96_q-SD-Image.img.xz) 
But it's stuck here:
DRAM: 64 MiB

warpme, 2024-05-20:
qll!
I'll provide you miniarch image soon - but first: is Eth working for you?

0x4A4D00, 2024-05-22:
Yes, it seems it's working, It got an IP from my router via LAN Cable.

warpme, 2024-05-22:
pls try: MiniArch-20240320-6.9.1-board-h313.x96_q_v5.1-SD-Image.img.xz

0x4A4D00, 2024-05-23:
Great! Thank you! It's working perfectly.

pdouby, 2024-06-04:
Hello, I really appreciate your work! This new image is booting perfectly!
But I also have the BA 301 V5.1 version, but ethernet does not work for me. When i run dmesg | grep dwmac-sun8i, it shows me messeges like:
dwmac-sun8i 1stmmac_dvr_remove: removing driver
dwmac-sun8i: probe of 5030000.ethernet failed with error -110
Interestingly. when i boot [MiniArch-20240522-6.9.3-board-h616.tanix_tx6s_axp313-SD-Image.img.xz] (it usualy boots like 1 out of 10 attempts) Ethernet seems working.
Is there any way to "transfer" ethernet config/driver from the image I mentioned before to the X96Q V5.1 image? Or is it some thing with device-tree? Sorry, but this is  beyond my Linux knowledge.