warpme/miniarch #37: Don't Install on eMMC

Blue-Mink, 2024-06-01:
`
--- MiniMyth2 to eMMC installer v2.0 (c)Piotr Oniszczuk ---

/usr/local/bin/install-on-emmc: line 98: warning: command substitution: ignored null byte in input
Ooo...you are running tanix,tx6sallwinner,sun50i-h616
Allwinner detected...Good!

SDcard device        : mmcblk0
eMMC device          : mmcblk2
System booted from   : mmcblk0p
ERROR: can't determine usage on mmcblk0p! Exiting `

warpme, 2024-06-06:
May you pls try attached script and provide me log?
[install-on-emmc.zip](https://github.com/user-attachments/files/15608142/install-on-emmc.zip)

Blue-Mink, 2024-06-06:
Yes, thank you very much for your reply and help, I have tried this script and here is the error log
`
---- MiniArch convienience script v1.2, (c)Piotr Oniszczuk ----

Please choose action by pressing key [1..9]

  (1) Full update (ArchLinuxArm codebase)
  (2) Init pacman keyring and populate packages signing keys
  (3) Resize footfs partition
  (4) Update MiniArch artefacts
  (5) Install EndeavourOS
  (0) Install on eMMC

or press Enter to exit...

0
error: could not register 'endeavouros' database (database already registered)
error: could not register 'endeavouros' database (database already registered)
warning: parted-3.6-1 is up to date -- skipping
 there is nothing to do

--- MiniMyth2 to eMMC installer v2.1 (c)Piotr Oniszczuk ---

/usr/local/bin/install-on-emmc: line 126: warning: command substitution: ignored null byte in input
Ooo...you are running tanix,tx6sallwinner,sun50i-h616
Allwinner detected...Good!

SDcard device        : mmcblk0
eMMC device          : mmcblk2
System booted from   : mmcblk0p
ERROR: can't determine usage on mmcblk0p! Exiting ...
You can now poweroff, remove SD card and boot again. System should boot from eMMC ...

Done ... :-)
`

warpme, 2024-06-06:
May you send me /var/log/install-on-emmc.log ?

Blue-Mink, 2024-06-06:
[root@H618 H618]# cat /var/log/install-on-emmc.log
18:36:52:
18:36:52: --- MiniMyth2 to eMMC installer v2.1 (c)Piotr Oniszczuk ---
18:36:52:
18:36:52: platform string: tanix,tx6sallwinner,sun50i-h616
18:36:52: Ooo...you are running tanix,tx6sallwinner,sun50i-h616
18:36:52: Allwinner detected...Good!
18:36:52:
18:36:52: SDcard device        : mmcblk0
18:36:52: eMMC device          : mmcblk2
18:36:52: System booted from   : mmcblk0p
18:36:52: reading /sys/block/mmcblk0p/mmcblk0pp6 for start returns:
18:36:52: reading /sys/block/mmcblk0p/mmcblk0pp5 for start returns:
18:36:52: reading /sys/block/mmcblk0p/mmcblk0pp4 for start returns:
18:36:52: reading /sys/block/mmcblk0p/mmcblk0pp3 for start returns:
18:36:52: reading /sys/block/mmcblk0p/mmcblk0pp2 for start returns:
18:36:52: reading /sys/block/mmcblk0p/mmcblk0pp1 for start returns:
18:36:52: reading /sys/block/mmcblk0p/mmcblk0pp0 for start returns:
18:36:52: reading /sys/block/mmcblk0p/mmcblk0p6 for start returns:
18:36:52: reading /sys/block/mmcblk0p/mmcblk0p5 for start returns:
18:36:52: reading /sys/block/mmcblk0p/mmcblk0p4 for start returns:
18:36:52: reading /sys/block/mmcblk0p/mmcblk0p3 for start returns:
18:36:52: reading /sys/block/mmcblk0p/mmcblk0p2 for start returns:
18:36:52: reading /sys/block/mmcblk0p/mmcblk0p1 for start returns:
18:36:52: reading /sys/block/mmcblk0p/mmcblk0p0 for start returns:
18:36:52: ERROR: can't determine usage on mmcblk0p! Exiting ...

Blue-Mink, 2024-06-06:
I'm so sorry, I think it might be a time zone issue and I didn't get your response and feedback in time.

Blue-Mink, 2024-06-06:
[Uploading resize-rootfs & install-on-emmc log.zip…]()

Blue-Mink, 2024-06-06:
When I type in the command fdisk -l as follows：
Disk /dev/mmcblk0: 58.24 GiB, 62534975488 bytes, 122138624 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x346d1a6a

Device         Boot  Start       End   Sectors   Size Id Type
/dev/mmcblk0p1 *     32768    373553    340786 166.4M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      374784 120913920 120539137  57.5G 83 Linux

Disk /dev/mmcblk2: 116.48 GiB, 125069950976 bytes, 244277248 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: A7EB320D-752F-471A-9686-F97312EE4DC6

Device         Start       End   Sectors   Size Type
/dev/mmcblk2p1  2048 244275199 244273152 116.5G Linux filesystem

Disk /dev/mmcblk2boot0: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes

Disk /dev/mmcblk2boot1: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes

warpme, 2024-06-06:
Definitely issue is in boot dev detection.
May you pls provide me output of `mount` command  ?

Blue-Mink, 2024-06-06:
[root@H618 bin]# df -H -T
Filesystem     Type      Size  Used Avail Use% Mounted on
/dev/root      ext4       60G   14G   44G  25% /
devtmpfs       devtmpfs  2.1G     0  2.1G   0% /dev
tmpfs          tmpfs     2.1G     0  2.1G   0% /dev/shm
tmpfs          tmpfs     824M  975k  823M   1% /run
tmpfs          tmpfs     2.1G  8.2k  2.1G   1% /tmp
tmpfs          tmpfs     412M   74k  412M   1% /run/user/1000
/dev/mmcblk0p1 vfat      175M   82M   94M  47% /run/media/H618/BOOT

Blue-Mink, 2024-06-06:
`[root@H618 bin]# df -H -T
Filesystem     Type      Size  Used Avail Use% Mounted on
/dev/root      ext4       60G   14G   44G  25% /
devtmpfs       devtmpfs  2.1G     0  2.1G   0% /dev
tmpfs          tmpfs     2.1G     0  2.1G   0% /dev/shm
tmpfs          tmpfs     824M  975k  823M   1% /run
tmpfs          tmpfs     2.1G  8.2k  2.1G   1% /tmp
tmpfs          tmpfs     412M   74k  412M   1% /run/user/1000
/dev/mmcblk0p1 vfat      175M   82M   94M  47% /run/media/H618/BOOT`

Blue-Mink, 2024-06-06:
> [root@H618 bin]# df -H -T
Filesystem     Type      Size  Used Avail Use% Mounted on
/dev/root      ext4       60G   14G   44G  25% /
devtmpfs       devtmpfs  2.1G     0  2.1G   0% /dev
tmpfs          tmpfs     2.1G     0  2.1G   0% /dev/shm
tmpfs          tmpfs     824M  975k  823M   1% /run
tmpfs          tmpfs     2.1G  8.2k  2.1G   1% /tmp
tmpfs          tmpfs     412M   74k  412M   1% /run/user/1000
/dev/mmcblk0p1 vfat      175M   82M   94M  47% /run/media/H618/BOOT

Blue-Mink, 2024-06-06:
`[root@H618 bin]# mount
/dev/mmcblk0p2 on / type ext4 (rw,relatime)
devtmpfs on /dev type devtmpfs (rw,relatime,size=1973172k,nr_inodes=493293,mode=755)
proc on /proc type proc (rw,nosuid,nodev,noexec,relatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,nodev,size=803888k,nr_inodes=819200,mode=755)
cgroup2 on /sys/fs/cgroup type cgroup2 (rw,nosuid,nodev,noexec,relatime,nsdelegate,memory_recursiveprot)
pstore on /sys/fs/pstore type pstore (rw,nosuid,nodev,noexec,relatime)
bpf on /sys/fs/bpf type bpf (rw,nosuid,nodev,noexec,relatime,mode=700)
mqueue on /dev/mqueue type mqueue (rw,nosuid,nodev,noexec,relatime)
debugfs on /sys/kernel/debug type debugfs (rw,nosuid,nodev,noexec,relatime)
tracefs on /sys/kernel/tracing type tracefs (rw,nosuid,nodev,noexec,relatime)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,size=2009720k,nr_inodes=1048576)
configfs on /sys/kernel/config type configfs (rw,nosuid,nodev,noexec,relatime)
tmpfs on /run/user/1000 type tmpfs (rw,nosuid,nodev,relatime,size=401940k,nr_inodes=100485,mode=700,uid=1000,gid=1000)
/dev/mmcblk0p1 on /run/media/H618/BOOT type vfat (rw,nosuid,nodev,relatime,uid=1000,gid=1000,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,showexec,utf8,flush,errors=remount-ro,uhelper=udisks2)`

Blue-Mink, 2024-06-06:
/dev/mmcblk0p2 on / type ext4 (rw,relatime)
devtmpfs on /dev type devtmpfs (rw,relatime,size=1973172k,nr_inodes=493293,mode=755)
proc on /proc type proc (rw,nosuid,nodev,noexec,relatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,nodev,size=803888k,nr_inodes=819200,mode=755)
cgroup2 on /sys/fs/cgroup type cgroup2 (rw,nosuid,nodev,noexec,relatime,nsdelegate,memory_recursiveprot)
pstore on /sys/fs/pstore type pstore (rw,nosuid,nodev,noexec,relatime)
bpf on /sys/fs/bpf type bpf (rw,nosuid,nodev,noexec,relatime,mode=700)
mqueue on /dev/mqueue type mqueue (rw,nosuid,nodev,noexec,relatime)
debugfs on /sys/kernel/debug type debugfs (rw,nosuid,nodev,noexec,relatime)
tracefs on /sys/kernel/tracing type tracefs (rw,nosuid,nodev,noexec,relatime)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,size=2009720k,nr_inodes=1048576)
configfs on /sys/kernel/config type configfs (rw,nosuid,nodev,noexec,relatime)
tmpfs on /run/user/1000 type tmpfs (rw,nosuid,nodev,relatime,size=401940k,nr_inodes=100485,mode=700,uid=1000,gid=1000)
/dev/mmcblk0p1 on /run/media/H618/BOOT type vfat (rw,nosuid,nodev,relatime,uid=1000,gid=1000,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,showexec,utf8,flush,errors=remount-ro,uhelper=udisks2)

Blue-Mink, 2024-06-06:
`/dev/mmcblk0p2 on / type ext4 (rw,relatime)
devtmpfs on /dev type devtmpfs (rw,relatime,size=1973172k,nr_inodes=493293,mode=755)
proc on /proc type proc (rw,nosuid,nodev,noexec,relatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,nodev,size=803888k,nr_inodes=819200,mode=755)
cgroup2 on /sys/fs/cgroup type cgroup2 (rw,nosuid,nodev,noexec,relatime,nsdelegate,memory_recursiveprot)
pstore on /sys/fs/pstore type pstore (rw,nosuid,nodev,noexec,relatime)
bpf on /sys/fs/bpf type bpf (rw,nosuid,nodev,noexec,relatime,mode=700)
mqueue on /dev/mqueue type mqueue (rw,nosuid,nodev,noexec,relatime)
debugfs on /sys/kernel/debug type debugfs (rw,nosuid,nodev,noexec,relatime)
tracefs on /sys/kernel/tracing type tracefs (rw,nosuid,nodev,noexec,relatime)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,size=2009720k,nr_inodes=1048576)
configfs on /sys/kernel/config type configfs (rw,nosuid,nodev,noexec,relatime)
tmpfs on /run/user/1000 type tmpfs (rw,nosuid,nodev,relatime,size=401940k,nr_inodes=100485,mode=700,uid=1000,gid=1000)
/dev/mmcblk0p1 on /run/media/H618/BOOT type vfat (rw,nosuid,nodev,relatime,uid=1000,gid=1000,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,showexec,utf8,flush,errors=remount-ro,uhelper=udisks2)`

Blue-Mink, 2024-06-06:
[image]

Blue-Mink, 2024-06-06:
[image]

Blue-Mink, 2024-06-06:
Thank you very much for your patience and help！

warpme, 2024-06-06:
Ah - i was kicked again by lack of time to test code always on my both projects: minimyth2 and miniarch....
pls try attached script   
It's a bit late here...but i hope i found issue....
[install-on-emmc.zip](https://github.com/user-attachments/files/15685718/install-on-emmc.zip)

Blue-Mink, 2024-06-06:
Thank you! You're really looking for my question, I'm trying the latest script right now, I'll let you know the result of the script running a little later.

Blue-Mink, 2024-06-06:
This is the result of the latest script, but I'm not sure if it's useful, can you help me see the result of this runing?
[image]

Blue-Mink, 2024-06-07:
This latest script is very useful and it has solved my problem, thank you again!

Blue-Mink, 2024-06-07:
[image]

Blue-Mink, 2024-06-07:
[image]

Blue-Mink, 2024-06-07:
[image]

warpme, 2024-06-07:
Fixed in https://github.com/warpme/minimyth2/commit/009306943e54743195b5c22cb737a818efb644d1
Thx for help!

closed 2024-06-07 by warpme

Blue-Mink, 2024-06-07:
Miniarch is a great project. Thank you for your work on this project and for your  help. I have successfully installed on my TVbox and written to emmc, it can start and work smoothly, now except the rtl8211fv network can not work , everything else is normal, my wireless network can work, I think arch Linux will bring great help to my study. At present, I don't have more equipment to test minimyth2. Maybe I will test minimyth2 in the future.