warpme/miniarch #38: Ethernet not connected

letsplaysfax, 2024-06-07:
I try to install 
MiniArch-20240522-6.9.3-board-h313.x96_q_v5.1-SD-Image.img.xz
Stb x96q pro h313 
Work 
Ethernet not working 
How to install driver please

Proguard007, 2024-06-07:
Don't know how to install drivers but I used my phone with usb tethering for internet.

warpme, 2024-06-07:
Hi,
I already know at least 5 different HW versions of x96-q.
Following this is simply impossible from maintenance point of view.
Without HW in hands solving this issue is quite/very difficult.
If somebody provide me HW - than i can look on this. 
Blind work without HW is rather waste of time.....

letsplaysfax, 2024-06-07:
> Don't know how to install drivers but I used my phone with usb tethering for internet.

and me i used my phone usb tethering 
i need ethernet rj45 for my project

letsplaysfax, 2024-06-07:
> Hi, I already know at least 5 different HW versions of x96-q. Following this is simply impossible from maintenance point of view. Without HW in hands solving this issue is quite/very difficult. If somebody provide me HW - than i can look on this. Blind work without HW is rather waste of time.....

how to get HW info

letsplaysfax, 2024-06-09:
> > Hi, I already know at least 5 different HW versions of x96-q. Following this is simply impossible from maintenance point of view. Without HW in hands solving this issue is quite/very difficult. If somebody provide me HW - than i can look on this. Blind work without HW is rather waste of time.....
> 
> Thank you I provide id and motherboard ![Messenger_creation_ba28010a-6f35-42a1-8d37-4a3540d956e0](https://private-user-images.githubusercontent.com/161867358/337795691-f9fe5de8-e740-46c3-bd62-d3ac549ad662.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc5NDI3MTgsIm5iZiI6MTcxNzk0MjQxOCwicGF0aCI6Ii8xNjE4NjczNTgvMzM3Nzk1NjkxLWY5ZmU1ZGU4LWU3NDAtNDZjMy1iZDYyLWQzYWM1NDlhZDY2Mi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjA5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYwOVQxNDEzMzhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wNDBlZWE2OGFjNzE4ZDRlM2NhMTgyODBiNWUzZDNlODRjNDdiMGMxZmQyYWJiNmUwMGI1YTRmMWEzNTc4NGU5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.DlBFh9P51nXsiJHZU7_MmcRIvOOl4RvpbbTAAOLjjtY) ![20240607_220154](https://private-user-images.githubusercontent.com/161867358/337795697-e09ccbe0-8157-4e81-8d7a-113aaf143ce3.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc5NDI3MTgsIm5iZiI6MTcxNzk0MjQxOCwicGF0aCI6Ii8xNjE4NjczNTgvMzM3Nzk1Njk3LWUwOWNjYmUwLTgxNTctNGU4MS04ZDdhLTExM2FhZjE0M2NlMy5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjA5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYwOVQxNDEzMzhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wNmI1MDZlN2ExMzI1OTNlNmQyYWM2OGJmYzFjMzQ2NGRmZTRhMGJiNmFmMzQ5MzEwNjMwZjYzMmVkYTQ1MWMyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.1ZKSRfHYSq7Gf2WZOXDD4ANfSAnpdGZy1md3w1d1byQ) ![20240607_220134](https://private-user-images.githubusercontent.com/161867358/337795718-50eafd6e-b164-47d3-a4a1-35e9a619f91d.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc5NDI3MTgsIm5iZiI6MTcxNzk0MjQxOCwicGF0aCI6Ii8xNjE4NjczNTgvMzM3Nzk1NzE4LTUwZWFmZDZlLWIxNjQtNDdkMy1hNGExLTM1ZTlhNjE5ZjkxZC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjA5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYwOVQxNDEzMzhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hNjgzODAwYWZjOGFjNGM1ZjRmZjk4MGI5YzY2NzJjOWI2NWFhMDE3MjBlNDE4NzdkNzM1OWU4OGM4NGRlNDBkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.C1xCoY4NzwBb6Hm1WzOqgHvIoc0kt3q2WTBh331NrPU)

hi this information help you or not. you can find me a solution

letsplaysfax, 2024-06-12:
Haven't you found the solution yet?

warpme, 2024-06-19:
Without HW in hands solving this issue is quite/very difficult. Provide me HW - then i can look on this....

closed 2024-07-26 by warpme