warpme/miniarch #39: Opnvfd on tanix tx6s

sunilsankar, 2024-06-09:
Is there a way to enable the led on miniarch

warpme, 2024-06-09:
Well - miniarch uses the same kernel like minimyth2 so openvfd kernel module should work out-of-box...
Do you have in dmesg any openvfd entries?

sunilsankar, 2024-06-09:
Dont see any module 
```
root@tanix ~]# dmesg |grep -i vfd
[root@tanix ~]#

```

warpme, 2024-06-09:
Right. If you do `modprobe openvfd` then dmesg should show us kernel talks to 7-seg display hw.
To use 7-seg display you need userspace app https://github.com/arthur-liberman/linux_openvfd It talks to kernel.

sunilsankar, 2024-06-10:
Thanks now it works 😄.Is it possible to share the miniarch code base I will include the steps and also the service file

warpme, 2024-06-10:
Well - for miniarch code - i'm already doing this: https://github.com/warpme/minimyth2/tree/master/script/miniarch
User-space of miniarch is in fact ArchLinux ARM - so i think isn't be best when you will start AUR project for user-space OpenVFD?

sunilsankar, 2024-06-10:
> Well - for miniarch code - i'm already doing this: https://github.com/warpme/minimyth2/tree/master/script/miniarch
> User-space of miniarch is in fact ArchLinux ARM - so i think isn't be best when you will start AUR project for user-space OpenVFD? 

Agreed

closed 2024-06-10 by sunilsankar