warpme/miniarch #42: Can't install desktop environment - X96Q

bladyle, 2024-06-15:
Hello, 
I'm glad that there is finally support for this TV box and I give it a try. 
I'm installing on SD card and all good until last step - install https://arm.endeavouros.com/ by typing 'start' then (5)
Trying to follow the instruction but it can't finish successful installation: 

Installing EndeavourOS Base Addons...
./endeavour-ARM-install-V2.7.sh: line 111: eos-packagelist: command not found
error: argument '-' specified with empty stdin

Installing EndeavourOS Base Addons   OK

...

Creating ll alias...
Creating ll alias  OK
./endeavour-ARM-install-V2.7.sh: line 909: eos-rankmirrors: command not found
error: could not register 'endeavouros' database (database already registered)

...

nstalling XFCE4 ...
./endeavour-ARM-install-V2.7.sh: line 653: eos-packagelist: command not found
error: argument '-' specified with empty stdin

...

kipz, 2024-07-04:
@bladyle yeah, I hard that problem too. I was able to install an environment manually using `pacman` directly though.

bladyle, 2024-07-08:
> @bladyle yeah, I hard that problem too. I was able to install an environment manually using `pacman` directly though.

Can you share some quick instruction? I can't see to load to login screen.

kipz, 2024-07-09:
> > @bladyle yeah, I hard that problem too. I was able to install an environment manually using `pacman` directly though.
> 
> Can you share some quick instruction? I can't see to load to login screen.

It's all a bit hacky to be honest, but you you need packages that give you xinit/startx and a window manager (e.g. fluxbox).

I don't recall exactly what I did, but hopefully that's enough to get going.

These are my notes:
https://gist.github.com/kipz/93caf0f38944aa726d48135d7261b605

warpme, 2024-07-26:
As this is not miniarch bug but rather archlinux arm bug - may we close this ticket?

bladyle, 2024-07-27:
@warpme are you able to do successful step (5) - Installing EndeavourOS on this board ?

warpme, 2024-08-16:
Please - investigate this issue with EndeavourOS team.
Also I have frequent issues with install of EndeavourOS - mainly due various issues with EndeavourOS install script :-(

bladyle, 2024-08-19:
Ok, don't worry you can close this issue.

closed 2024-08-19 by warpme