warpme/miniarch #44: The latest builds won't run on my board

AligatorBkmz, 2024-07-02:
[image]
The last image that worked on my board is  [MiniArch-20230212-6.3.5-board-h313.x96_q-SD-Image](https://drive.google.com/file/d/1Y6fsH13QyNql85PRuIs6Y6gY3GTWdTjV/view?usp=drive_link)

AligatorBkmz, 2024-07-02:
however, the package manager breaks due to dependencies when trying to update on old image

Lana-chan, 2024-07-09:
i have the same board, do you still have the old image? the github releases only has latest

warpme, 2024-08-16:
Without physical access to HW I can't do anything here :-(
Im going to close this ticket....

Lana-chan, 2024-08-16:
@warpme would the UART show anything helpful about this? i'm able to connect an FTDI to mine if so, i just haven't tried it (i am not very familiar with low-level linux kernel stuff)

alternatively, could you reupload the previous version which did work?

thanks!

warpme, 2024-08-17:
Well - for sure uart may help to understand what's going on...
re previous versions: unfortunatelly I'm not archiving old versions - so only solution I see is: 
-you will get old sources from my git
-you will build it again

realwk, 2024-09-06:
@Lana-chan Hello. Did you build the image?

warpme, 2024-10-09:
closed due no activity

closed 2024-10-09 by warpme