warpme/miniarch #47: X96Q - won't boot

mi6e4ka, 2024-07-19:
Hello! I try to install MiniArch-20240715-6.10.0-board-h313.x96_q-SD-Image.img on my x96q tv box, with label "BA301_141 V4.0" on board. There uart logs:
```
U-Boot SPL 2024.04 (Jul 15 2024 - 09:49:43 +0200)
DRAM: 2048 MiB
spl: mmc init failed with error: -1
SPL: failed to boot from all boot devices
### ERROR ### Please RESET the board ###
```
I try with two different sd card (16gb and 32gb, sdhc, class 4 and 10), but result is the same.

warpme, 2024-07-24:
well - without having this hw in my lab I can help...
sorry

mi6e4ka, 2024-07-24:
@warpme What could be the problem? I can provide all the necessary information

warpme, 2024-07-26:
...without having this hw in my lab....

closed 2024-07-28 by mi6e4ka