warpme/miniarch #50: Ethernet does not get ip on my h618 box

longbai, 2024-07-24:
Theank you for you work, I can run my h618 4G-128-RTL8211F-Axp313A, but the ethernet doesnot work.
It has obtain mac, but does not get IP,  here is the origin android dtb and the dmesg.
[dmsg.tgz](https://github.com/user-attachments/files/16362256/dmsg.tgz)

Thank you very much.

longbai, 2024-07-24:
It is any method to switch 1000M rgmii? I tried, but the driver init failed.

warpme, 2024-07-24:
Well - it looks like your box has GE on emac0. I see you are using vontar h618 which has 100Mbps Eth on emac1. 
It will not work. You may try with h618 image for boards with GE e.g. zero3....

longbai, 2024-07-25:
Thank you for your reply, the h616 datasheet said there are 2 interface ema0/1. i have trieed emac0, it failed when the phy link brige up.

longbai, 2024-07-26:
maybe fixed by this https://github.com/armbian/build/pull/6617

warpme, 2024-07-26:
If you are using zero2/3/2w images - pls re-download. I updated images with some corrections....

closed 2024-08-16 by warpme