warpme/miniarch #52: Help with headless kodi

jaum20, 2024-08-07:
Hi, I installed theh image `MiniArch-20240715-6.10.0-board-h616.tanix_tx6s-SD-Image.img.xz` and followed the instructions on the README, but didn't install endeavouros, since my objective is using kodi only. I am trying linux cause I can't get 5.1 surround audio using the android os. 

The problem is, after installing kodi and open it, the interface is stuck. Can navigate the UI using both keyboard and mouse. When I fist run the program I get an error message: `[E] pw.loop [loop.c:69 pw_loop_new()] 0xaaab1d36de80: can't make support.system handle: No such file or directory`

Any ideas?

Thanks

warpme, 2024-08-16:
As this is user app issue atop of archlinux arm os - I'm not sure how I can help here?

jaum20, 2024-08-17:
I already give up making kodi work in this build. I am trying minimyth now, but [no luck yet](https://github.com/warpme/minimyth2/issues/62)

closed 2024-08-17 by jaum20