warpme/miniarch #53: [REQUEST] Build instructions

dmig, 2024-08-13:
I have a vontar h618 based board and would like to make some experiments with device tree, to make hardware functioning.

I'm looking for a quick and simple way to build DTB. Tried https://github.com/warpme/minimyth2/wiki/Build-Instructions#quick-build-guide-on-archlinux, but this takes insanely long time (and failed in the middle).

Also, having DTB build HOWTO will attract more enthusiasts -- a lot of them are trying both MiniArch and custom Armbian port and staying with latter because of wider hardware support and documented build process.

dmig, 2024-08-15:
Today complete build fails on:
```
[76/564] Compiling C object glib/libglib-2.0.so.0.7200.3.p/gtimezone.c.o
In function ‘init_zone_from_iana_info’,
    inlined from ‘g_time_zone_new_identifier’ at ../glib/gtimezone.c:1870:11:
../glib/gtimezone.c:776:15: warning: ‘last_explicit_transition_time’ may be used uninitialized [-Wmaybe-uninitialized]
  775 |           if (time_count <= 0
      |               ~~~~~~~~~~~~~~~
  776 |               || last_explicit_transition_time < footer_transition->time)
      |               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../glib/gtimezone.c: In function ‘g_time_zone_new_identifier’:
../glib/gtimezone.c:672:10: note: ‘last_explicit_transition_time’ was declared here
  672 |   gint64 last_explicit_transition_time;
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[98/564] Compiling C object glib/libglib-2.0.so.0.7200.3.p/gvariant.c.o
../glib/gvariant.c: In function ‘g_variant_dict_new’:
../glib/gvariant.c:3947:8: warning: allocation of insufficient size ‘32’ for type ‘GVariantDict’ {aka ‘struct _GVariantDict’} with size ‘128’ [-Walloc-size]
 3947 |   dict = g_slice_alloc (sizeof (struct heap_dict));
      |        ^
[105/564] Linking target glib/libglib-2.0.so.0.7200.3
FAILED: glib/libglib-2.0.so.0.7200.3 
//home/dmig/workspace/minimyth2/images/build/usr/bin/gcc  -o glib/libglib-2.0.so.0.7200.3 glib/libglib-2.0.so.0.7200.3.p/deprecated_gallocator.c.o glib/libglib-2.0.so.0.7200.3.p/deprecated_gcache.c.o glib/libglib-2.0.so.0.7200.3.p/deprecated_gcompletion.c.o glib/libglib-2.0.so.0.7200.3.p/deprecated_grel.c.o glib/libglib-2.0.so.0.7200.3.p/deprecated_gthread-deprecated.c.o glib/libglib-2.0.so.0.7200.3.p/garcbox.c.o glib/libglib-2.0.so.0.7200.3.p/garray.c.o glib/libglib-2.0.so.0.7200.3.p/gasyncqueue.c.o glib/libglib-2.0.so.0.7200.3.p/gatomic.c.o glib/libglib-2.0.so.0.7200.3.p/gbacktrace.c.o glib/libglib-2.0.so.0.7200.3.p/gbase64.c.o glib/libglib-2.0.so.0.7200.3.p/gbitlock.c.o glib/libglib-2.0.so.0.7200.3.p/gbookmarkfile.c.o glib/libglib-2.0.so.0.7200.3.p/gbytes.c.o glib/libglib-2.0.so.0.7200.3.p/gcharset.c.o glib/libglib-2.0.so.0.7200.3.p/gchecksum.c.o glib/libglib-2.0.so.0.7200.3.p/gconvert.c.o glib/libglib-2.0.so.0.7200.3.p/gdataset.c.o glib/libglib-2.0.so.0.7200.3.p/gdate.c.o glib/libglib-2.0.so.0.7200.3.p/gdatetime.c.o glib/libglib-2.0.so.0.7200.3.p/gdir.c.o glib/libglib-2.0.so.0.7200.3.p/genviron.c.o glib/libglib-2.0.so.0.7200.3.p/gerror.c.o glib/libglib-2.0.so.0.7200.3.p/gfileutils.c.o glib/libglib-2.0.so.0.7200.3.p/ggettext.c.o glib/libglib-2.0.so.0.7200.3.p/ghash.c.o glib/libglib-2.0.so.0.7200.3.p/ghmac.c.o glib/libglib-2.0.so.0.7200.3.p/ghook.c.o glib/libglib-2.0.so.0.7200.3.p/ghostutils.c.o glib/libglib-2.0.so.0.7200.3.p/giochannel.c.o glib/libglib-2.0.so.0.7200.3.p/gkeyfile.c.o glib/libglib-2.0.so.0.7200.3.p/glib-init.c.o glib/libglib-2.0.so.0.7200.3.p/glib-private.c.o glib/libglib-2.0.so.0.7200.3.p/glist.c.o glib/libglib-2.0.so.0.7200.3.p/gmain.c.o glib/libglib-2.0.so.0.7200.3.p/gmappedfile.c.o glib/libglib-2.0.so.0.7200.3.p/gmarkup.c.o glib/libglib-2.0.so.0.7200.3.p/gmem.c.o glib/libglib-2.0.so.0.7200.3.p/gmessages.c.o glib/libglib-2.0.so.0.7200.3.p/gnode.c.o glib/libglib-2.0.so.0.7200.3.p/goption.c.o glib/libglib-2.0.so.0.7200.3.p/gpattern.c.o glib/libglib-2.0.so.0.7200.3.p/gpoll.c.o glib/libglib-2.0.so.0.7200.3.p/gprimes.c.o glib/libglib-2.0.so.0.7200.3.p/gqsort.c.o glib/libglib-2.0.so.0.7200.3.p/gquark.c.o glib/libglib-2.0.so.0.7200.3.p/gqueue.c.o glib/libglib-2.0.so.0.7200.3.p/grand.c.o glib/libglib-2.0.so.0.7200.3.p/grcbox.c.o glib/libglib-2.0.so.0.7200.3.p/grefcount.c.o glib/libglib-2.0.so.0.7200.3.p/grefstring.c.o glib/libglib-2.0.so.0.7200.3.p/gregex.c.o glib/libglib-2.0.so.0.7200.3.p/gscanner.c.o glib/libglib-2.0.so.0.7200.3.p/gsequence.c.o glib/libglib-2.0.so.0.7200.3.p/gshell.c.o glib/libglib-2.0.so.0.7200.3.p/gslice.c.o glib/libglib-2.0.so.0.7200.3.p/gslist.c.o glib/libglib-2.0.so.0.7200.3.p/gstdio.c.o glib/libglib-2.0.so.0.7200.3.p/gstrfuncs.c.o glib/libglib-2.0.so.0.7200.3.p/gstring.c.o glib/libglib-2.0.so.0.7200.3.p/gstringchunk.c.o glib/libglib-2.0.so.0.7200.3.p/gstrvbuilder.c.o glib/libglib-2.0.so.0.7200.3.p/gtestutils.c.o glib/libglib-2.0.so.0.7200.3.p/gthread.c.o glib/libglib-2.0.so.0.7200.3.p/gthreadpool.c.o glib/libglib-2.0.so.0.7200.3.p/gtimer.c.o glib/libglib-2.0.so.0.7200.3.p/gtimezone.c.o glib/libglib-2.0.so.0.7200.3.p/gtrace.c.o glib/libglib-2.0.so.0.7200.3.p/gtranslit.c.o glib/libglib-2.0.so.0.7200.3.p/gtrashstack.c.o glib/libglib-2.0.so.0.7200.3.p/gtree.c.o glib/libglib-2.0.so.0.7200.3.p/guniprop.c.o glib/libglib-2.0.so.0.7200.3.p/gutf8.c.o glib/libglib-2.0.so.0.7200.3.p/gunibreak.c.o glib/libglib-2.0.so.0.7200.3.p/gunicollate.c.o glib/libglib-2.0.so.0.7200.3.p/gunidecomp.c.o glib/libglib-2.0.so.0.7200.3.p/guri.c.o glib/libglib-2.0.so.0.7200.3.p/gutils.c.o glib/libglib-2.0.so.0.7200.3.p/guuid.c.o glib/libglib-2.0.so.0.7200.3.p/gvariant.c.o glib/libglib-2.0.so.0.7200.3.p/gvariant-core.c.o glib/libglib-2.0.so.0.7200.3.p/gvariant-parser.c.o glib/libglib-2.0.so.0.7200.3.p/gvariant-serialiser.c.o glib/libglib-2.0.so.0.7200.3.p/gvarianttypeinfo.c.o glib/libglib-2.0.so.0.7200.3.p/gvarianttype.c.o glib/libglib-2.0.so.0.7200.3.p/gversion.c.o glib/libglib-2.0.so.0.7200.3.p/gwakeup.c.o glib/libglib-2.0.so.0.7200.3.p/gprintf.c.o glib/libglib-2.0.so.0.7200.3.p/glib-unix.c.o glib/libglib-2.0.so.0.7200.3.p/gspawn.c.o glib/libglib-2.0.so.0.7200.3.p/giounix.c.o glib/libglib-2.0.so.0.7200.3.p/gthread-posix.c.o -Wl,--as-neede [... cut ...]
make[3]: *** [../../gar.lib.mk:581: ../../system/eudev/cookies/main.d/install] Error 2
make[3]: Leaving directory '/home/dmig/workspace/minimyth2/script/utils/alsa-utils'
make[2]: *** [../../gar.lib.mk:600: imgdep-main] Error 2
make[2]: Leaving directory '/home/dmig/workspace/minimyth2/script/utils/alsa-utils'
make[1]: *** [../../gar.lib.mk:581: ../../utils/alsa-utils/cookies/main.d/install] Error 2
make[1]: Leaving directory '/home/dmig/workspace/minimyth2/script/meta/minimyth'
make: *** [../../gar.lib.mk:600: imgdep-main] Error 2
```

warpme, 2024-08-15:
Hi Dmitry,
May you pls try building current mm2 master?
Generally - i prefer not to fix older versions of mm2 (due lack of time...)

dmig, 2024-08-15:
This was todays master.

Trying to build using `script/meta/miniarch/build-image-for-board.sh 27`, it stops failing to resolve one of patches:
```
 ==> Running checksum on 1150-arm64-dtsi-rockchip-add-3528.dtsi.patch
e04aa139944d0c9bbdb824f9e7d791cd  download/1150-arm64-dtsi-rockchip-add-3528.dtsi.patch
file 1150-arm64-dtsi-rockchip-add-3528.dtsi.patch passes checksum test!
 ==> Running checksum on 1151-arm64-dts-rockchip-add-dts-for-vontar_r3.patch
e1d1d0dfa07562c0df52e4dccfbe058b  download/1151-arm64-dts-rockchip-add-dts-for-vontar_r3.patch
file 1151-arm64-dts-rockchip-add-dts-for-vontar_r3.patch passes checksum test!
 ==> Running checksum on 1152-arm64-dts-rockchip-rk3528-add-dts-for-rock2a.patch
858ae86c201ce951c59d616e03ca9075  download/1152-arm64-dts-rockchip-rk3528-add-dts-for-rock2a.patch
file 1152-arm64-dts-rockchip-rk3528-add-dts-for-rock2a.patch passes checksum test!
	[checksum] complete for linux.
 ==> Extracting download/linux-6.10.4.tar.gz
	[extract] complete for linux.
 ==> Applying patch download/0756-arm64-dts-rockchip-rk33xx-add-eth-wifi-aliases.patch
can't find file to patch at input line 4
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|diff --speed-large-files --no-dereference --minimal -Naur linux-6.10-rc5/arch/arm64/boot/dts/rockchip/rk3399-orangepi-4.dts linux-6.10-rc5/arch/arm64/boot/dts/rockchip/rk3399-orangepi-4.dts
|--- linux-6.10-rc5/arch/arm64/boot/dts/rockchip/rk3399-orangepi-4.dts	2024-06-26 14:44:49.580956438 +0200
|+++ linux-6.10-rc5/arch/arm64/boot/dts/rockchip/rk3399-orangepi-4.dts	2024-06-26 12:49:55.411104868 +0200
--------------------------
File to patch: 
```
Still figuring out, where is that file...

dmig, 2024-08-15:
Ok, 6.10.4 doesn't have `rk3399-orangepi-4.dts`/`rk3399-orangepi-4-lts.dts`, it has `rk3399-orangepi.dts`.
Patch `0756-arm64-dts-rockchip-rk33xx-add-eth-wifi-aliases.patch` fails on `rk3399-orangepi-4.dts` and `rk3399-rock-4c-plus.dts`.
After that whole build process fails:
```
 ==> Running checksum on 1152-arm64-dts-rockchip-rk3528-add-dts-for-rock2a.patch
858ae86c201ce951c59d616e03ca9075  download/1152-arm64-dts-rockchip-rk3528-add-dts-for-rock2a.patch
file 1152-arm64-dts-rockchip-rk3528-add-dts-for-rock2a.patch passes checksum test!
	[checksum] complete for linux.
 ==> Extracting download/linux-6.10.4.tar.gz
	[extract] complete for linux.
 ==> Applying patch download/0756-arm64-dts-rockchip-rk33xx-add-eth-wifi-aliases.patch
can't find file to patch at input line 4
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|diff --speed-large-files --no-dereference --minimal -Naur linux-6.10-rc5/arch/arm64/boot/dts/rockchip/rk3399-orangepi-4.dts linux-6.10-rc5/arch/arm64/boot/dts/rockchip/rk3399-orangepi-4.dts
|--- linux-6.10-rc5/arch/arm64/boot/dts/rockchip/rk3399-orangepi-4.dts	2024-06-26 14:44:49.580956438 +0200
|+++ linux-6.10-rc5/arch/arm64/boot/dts/rockchip/rk3399-orangepi-4.dts	2024-06-26 12:49:55.411104868 +0200
--------------------------
File to patch: linux-6.10.4/arch/arm64/boot/dts/rockchip/rk3399-orangepi.dts
linux-6.10.4/arch/arm64/boot/dts/rockchip/rk3399-orangepi.dts: No such file or directory
Skip this patch? [y] n
File to patch: script/kernel/linux-6.10/work/main.d/linux-6.10.4/arch/arm64/boot/dts/rockchip/rk3399-orangepi.dts
script/kernel/linux-6.10/work/main.d/linux-6.10.4/arch/arm64/boot/dts/rockchip/rk3399-orangepi.dts: No such file or directory
Skip this patch? [y] n
File to patch: /home/dmig/workspace/minimyth2/script/kernel/linux-6.10/work/main.d/linux-6.10.4/arch/arm64/boot/dts/rockchip/rk3399-orangepi.dts
patching file /home/dmig/workspace/minimyth2/script/kernel/linux-6.10/work/main.d/linux-6.10.4/arch/arm64/boot/dts/rockchip/rk3399-orangepi.dts
Hunk #1 FAILED at 25.
1 out of 1 hunk FAILED -- saving rejects to file /home/dmig/workspace/minimyth2/script/kernel/linux-6.10/work/main.d/linux-6.10.4/arch/arm64/boot/dts/rockchip/rk3399-orangepi.dts.rej
can't find file to patch at input line 15
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|diff --speed-large-files --no-dereference --minimal -Naur linux-6.10-rc5/arch/arm64/boot/dts/rockchip/rk3399-orangepi-4-lts.dts linux-6.10-rc5/arch/arm64/boot/dts/rockchip/rk3399-orangepi-4-lts.dts
|--- linux-6.10-rc5/arch/arm64/boot/dts/rockchip/rk3399-orangepi-4-lts.dts	2024-06-26 14:44:49.580956438 +0200
|+++ linux-6.10-rc5/arch/arm64/boot/dts/rockchip/rk3399-orangepi-4-lts.dts	2024-06-26 12:36:13.394455891 +0200
--------------------------
File to patch: /home/dmig/workspace/minimyth2/script/kernel/linux-6.10/work/main.d/linux-6.10.4/arch/arm64/boot/dts/rockchip/rk3399-orangepi.dts
patching file /home/dmig/workspace/minimyth2/script/kernel/linux-6.10/work/main.d/linux-6.10.4/arch/arm64/boot/dts/rockchip/rk3399-orangepi.dts
Hunk #1 FAILED at 25.
1 out of 1 hunk FAILED -- saving rejects to file /home/dmig/workspace/minimyth2/script/kernel/linux-6.10/work/main.d/linux-6.10.4/arch/arm64/boot/dts/rockchip/rk3399-orangepi.dts.rej
patching file arch/arm64/boot/dts/rockchip/rk3399-rock-4c-plus.dts
Hunk #1 FAILED at 16.
1 out of 1 hunk FAILED -- saving rejects to file arch/arm64/boot/dts/rockchip/rk3399-rock-4c-plus.dts.rej
patching file arch/arm64/boot/dts/rockchip/rk3399-rock-4se.dts
Hunk #1 succeeded at 14 with fuzz 2 (offset -2 lines).
patching file arch/arm64/boot/dts/rockchip/rk3399-rock-pi-4b.dts
Hunk #1 succeeded at 14 with fuzz 2 (offset -2 lines).
patching file arch/arm64/boot/dts/rockchip/rk3399-rock-pi-4b-plus.dts
Hunk #1 succeeded at 14 with fuzz 2 (offset -2 lines).
patching file arch/arm64/boot/dts/rockchip/rk3399-rock-pi-4c.dts
Hunk #1 succeeded at 15 with fuzz 2 (offset -2 lines).
make[8]: *** [../../gar.lib.mk:252: normal-patch-0756-arm64-dts-rockchip-rk33xx-add-eth-wifi-aliases.patch] Error 1
make[8]: Leaving directory '/home/dmig/workspace/minimyth2/script/kernel/linux-6.10'
make[7]: *** [../../gar.lib.mk:581: ../../kernel/linux-6.10/cookies/main.d/install] Error 2
make[7]: Leaving directory '/home/dmig/workspace/minimyth2/script/kernel/kernel'
make[6]: *** [../../gar.lib.mk:600: imgdep-main] Error 2
make[6]: Leaving directory '/home/dmig/workspace/minimyth2/script/kernel/kernel'
make[5]: *** [../../gar.lib.mk:581: ../../kernel/kernel/cookies/main.d/install] Error 2
make[5]: Leaving directory '/home/dmig/workspace/minimyth2/script/miniarch/linux-pkg'
make[4]: *** [../../gar.lib.mk:600: imgdep-main] Error 2
make[4]: Leaving directory '/home/dmig/workspace/minimyth2/script/miniarch/linux-pkg'
make[3]: *** [../../gar.lib.mk:581: ../../miniarch/linux-pkg/cookies/main.d/install] Error 2
make[3]: Leaving directory '/home/dmig/workspace/minimyth2/script/miniarch/miniarch-rootfs'
make[2]: *** [../../gar.lib.mk:600: imgdep-main] Error 2
make[2]: Leaving directory '/home/dmig/workspace/minimyth2/script/miniarch/miniarch-rootfs'
make[1]: *** [../../gar.lib.mk:581: ../../miniarch/miniarch-rootfs/cookies/main.d/install] Error 2
make[1]: Leaving directory '/home/dmig/workspace/minimyth2/script/meta/miniarch'
make: *** [../../gar.lib.mk:600: imgdep-main] Error 2
...
```

warpme, 2024-08-15:
Well - I'll add build instructions soon...

Re: your build issues: definitely something not right is with your setup.
I just done test build of fresh miniarch. It went for without single issue.
What I done:
-get vanilla archlinux in kvm.
-do standard `pacman -Syu ` 
-setup build system by following minimyth2 build instructions from https://github.com/warpme/minimyth2/wiki/Build-Instructions
-instead of `cd ~/minimyth2/script/meta/minimyth/` I done `cd ~/minimyth2/script/meta/miniarch/`
-run `./build-image-for-board.sh`
-select desired board then press Enter
-after 2h08m I got sd card image for my h618 zero2w in `~/build`
-flash to sd card. it works ok

please let me know how it goes for you...

dmig, 2024-08-15:
Thanks. I'm running a completely clean build right now.
Removing first 2 hunks from `0756-arm64-dts-rockchip-rk33xx-add-eth-wifi-aliases.patch` should solve the issue (changes are already included).

warpme, 2024-08-15:
hmm - if you need do any changes to get building - then **something is not right with your setup**. Current master should give you working image just with procedure I described 5min ago.

dmig, 2024-08-15:
I'm using kernel 6.10, which resolves to 6.10.4:
```
mm_KERNEL_VERSION         ?= 6.10
```

dmig, 2024-08-16:
Ok, overnight it finished the toolchain and compiled a kernel. Then failed:
```
symlinking sources ...
update kernel ver in PKGBUILD ...
starting packaging with MAKEPKG ...
//home/dmig/workspace/minimyth2/images/build/bin/bash: line 1: //home/dmig/workspace/minimyth2/images/build/usr/bin/makepkg: No such file or directory
make[4]: *** [Makefile:37: build-custom] Error 127
make[4]: Leaving directory '/home/dmig/workspace/minimyth2/script/miniarch/linux-pkg'
make[3]: *** [../../gar.lib.mk:581: ../../miniarch/linux-pkg/cookies/main.d/install] Error 2
make[3]: Leaving directory '/home/dmig/workspace/minimyth2/script/miniarch/miniarch-rootfs'
make[2]: *** [../../gar.lib.mk:600: imgdep-main] Error 2
make[2]: Leaving directory '/home/dmig/workspace/minimyth2/script/miniarch/miniarch-rootfs'
make[1]: *** [../../gar.lib.mk:581: ../../miniarch/miniarch-rootfs/cookies/main.d/install] Error 2
make[1]: Leaving directory '/home/dmig/workspace/minimyth2/script/meta/miniarch'
make: *** [../../gar.lib.mk:600: imgdep-main] Error 2
```
Nevertheless, DTBs are here, and I need to find a way to compile one.

warpme, 2024-08-16:
i added some basic building instructions. pls look....

dmig, 2024-08-16:
Thanks!
5. go to `~/minimyth2/script/miniarch`
6. run `make update-kernel`
fails:
```
> make update-kernel
make: *** No rule to make target 'update-kernel'.  Stop.
> make 
make: *** No targets specified and no makefile found.  Stop.
```
Should probably be `~/minimyth2/script/meta`

dmig, 2024-08-16:
Well, miniarch-rootfs rebuild produced same error:
```
miniarch-rootfs (master) [2]> make clean install
 
cleaning MiniArch-aarch64-20240715-6.10.5 ...
 
[===== NOW BUILDING:	miniarch-rootfs-20240715	=====]
install -d cookies/main.d
	[fetch] complete for miniarch-rootfs.
 ==> Running checksum on ArchLinuxARM-aarch64-latest.tar.gz
bdef3220a954dadacf03f18d18544204  download/ArchLinuxARM-aarch64-latest.tar.gz
file ArchLinuxARM-aarch64-latest.tar.gz passes checksum test!
	[checksum] complete for miniarch-rootfs.
install -d work/main.d
 
unpacking ArchLinuxARM-aarch64-latest ...
./usr/bin/newgidmap: Cannot restore extended attributes: security.capability security.capability: Operation not permitted
./usr/bin/newuidmap: Cannot restore extended attributes: security.capability security.capability: Operation not permitted
bsdtar: Error exit delayed from previous errors.
"adding some fixups in archlinuxarm image to make mkinitcpio happy ..."
	[extract] complete for miniarch-rootfs.
	[patch] complete for miniarch-rootfs.
make[1]: Entering directory '/home/dmig/workspace/minimyth2/script/miniarch/miniarch-rootfs'
make[1]: '../../devel/build-system-bins/cookies/build.d/install' is up to date.
make[1]: '../../miniarch/pacman/cookies/build.d/install' is up to date.
make[1]: '../../miniarch/mkinitcpio/cookies/build.d/install' is up to date.
make[1]: '../../devel/fakeroot/cookies/build.d/install' is up to date.
make[1]: Leaving directory '/home/dmig/workspace/minimyth2/script/miniarch/miniarch-rootfs'
make[1]: Entering directory '/home/dmig/workspace/minimyth2/script/miniarch/miniarch-rootfs'
make[1]: '../../miniarch/miniarch-pkg/cookies/main.d/install' is up to date.
 ==> Building miniarch/linux-pkg as a dependency
make[2]: Entering directory '/home/dmig/workspace/minimyth2/script/miniarch/linux-pkg'
[===== NOW BUILDING:	miniarch-linux.pkg-1.0	=====]
	[fetch] complete for miniarch-linux.pkg.
	[checksum] complete for miniarch-linux.pkg.
	[extract] complete for miniarch-linux.pkg.
	[patch] complete for miniarch-linux.pkg.
	[configure] complete for miniarch-linux.pkg.
symlinking sources ...
update kernel ver in PKGBUILD ...
starting packaging with MAKEPKG ...
//home/dmig/workspace/minimyth2/images/build/bin/bash: line 1: //home/dmig/workspace/minimyth2/images/build/usr/bin/makepkg: No such file or directory
make[2]: *** [Makefile:37: build-custom] Error 127
make[2]: Leaving directory '/home/dmig/workspace/minimyth2/script/miniarch/linux-pkg'
make[1]: *** [../../gar.lib.mk:581: ../../miniarch/linux-pkg/cookies/main.d/install] Error 2
make[1]: Leaving directory '/home/dmig/workspace/minimyth2/script/miniarch/miniarch-rootfs'
make: *** [../../gar.lib.mk:600: imgdep-main] Error 2
```

warpme, 2024-08-16:
> Thanks! 5. go to `~/minimyth2/script/miniarch` 6. run `make update-kernel` fails:
> 
> ```
> > make update-kernel
> make: *** No rule to make target 'update-kernel'.  Stop.
> > make 
> make: *** No targets specified and no makefile found.  Stop.
> ```
> 
> Should probably be `~/minimyth2/script/meta`

well - what is output of `cat ./Makefile `?
make rule called `update-kernel `is there:  https://github.com/warpme/minimyth2/blob/ad990ff1bacce4246159eabcfbc90ad94c16a17b/script/meta/miniarch/Makefile#L40

warpme, 2024-08-16:
> //home/dmig/workspace/minimyth2/images/build/bin/bash: line 1: //home/dmig/workspace/minimyth2/images/build/usr/bin/makepkg: No such file or directory
> make[2]: *** [Makefile:37: build-custom] Error 127

this means we still don't have pacman properly build&installed
may you go to `~/minimyth2/script/miniarch/pacman` and do `make clean install`

(I just build it on my vanilla archlinux. my build log: https://gist.github.com/warpme/31d5e013ca6471a197ea2b87c3cb446b)

dmig, 2024-08-16:
Ok, pacman is building and installing successfully:
```
[0/1] Installing files.
Installing scripts/makepkg-template to /home/dmig/workspace/minimyth2/images/main/usr/bin
Installing scripts/makepkg to /home/dmig/workspace/minimyth2/images/main/usr/bin
Installing scripts/pacman-db-upgrade to /home/dmig/workspace/minimyth2/images/main/usr/bin
Installing scripts/pacman-key to /home/dmig/workspace/minimyth2/images/main/usr/bin
Installing scripts/repo-add to /home/dmig/workspace/minimyth2/images/main/usr/bin
...
```

But the path is different from expected:
```
//home/dmig/workspace/minimyth2/images/build/bin/bash: line 1: //home/dmig/workspace/minimyth2/images/build/usr/bin/makepkg: No such file or directory
```
So makepkg from build tools is missing, not target one.

warpme, 2024-08-16:
> So makepkg from build tools is missing, not target one.

ah - my bad!
pls 
1. go to `~/minimyth2/script/miniarch/pacman` and do `make clean`
2. do `make clean install DESTIMG="build"`

dmig, 2024-08-16:
Ok, not `getopt` is missing...
```
unpacking ArchLinuxARM-aarch64-latest ...
./usr/bin/newgidmap: Cannot restore extended attributes: security.capability security.capability: Operation not permitted
./usr/bin/newuidmap: Cannot restore extended attributes: security.capability security.capability: Operation not permitted
bsdtar: Error exit delayed from previous errors.

...

==> Retrieving sources...
  -> Found kernel.keyblock
  -> Found kernel_data_key.vbprivk
  -> Found linux.preset
  -> Found 60-linux.hook
  -> Found 90-linux.hook
==> WARNING: Skipping all source file integrity checks.
==> Extracting sources...
==> Entering fakeroot environment...
//home/dmig/workspace/minimyth2/images/build/usr/bin/fakeroot: line 46: getopt: command not found
//home/dmig/workspace/minimyth2/images/build/usr/bin/fakeroot: line 50: getopt: command not found
fakeroot, create a fake root environment.
   usage: fakeroot [-l|--lib fakerootlib] [-f|--faked fakedbin]
                   [-i file] [-s file] [-u|--unknown-is-real]
		  [-b|--fd-base fd] [-h|--help] [-v|--version]
                   [--] [command]
make[5]: *** [Makefile:37: build-custom] Error 1
```

dmig, 2024-08-16:
On the second run without `clean` it's even more interesting:
```
 ==> Building miniarch/linux-pkg as a dependency
make[2]: Entering directory '/home/dmig/workspace/minimyth2/script/miniarch/linux-pkg'
[===== NOW BUILDING:	miniarch-linux.pkg-1.0	=====]
	[fetch] complete for miniarch-linux.pkg.
	[checksum] complete for miniarch-linux.pkg.
	[extract] complete for miniarch-linux.pkg.
find: ‘./work/main.d/pkg’: Permission denied
make[2]: *** [../../gar.mk:397: patch] Error 1
make[2]: Leaving directory '/home/dmig/workspace/minimyth2/script/miniarch/linux-pkg'
make[1]: *** [../../gar.lib.mk:581: ../../miniarch/linux-pkg/cookies/main.d/install] Error 2
make[1]: Leaving directory '/home/dmig/workspace/minimyth2/script/miniarch/miniarch-rootfs'
make: *** [../../gar.lib.mk:600: imgdep-main] Error 2
```

warpme, 2024-08-16:
> unpacking ArchLinuxARM-aarch64-latest ...
> ./usr/bin/newgidmap: Cannot restore extended attributes: security.capability security.capability: Operation not permitted
> ./usr/bin/newuidmap: Cannot [...]

what was:
dir where you issue command
command you issued

pls - when you report output of given command  - add info about dir and command you issued....

dmig, 2024-08-16:
That was caused by: `make clean install` from `/home/dmig/workspace/minimyth2/script/miniarch/miniarch-rootfs`

dmig, 2024-08-16:
Found a way to build getopt, as pacman. Now getting permission denied:
```
~/w/m/s/m/miniarch-rootfs (master) [2]> make clean install
 
cleaning MiniArch-aarch64-20240715-6.10.5 ...
 
[===== NOW BUILDING:	miniarch-rootfs-20240715	=====]
install -d cookies/main.d
	[fetch] complete for miniarch-rootfs.
 ==> Running checksum on ArchLinuxARM-aarch64-latest.tar.gz
bdef3220a954dadacf03f18d18544204  download/ArchLinuxARM-aarch64-latest.tar.gz
file ArchLinuxARM-aarch64-latest.tar.gz passes checksum test!
	[checksum] complete for miniarch-rootfs.
install -d work/main.d
 
unpacking ArchLinuxARM-aarch64-latest ...
./usr/bin/newgidmap: Cannot restore extended attributes: security.capability security.capability: Operation not permitted
./usr/bin/newuidmap: Cannot restore extended attributes: security.capability security.capability: Operation not permitted
bsdtar: Error exit delayed from previous errors.
"adding some fixups in archlinuxarm image to make mkinitcpio happy ..."
	[extract] complete for miniarch-rootfs.
	[patch] complete for miniarch-rootfs.
make[1]: Entering directory '/home/dmig/workspace/minimyth2/script/miniarch/miniarch-rootfs'
make[1]: '../../devel/build-system-bins/cookies/build.d/install' is up to date.
make[1]: '../../miniarch/pacman/cookies/build.d/install' is up to date.
make[1]: '../../miniarch/mkinitcpio/cookies/build.d/install' is up to date.
make[1]: '../../devel/fakeroot/cookies/build.d/install' is up to date.
make[1]: Leaving directory '/home/dmig/workspace/minimyth2/script/miniarch/miniarch-rootfs'
make[1]: Entering directory '/home/dmig/workspace/minimyth2/script/miniarch/miniarch-rootfs'
make[1]: '../../miniarch/miniarch-pkg/cookies/main.d/install' is up to date.
 ==> Building miniarch/linux-pkg as a dependency
make[2]: Entering directory '/home/dmig/workspace/minimyth2/script/miniarch/linux-pkg'
[===== NOW BUILDING:	miniarch-linux.pkg-1.0	=====]
	[fetch] complete for miniarch-linux.pkg.
	[checksum] complete for miniarch-linux.pkg.
	[extract] complete for miniarch-linux.pkg.
find: ‘./work/main.d/pkg’: Permission denied
make[2]: *** [../../gar.mk:397: patch] Error 1
make[2]: Leaving directory '/home/dmig/workspace/minimyth2/script/miniarch/linux-pkg'
make[1]: *** [../../gar.lib.mk:581: ../../miniarch/linux-pkg/cookies/main.d/install] Error 2
make[1]: Leaving directory '/home/dmig/workspace/minimyth2/script/miniarch/miniarch-rootfs'
make: *** [../../gar.lib.mk:600: imgdep-main] Error 2
```

warpme, 2024-08-16:
I just build successfully sd card image on vanilla archlinux via  
cd ~/minimyth2/script/meta/miniarch/
run ./build-image-for-board.sh

As you have issue with building roofs on vanilla archlinux (your case; before first successful run ./build-image-for-board.sh) - this means some initial dependencies are missing + they are resolved by ./build-image-for-board.sh successful run (my case).

I'll try to reproduce this on my vanilla archlinux...

As procedure from https://github.com/warpme/miniarch?tab=readme-ov-file#quick-start-1 works ok - I think you have 2 choices:
1. wait will I fix bug
2. delete whole minimyth2 dir and exactly follow https://github.com/warpme/miniarch?tab=readme-ov-file#quick-start-1

dmig, 2024-08-16:
I'll try to debug myself. Waiting 7-8 hours for a completely clean build is not an option.

dmig, 2024-08-16:
`ln -s /usr/bin/getopt /home/dmig/workspace/minimyth2/images/build/bin-build-system` solved missing getopt

warpme, 2024-08-16:
I checked and....it looks all is ok.
build system is designed in a way that very first build of minimyth2 / miniarch MUST be always from `meta/minimyth` `meta/miniarch`
This is a must becasue whole dependencies tree is designed with assumption that: dependences root is in `meta/<project>` dir.
As minimyth2 and miniarch are building for me without issue on current archlinux (fedora40 also tested) - I'm going to close this ticket.
If you will still have build issues - pls:
-for first build follow EXACTLY https://github.com/warpme/miniarch?tab=readme-ov-file#quick-start-1 procedure
-provide me FULL build log

Thx for you cooperation!

closed 2024-08-16 by warpme

dmig, 2024-08-16:
Fixed getopt symlink, kernel update to the latest and rebuild succeded, now failing on initrd build:
```
~/w/m/s/m/miniarch (master)> make update-kernel
...
==> Finished making: linux-firmware 20240709-1 (Fri 16 Aug 2024 21:58:10)
	[build] complete for miniarch-firmware.pkg.
echo " ==> Installing GPL2 license text"
 ==> Installing GPL2 license text
install -d /home/dmig/workspace/minimyth2/images/main/usr/licenses/miniarch-firmware.pkg/
install -m 644 ../../licenses/GPL2.txt /home/dmig/workspace/minimyth2/images/main/usr/licenses/miniarch-firmware.pkg/
mkdir -p cookies/main.d/. && date >> cookies/main.d/install-license-GPL2
	[install] complete for miniarch-firmware.pkg.
make[5]: Leaving directory '/home/dmig/workspace/minimyth2/script/miniarch/linux-firmware-pkg'
make[4]: Leaving directory '/home/dmig/workspace/minimyth2/script/miniarch/miniarch-rootfs'
	[configure] complete for miniarch-rootfs.
 
updating kernel ...
warning: no 'XferCommand' configured
loading packages...
resolving dependencies...
looking for conflicting packages...

...

Packages (1) linux-firmware-20240709-1

Total Installed Size:    15.24 MiB
Net Upgrade Size:      -421.28 MiB

:: Proceed with installation? [Y/n] 
(1/1) checking package integrity                                                                              [#################################################################] 100%
(1/1) loading package files                                                                                   [#################################################################] 100%
(1/1) checking for file conflicts                                                                             [#################################################################] 100%
(1/1) checking available disk space                                                                           [#################################################################] 100%
:: Processing package changes...
(1/1) upgrading linux-firmware                                                                                [#################################################################] 100%
could not change the root directory (Operation not permitted)
error: command failed to execute correctly
 
running mkinitcpio after kernel update ...
//home/dmig/workspace/minimyth2/images/build/bin/bash: line 4: //home/dmig/workspace/minimyth2/images/build/usr/bin/mkinitcpio: No such file or directory
"moving generated initramfs-linux.img to mm2 boot dir ..."
hack removing /usr/lib/modprobe.d/bluetooth-usb.conf to make pacman happy when user is installing bluez pkg...
 
installing miniarch-meta ...
warning: no 'XferCommand' configured
loading packages...
error: 'work/main.d/../../../miniarch-pkg/work/main.d/miniarch-meta-*': could not find or read package
make[3]: *** [Makefile:48: install-miniarch] Error 1
make[3]: Leaving directory '/home/dmig/workspace/minimyth2/script/miniarch/miniarch-rootfs'
make[2]: *** [../../gar.lib.mk:581: ../../miniarch/miniarch-rootfs/cookies/main.d/install] Error 2
make[2]: Leaving directory '/home/dmig/workspace/minimyth2/script/meta/miniarch'
make[1]: *** [../../gar.lib.mk:600: imgdep-main] Error 2
make[1]: Leaving directory '/home/dmig/workspace/minimyth2/script/meta/miniarch'
make: *** [Makefile:45: update-kernel] Error 2
```
Fixed by building mkinitcpio. Build doesn't finish, but at least I got kernel, DTBs and initramfs.