warpme/miniarch #54: How to turn on hardware acceleration on Radxa 3W?

bubnenkoff, 2024-09-02:
Hi! Thanks for your distribution. It’s only works fine on my Radxa 3W. The only issue that it seems that HW acceleration do not works on youtube. 
And I can’t understand how to check does it works or only software rendering works. 

It there any way to check it?

warpme, 2024-09-02:
well - it works well if: user software (your player) can use standardized hw accel. kernel in MiniArch provides.
3W is rk3566 and currently: h264, mpeg2 and vp8 are supported in hardware.
I have users using 3W daily as video player with: MythTV and Kodi.
H264 liveTV plays there with full hw acceleration. 
What video format you trying to play and what is videoplayer?

bubnenkoff, 2024-09-02:
I am trying to view youtube directly from FireFox.

warpme, 2024-09-02:
then non-working HW accell is issue of Firefox. 
MiniArch kernel provides fully to v4l2_request standard complaint APIs for h264 decoding....

bubnenkoff, 2024-09-02:
Which browser can I try?
How I can monitor of using GPU?

warpme, 2024-09-02:
AFAIK there is no Linux browser supporting HW accel on Linux at the moment....but I might be wrong. 
Internet search is your friend :-)

bubnenkoff, 2024-09-02:
Is there any ideas how to watch youtube with Hw acceleration?
I only need more or less usable UI for player.

warpme, 2024-09-02:
IDK
Internet search is your friend :-)

closed 2024-09-09 by warpme