warpme/miniarch #56: How to add WireGuard support?

bubnenkoff, 2024-09-12:
It seems that miniArch do not have WireGuard in kernel. I tried to read how to recompile kernel with WireGuard support, but it was very hard for me. 

Could you add WireGuard support or explain how can I reconfigure kernel to its support?

warpme, 2024-09-16:
Hi,
I added WireGuard in 6.10.10 kernel. Pls update MiniMarch artefacts and report how it goes :-)

bubnenkoff, 2024-09-16:
Thanks! Should I update with pacman or should I use `start`?

warpme, 2024-09-16:
pls go with start (start does special things needed to update kernel)

bubnenkoff, 2024-09-16:
Thanks! I did it. The final goals was to get Amnezia-wg works. But I faced with 2 problems:
1. Afaik there is user space implementation https://github.com/amnezia-vpn/amneziawg-go
But after every running it it claim me that I should use kernel module https://github.com/amnezia-vpn/amneziawg-linux-kernel-module
(it seems that it detect that kernel support it).
1.1 If I right understand amneziawg-go should works if WG is present in Linux kernel (now it's present).

2. But kernel module require source codes for it's compiling and I do not know where to get them.

warpme, 2024-10-09:
Closed as completed (adding wire guard kernel module)