warpme/miniarch #59: TX1 install issues.

guile124, 2024-10-17:
I'm trying to install miniarch in a Tanix TX1 clone (wudong manufacturer).
I followed all steps of the "how to install" file in the TX1 image. The TX1 device is in a fel mode and detected by linux laptop (lsusb) with result:
ALLWINNER technology sunxi Soc OTG connector in FEL/flashing mode

However when i launch the 'boot-kernel-via-usb.sh' i get: 
Awaiting for device ...
./boot-hernel-via-usb.sh: 10: not found

The result of ls -l boot-kernel-via-usb.sh sunxi-fel

-rwxr-xr-x 1 for boot-kernel and same for sunxi-fel

What am i doing wrong? Any suggestion to test?

guile124, 2024-10-17:
Ok, solved using sudo bash boot-kernel-via-usb.sh. However, now I get this:
Awaiting device ... 
Device discovered. Good!!! 
Starting upload U-Boot, Linux Kernel, DTB to device ... 
Found DT name in SPL header: sun50i-h313-tanix-tx1 
Stack pointers: sp_irq=0x00021400, sp=0x00053FFC 
MMU is not enabled by BROM. 
=> Executing the SPL... done. 
usb_bulk_send() ERROR -7: Operation timed out. 
Booting Linux kernel on device. Please wait ... 
Now you can (re)connect the USB key with rootfs ...

After this, the next step fails when I change the USB to TX1; nothing happens.

Tried many things: new USB, re-burned the image, rebooted both devices. Any help is welcome.

warpme, 2024-10-18:
Hi 
I uploaded new image for tx1.
May you download and try?

guile124, 2024-10-18:
I have tried it, but the same problem persists: usb_bulk_send() ERROR -7: Operation timed out, when I run the script. Interestingly, it then prints 'Booting Linux kernel on device. Please wait... Now you can (re)connect the USB key with rootfs...I don't know if the error is ignored and the script completes its function or not. However the next step connecting usb to TX1 do not work. Do you have any suggestions I could try?
I really appreciate your time. Thank you!

warpme, 2024-10-19:
This looks definitelly like issue with running sunxi-fel on Your system.
I updated script and also built sunxi-fel from latest sources (see attached files).
Try with these.
If this will still report issues - you may try install sunxi-tools package on your system and then try to use script.

btw: script uses sunxi-fel from dir where script resides - so always copy sunxi-fel to script dir
[boot-kernel-via-usb.sh.zip](https://github.com/user-attachments/files/17444759/boot-kernel-via-usb.sh.zip)
[sunxi-fel.zip](https://github.com/user-attachments/files/17444760/sunxi-fel.zip)

guile124, 2024-10-19:
Thanks Piotr!, same problem persists with the new files: usb_bulk_send() ERROR -7: Operation timed out.
I also tried with another system Manjaro based with different Arquitecture, tried also with the last and 1.4.1 of sunxi-tools, tried with many diferents usb sticks and usb cables 2.0 and 3.0. 
I'm starting to think that the problem might be with my device. Since it's not a Tanix but a Wudung, maybe the assembler has modified something. Have other users reported a successful installation?

warpme, 2024-10-20:
May you send me pictures of pcb?

guile124, 2024-10-20:
I've tried Lubuntu 22.04.5 LTS and Manjaro xfce 24.1.1 if you suggest any other i can try.
_
[image]

The heat sink doesn't allow you to see much, I could remove it if necessary, or if you need something in more detail, I'll proceed.

warpme, 2024-10-28:
Ah - board is v2.0. 
I develop support for v1.1 and v1.2 as i have such hw in my lab.  
I'm afraid without having physical HW in my lab it might be difficult to get it with mainline Linux (different dram chip, potentially different h313 variant, etc)

guile124, 2024-10-31:
Got it, this explains it all. Thank you!

warpme, 2024-10-31:
I'm closing this ticket due to no hw to work on.

closed 2024-10-31 by warpme