warpme/miniarch #62: Cannot install on vontar - h618 - database cannot be established (created)

yeungon, 2024-11-04:
Hi there! I love your idea and your endeavor to make this project a reality. Recently, I attempted to install MiniArch on my brand-new TV box (Votar, H618, Allwinner), but it didn't work. 

I might have done something wrong. Could you please take a look?

[image]

warpme, 2024-11-04:
It looks you are using minimyth2 image....
For MiniMarch pls use miniarch image (https://github.com/warpme/miniarch/releases)

yeungon, 2024-11-04:
> It looks you are using minimyth2 image.... For MiniMarch pls use miniarch image (https://github.com/warpme/miniarch/releases)

I will report about my progress. Thanks for your quick reply.

yeungon, 2024-11-04:
> It looks you are using minimyth2 image.... For MiniMarch pls use miniarch image (https://github.com/warpme/miniarch/releases)

It works pretty well with the correct image you mentioned. Thanks.