warpme/miniarch #63: Rk3566 X96 X6 box : ethernet not working

chris4git, 2024-11-16:
Hi,
I hope to use my nice X96 X6 TV box as a mini server and just discovered you have images for this box :-)
So thanks for sharing your work !

I uploaded MiniArch-20240715-6.10.10-board-rk3566.x96_x6-SD-Image.img image on a SD card and it boots.
Unfortunately the ethernet interface don't work : it is up (1Gbps/Full - flow control rx/tx) but don't acquire address from the router (who don't receive dhcp requests).
**If I replace the rk3566-x96_x6.dtb by rk3566-box-demo.dtb then it works** (but other problems) !
So it seems this is related to the dtb rk3566-x96_x6.dtb
Any idea to help resolve the problem ?

Thanks in advance
Chris

Kernel logs :
```
[    1.786966] rk_gmac-dwmac fe010000.ethernet: IRQ eth_lpi not found
[    1.787001] rk_gmac-dwmac fe010000.ethernet: IRQ sfty not found
[    1.787317] rk_gmac-dwmac fe010000.ethernet: supply phy not found, using dummy regulator
[    1.788296] rk_gmac-dwmac fe010000.ethernet: clock input or output? (input).
[    1.788315] rk_gmac-dwmac fe010000.ethernet: TX delay(0x4f).
[    1.788329] rk_gmac-dwmac fe010000.ethernet: RX delay(0x2d).
[    1.788351] rk_gmac-dwmac fe010000.ethernet: integrated PHY? (no).
[    1.788437] rk_gmac-dwmac fe010000.ethernet: clock input from PHY
[    1.793464] rk_gmac-dwmac fe010000.ethernet: init for RGMII
[    1.793908] rk_gmac-dwmac fe010000.ethernet: User ID: 0x30, Synopsys ID: 0x51
[    1.793933] rk_gmac-dwmac fe010000.ethernet:     DWMAC4/5
[    1.793947] rk_gmac-dwmac fe010000.ethernet: DMA HW capability register supported
[    1.793957] rk_gmac-dwmac fe010000.ethernet: RX Checksum Offload Engine supported
[    1.793966] rk_gmac-dwmac fe010000.ethernet: TX Checksum insertion supported
[    1.793974] rk_gmac-dwmac fe010000.ethernet: Wake-Up On Lan supported
[    1.794087] rk_gmac-dwmac fe010000.ethernet: TSO supported
[    1.794100] rk_gmac-dwmac fe010000.ethernet: Enable RX Mitigation via HW Watchdog Timer
[    1.794116] rk_gmac-dwmac fe010000.ethernet: Enabled RFS Flow TC (entries=10)
[    1.794132] rk_gmac-dwmac fe010000.ethernet: TSO feature enabled
[    1.794143] rk_gmac-dwmac fe010000.ethernet: Using 32/32 bits DMA host/device width
[    5.296776] rk_gmac-dwmac fe010000.ethernet end0: renamed from eth0
[    6.210413] rk_gmac-dwmac fe010000.ethernet end0: Register MEM_TYPE_PAGE_POOL RxQ-0
[    6.302812] rk_gmac-dwmac fe010000.ethernet end0: PHY [stmmac-0:01] driver [RTL8211F Gigabit Ethernet] (irq=POLL)
[    6.312825] dwmac4: Master AXI performs any burst length
[    6.312870] rk_gmac-dwmac fe010000.ethernet end0: No Safety Features support found
[    6.312902] rk_gmac-dwmac fe010000.ethernet end0: IEEE 1588-2008 Advanced Timestamp supported
[    6.313267] rk_gmac-dwmac fe010000.ethernet end0: registered PTP clock
[    6.314116] rk_gmac-dwmac fe010000.ethernet end0: configuring for phy/rgmii link mode
[    9.393563] rk_gmac-dwmac fe010000.ethernet end0: Link is Up - 1Gbps/Full - flow control rx/tx
```

chris4git, 2024-11-17:
Precision :
if one boot with /dtbs/rockchip/rk3566-box-demo.dtb then reboot with /dtbs/rockchip/rk3566-x96_x6.dtb ethernet works !
If one power off the board between boots (instead of reboot command) , it doesn't works !

Chris

chris4git, 2024-11-21:
Hi,
I tried the latest image MiniArch-20240715-6.12.0-board-rk3566.x96_x6-SD-Image.img.xz without more success with ethernet interface which is still silent.

Any hint ?

Chris

chris4git, 2024-11-21:
After investigation it turns out that what is emitted par the router is received par the box (on end0) but what is emitted par the box is **not** received by the router.
What difference between the two dtbs can explain this ?

Chris

warpme, 2025-01-16:
Oh - X96 boxes are only hell to support.
This vendor has usual habit to:

1. change wirings on PCB (sometimes without any functional requirement; just for change)
2. using different chips (Eth, wifi) without any external sing (e.g. different box name, ver. rev, etc)
3. using different ram chips (ddr3, lpddr3, ddr3l, lpddr4) without any external sing (e.g. different box name, ver. rev, etc)
4. sometimes (i had i such case)  - sing different SoC

All above makes support of such vendor just impossible as:

- p1 makes dead Eth with any reason...
- p2 makes Eth or WiFi not working...
- p3 makes box non-bootable...
- p4.makes box no any sign of life...

In all cases - for fixing this I need:
- my time
- when I have time - having physical box at hand

Doing this project at spare time - I simply don't have time for chasing constantly moving target (like e.g. modified HW in in X96 boxes)

So I want to close this ticket as:
- no hw available to work on
- no motivation to do work on this (if you want to change this - pls consider donation for buying HW to my lab - then i may look what we can do...)   
 
Sorry for that....but this is reality 
(and i suspect it is reality not only of mine FOSS project...) 
   
br

chris4git, 2025-01-16:
Hi,
I thought [x96mini.com](https://x96mini.com/pages/about-us)  was rather serious. They even have linux tv boxes and a Debian digital player !
In the meantime the guy @maka on Armbian forum has posted a dtb which has fixed the ethernet problem...

Chris

warpme, 2025-01-24:
Closed due lack of HW to fix this issue

closed 2025-01-24 by warpme