warpme/miniarch #64: Add support for sun50i-h618-transpeed-8k618-t

ImWez, 2024-12-10:
It's very similar to the sun50i-h618-vontar-h618, you can even use the Vontar image, but you can't install it on the eMMC due to a difference in the dtb

firepower9966, 2024-12-11:
h618-transpeed-8k618-t.dtb file is in the allwinner folder but no image file. please add

ImWez, 2024-12-11:
I have found this repo. Armbian for sun50i-h618-transpeed-8k618-t, maybe help...

https://github.com/NickAlilovic/build

ImWez, 2024-12-11:
[sun50i-h618-transpeed-8k618-t.dtb](https://github.com/user-attachments/files/18100895/sun50i-h618-transpeed-8k618-t.zip)

warpme, 2025-01-09:
Oh adding support for new box I need:
- my time
- when I have time - having physical box at hand

Doing this project at spare time - I don't time for chasing constantly moving target (like e.g. X96 boxes)

So I want to close this ticket as:
- no hw available to work ok
- no motivation to do work for free (if you want to change this - pls consider donation - then i may look what we can do...)   
 
Sorry for that....but this is reality 
(and i suspect it is reality not only of mine FOSS project...) 

br

warpme, 2025-01-16:
can we close this ticket?

firepower9966, 2025-01-16:
Yes

closed 2025-01-16 by ImWez