warpme/miniarch #65: Vontar H618 no sound, bluetooth.

firepower9966, 2024-12-11:
no sound via line out and no bluetooth for vontar h618 image.

warpme, 2025-01-09:
Oh - Vontar boxes are quite difficult to support.
This vendor has usual habit to:

1. using different chips (Eth, wifi) without any external sing (e.g. different box name, ver. rev, etc)
2. using different ram chips (ddr3, lpddr3, ddr3l, lpddr4) without any external sing (e.g. different box name, ver. rev, etc)

All above makes support of such vendor just impossible as:
- p1 makes Eth or WiFi not working...
- p2 makes box non-bootable...

In all cases - for fixing this I need:
- my time
- when I have time - having physical box at hand

Doing this project at spare time - I don't time for chasing constantly moving target (like e.g. X96 boxes)

So I want to close this ticket as:
- no hw available to work ok
- no motivation to do work for free (if you want to change this - pls consider donation - then i may look what we can do...)   
 
Sorry for that....but this is reality 
(and i suspect it is reality not only of mine FOSS project...) 
   
br

warpme, 2025-01-16:
can we close this ticket?

warpme, 2025-01-24:
closed due lack of activity

closed 2025-01-24 by warpme