warpme/miniarch #66: Tanix tx6s-p not boot from SDcard

rock2rap, 2024-12-19:
I have a Tanix tx6s-p with an H616 chip, which normally boots into Android TV. However, I want to use this box for more than just watching movies and listening to music. I want to set up a small server for home use. So, I downloaded the correct image file for my model and successfully burned it onto an SD card using BalenaEtcher. But when I insert the SD card into the TF slot of the tx6s-p and power it on, nothing happens. It seems like it's not booting from the SD card. Why is this happening? Have I made a mistake somewhere? The box itself works fine.

[image]
[image]

warpme, 2025-01-09:
Oh - Tanix boxes are quite difficult to support.
This vendor has usual habit to:

1. using different chips (Eth, wifi) without any external sing (e.g. different box name, ver. rev, etc)
2. using different ram chips (ddr3, lpddr3, ddr3l, lpddr4) without any external sing (e.g. different box name, ver. rev, etc)

All above makes support of such vendor just impossible as:
- p1 makes Eth or WiFi not working...
- p2 makes box non-bootable...

In all cases - for fixing this I need:
- my time
- when I have time - having physical box at hand

Doing this project at spare time - I don't time for chasing constantly moving target (like e.g. X96 boxes)

So I want to close this ticket as:
- no hw available to work ok
- no motivation to do work for free (if you want to change this - pls consider donation - then i may look what we can do...)   
 
Sorry for that....but this is reality 
(and i suspect it is reality not only of mine FOSS project...) 
   
br

warpme, 2025-01-16:
Can we close this ticket?

closed 2025-01-24 by warpme