warpme/miniarch #67: X96Q (likely v5.1), no ethernet

Guybrush3pw00d, 2024-12-27:
My best regards and first of all thanks for all your efforts. 
I've just bought an X96Q and I've tried to put [this one](MiniArch-20240715-6.12.0-board-h313.x96_q_v5.1-SD-Image.img.xz) on it. 
I don't know the board version because, if possible, I'd like to avoid damaging the seal for warranty purposes but I'm sure it actually has 2Gb of RAM given that I get them all even under Linux and not only under Android which can provide fake informations in this regard. 
By what I've understood, an effective way to determine the board revision is a trial and check procedure using the miniarch images (available for all the currently known and therefore handled revisions): if it boots, the board is likely to be that one (or at least a very similar one with minor tweaking necessary); if it doesn't, try another image for a different revision.  This way, I've determined mine should be a v5.1. 
At this stage, I'm purposedly sticking with sdcard (for the same reason of leaving a door open for returning it if necessary). 
Miniarch boots but it can't start ethernet so I can't do almost anything. 
I'm starting to fear my assumption could be wrong and mine isn't actually a v5.1.
[Here](https://ibb.co/N38TWjb) is a screenshot I consider relevant. 
Have a nice day!

Tribalwarrior, 2024-12-29:
Same issue. Have 5.1 board, no Ethernet, and iw: command not fownd

Guybrush3pw00d, 2024-12-29:
I haven't taken a look at closed tickets before opening this one and doing that was a mistake on my side I apologize for. I've been able to complete the whole setup stage by using usb tethering as adviced in a previous (and therefore closed) ticket. It was somthing I could have thought of by myself anyway, another mistake on my side.
After successfully completing EOS setup at least wifi has begun working while still no output from lspci, still no ethernet and NY (as to be expected given no output from lspci).
What's even worse is that hw acceleration seems completely missing making everything almost unusable. I keep thinking my guess about the board revision could be wrong.

warpme, 2025-01-09:
Oh - X96 boxes are only hell to support.
This vendor has usual habit to:

1. change wirings on PCB (sometimes without any functional requirement; just for chnage)
2. using different chips (Eth, wifi) without any external sing (e.g. different box name, ver. rev, etc)
3. using different ram chips (ddr3, lpddr3, ddr3l, lpddr4) without any external sing (e.g. different box name, ver. rev, etc)
4. sometimes (i had i such case)  - sing different SoC

All above makes support of such vendor just impossible as:

- p1 makes dead Eth with any reason...
- p2 makes Eth or WiFi not working...
- p3 makes box non-bootable...
- p4.makes box no any sign of life...

In all cases - for fixing this I need:
- my time
- when I have time - having physical box at hand

Doing this project at spare time - I don't time for chasing constantly moving target (like e.g. X96 boxes)

So I want to close this ticket as:
- no hw available to work ok
- no motivation to do work for free (if you wan to chnage this - pls consider donation - then i may look what we can do...)   
 
Sorry for that....but this is reality 
(and i suspect it is reality not only of mine FOSS project...) 
   
br

closed 2025-01-09 by warpme