warpme/miniarch #68: Another possible variant of tx6s - unable to boot

zer0exia, 2024-12-29:
I've tried both tx6s and tx6s_axp313 sd image from releases, but the box aren't able to boot from any one of them. I do have output of UART that I can provide, and if you could help me build the appropriate dtb file for this version, it would be appreciated. 

I would gladly provide any more information that you require.

[androidbox.txt](https://github.com/user-attachments/files/18269415/androidbox.txt)

warpme, 2025-01-09:
Oh - developing dtb without hw at hand is almost impossible....
Moreover - issue might be i.e. with dram params in bootloader... 
If somebody will provide me hw - i can try....

warpme, 2025-01-16:
Oh - Tanix boxes are only hell to support.
This vendor has usual habit to:

1. change wirings on PCB (sometimes without any functional requirement; just for change)
2. using different chips (Eth, wifi) without any external sing (e.g. different box name, ver. rev, etc)
3. using different ram chips (ddr3, lpddr3, ddr3l, lpddr4) without any external sing (e.g. different box name, ver. rev, etc)
4. sometimes (i had i such case)  - sing different SoC

All above makes support of such vendor just impossible as:

- p1 makes dead Eth with any reason...
- p2 makes Eth or WiFi not working...
- p3 makes box non-bootable...
- p4.makes box no any sign of life...

In all cases - for fixing this I need:
- my time
- when I have time - having physical box at hand

Doing this project at spare time - I simply don't have time for chasing constantly moving target (like e.g. modified HW in Tanix boxes)

So I want to close this ticket as:
- no hw available to work on
- no motivation to do work on this moving target (if you want to change this - pls consider donation for buying HW to my lab - then i may look what we can do...)
 
Sorry for that....but this is reality 
(and i suspect it is reality not only of mine FOSS project...) 
   
br

warpme, 2025-01-24:
closed due lack of activity

closed 2025-01-24 by warpme