warpme/miniarch #69: Can't run AppImage - fuse.service status=FAILURE

q4a, 2025-01-03:
Hi. I can't run AppImage on miniarch+EndeavourOS.
modprobe@fuse.service has status=1/FAILURE. Here is my commands and checks.
Is it possible that fuse disabled in kernel?
```
[q@rock5b ~]$ ~/Downloads/rpcs3-v0.0.34-17293-a8f80ebf_linux_aarch64.AppImage 
fuse: device not found, try 'modprobe fuse' first                                                                                                                                             
                                                                                                                                                                                              
Cannot mount AppImage, please check your FUSE setup.                                                                                                                                          
You might still be able to extract the contents of this AppImage                                                                                                                              
if you run it with the --appimage-extract option.                                                                                                                                             
See https://github.com/AppImage/AppImageKit/wiki/FUSE                                                                                                                                         
for more information                                                                                                                                                                          
open dir error: No such file or directory                                                                                                                                                     
[q@rock5b ~]$ sudo modprobe fuse
[sudo] password for q: 
modprobe: FATAL: Module fuse not found in directory /lib/modules/6.12.7
[q@rock5b ~]$ sudo pacman -S fuse2 fuse3
warning: fuse2-2.9.9-5 is up to date -- reinstalling
warning: fuse3-3.16.2-1 is up to date -- reinstalling
resolving dependencies...
looking for conflicting packages...

Packages (2) fuse2-2.9.9-5  fuse3-3.16.2-1

Total Installed Size:  1.18 MiB
Net Upgrade Size:      0.00 MiB

:: Proceed with installation? [Y/n] n
[q@rock5b ~]$ mount |grep fuse
[q@rock5b ~]$ systemctl status sys-fs-fuse-connections.mount
○ sys-fs-fuse-connections.mount - FUSE Control File System
     Loaded: loaded (/usr/lib/systemd/system/sys-fs-fuse-connections.mount; static)
     Active: inactive (dead)
  Condition: start condition unmet at Fri 2025-01-03 20:27:30 MSK; 36min ago
      Where: /sys/fs/fuse/connections
       What: fusectl
       Docs: https://docs.kernel.org/filesystems/fuse.html
             https://systemd.io/API_FILE_SYSTEMS
[q@rock5b ~]$ systemctl status modprobe@fuse.service 
○ modprobe@fuse.service - Load Kernel Module fuse
     Loaded: loaded (/usr/lib/systemd/system/modprobe@.service; static)
     Active: inactive (dead) since Fri 2025-01-03 20:27:30 MSK; 37min ago
 Invocation: 00d5d04ea4c044cb8d80cfea781da4dd
       Docs: man:modprobe(8)
   Main PID: 346 (code=exited, status=1/FAILURE)
   Mem peak: 1.5M
        CPU: 16ms
[q@rock5b ~]$ sudo dmesg | grep fuse
[sudo] password for q:                                                                                                                                                                        
[    4.585523] systemd[1]: Starting Load Kernel Module fuse...
[q@rock5b ~]$ uname -a
Linux rock5b 6.12.7 #1 SMP PREEMPT Fri Dec 27 16:51:30 CET 2024 aarch64 GNU/Linux
```

warpme, 2025-01-09:
this looks to me like kernel is missing fuse modules.
may you investigate what exactly kernel modules are required for fuse?

warpme, 2025-01-10:
I update kernel and it should now include fuse module.
Can you update and try?

q4a, 2025-01-10:
Thanks! I updated kernel to:
```
Linux rock5b 6.12.9 #1 SMP PREEMPT Fri Jan 10 16:55:30 CET 2025 aarch64 GNU/Linux
```
And AppImage works now)

closed 2025-01-10 by q4a