warpme/miniarch #70: Orange Pi 5-Plus:  MiniArch-EndeavourOS Gnome Desktop - Firefox, Chromium or MPV No VPU Hardware Acceleration

JFLim1, 2025-01-04:
Hi, Just install MiniArch EndeavourOS with Gnome desktop environment on Orange Pi 5-Plus.  Firefox_v133 or Chromium_v131 both don't seem to have vpu hw acceleration when streaming Youtube h264 videos.  

Installed mpv and yt-dlp. No luck with vpu hardware acceleration with mpv too.

Had followed the instructions to Install MiniArch and EndeavourOS.as set out in "Quick Start".  Is there any other settings or packages needed to be installed to have VPU hardware acceleration on Firefox, Chromium and mpv?

Happy New Year.

warpme, 2025-01-09:
iirc - firefox supports hw accell only via vaapi api.
miniarch on op5plus offers v4l2_request api and there is not yet v4l2-vaapi backend available....
short answer: we need to wait till somebody writes such v4l2-vaapi backend software :-(

JFLim1, 2025-01-09:
> I think issue might be at v4l2_request at chromium building.
> Are you sure that your chromium build correctly detects/uses v4l2_request offered by kernel?

Am using the Chromium from alarm-repo and currently is  Version 131.0.6778.204 (Official Build) Arch Linux ARM (64-bit).  Am not sure whether v4l2_request is included.

warpme, 2025-01-10:
Yes - mpv supports v4l2_request
Here is output from mpv 0.39 playing h264 video
(i not provided drm_plane id - so only hw decondig is in use; there is no DMA transfers of decoded video frame to framebuffer)     

```
● Video  --vid=1               (h264 1920x1080 25 fps)
● Audio  --aid=1  --alang=pol  (mp2 2ch 48000 Hz 320 kbps)
○ Audio  --aid=2  --alang=org  (ac3 2ch 48000 Hz 448 kbps)
○ Subs   --sid=1  --slang=pol  (dvb_teletext)
○ Subs   --sid=2  --slang=pol  (dvb_subtitle)
[vo/gpu/opengl] Initializing GPU context 'drm'
[vo/gpu/drm] VT_GETMODE failed: Inappropriate ioctl for device
[vo/gpu/drm] Failed to set up VT switcher. Terminal switching will be unavailable.
[vo/gpu/drm] Picked DRM card 0, primary node /dev/dri/card0 as the default.
[vo/gpu/drm] Driver: rockchip 1.0.0 (0)
[vo/gpu/drm] Connector 65 currently connected to encoder 64
[vo/gpu/drm] Selected Encoder 64 with CRTC 42
[vo/gpu/drm] Selected mode: 1920x1080 (1920x1080@60.00Hz)
[vo/gpu/drm] DRM Atomic support found
[vo/gpu/drm] Using primary plane 32 as draw plane
[vo/gpu/drm] Failed to find drmprime plane with idx=-2. drmprime-overlay hwdec interop will not work
[vo/gpu] GBM_FORMAT_ARGB8888 supported by draw plane.
[vo/gpu] Supported modifier: 0x0
[vo/gpu] Creating GBM device
[vo/gpu] Initializing GBM surface (1920 x 1080)
[vo/gpu/opengl] Initializing EGL
[vo/gpu/opengl] EGL_VERSION=1.5
[vo/gpu/opengl] EGL_VENDOR=Mesa Project
[vo/gpu/opengl] EGL_CLIENT_APIS=OpenGL OpenGL_ES
[vo/gpu/opengl] Trying to create Desktop OpenGL context.
[vo/gpu/opengl] Attempting to find EGLConfig matching GBM_FORMAT_ARGB8888
[vo/gpu/opengl] Found matching EGLConfig for GBM_FORMAT_ARGB8888
[vo/gpu/opengl] Initializing EGL surface
[vo/gpu] GL_VERSION='3.1 Mesa 24.3.3'
[vo/gpu] Detected desktop OpenGL 3.1.
[vo/gpu] GL_VENDOR='Mesa'
[vo/gpu] GL_RENDERER='Mali-T860 (Panfrost)'
[vo/gpu] GL_SHADING_LANGUAGE_VERSION='1.40'
[vo/gpu] Loaded extension GL_ARB_sync.
[vo/gpu] Loaded extension GL_ARB_invalidate_subdata.
[vo/gpu] Loaded extension GL_ARB_get_program_binary.
[vo/gpu] Loaded extension GL_ARB_buffer_storage.
[vo/gpu] Loaded extension GL_ARB_shader_image_load_store.
[vo/gpu] Loaded extension GL_ARB_shader_storage_buffer_object.
[vo/gpu] Loaded extension GL_ARB_arrays_of_arrays.
[vo/gpu] Loaded extension GL_ARB_debug_output.
[vo/gpu] Loaded extension GL_KHR_debug.
[vo/gpu/opengl] Preparing framebuffer
[vo/gpu/drm] crtc is not VRR capable
[vo/gpu/drm] Monitor pixel aspect: 1
[vo/gpu/opengl] Could not find path to render node.
[vo/gpu] Testing FBO format rgba16f
[vo/gpu] Using FBO format rgba16f.
[vo/gpu] Disabling HDR peak computation (one or more of the following is not supported: compute shaders=0, SSBO=1).
[vo/gpu] Assuming 60.000000 FPS for display sync.
[vd] Container reported FPS: 25.000000
[vd] Codec list:
[vd]     h264 - H.264 / AVC / MPEG-4 AVC / MPEG-4 part 10
[vd]     h264_v4l2m2m (h264) - V4L2 mem2mem H.264 decoder wrapper
[vd]     h264_cuvid (h264) - Nvidia CUVID H264 decoder
[vd] Opening decoder h264
[vd] Looking at hwdec h264-nvdec...
[vo/gpu] Loading hwdec drivers for format: 'cuda'
[vo/gpu] Loading hwdec driver 'cuda'
Cannot load libcuda.so.1
[vo/gpu/cuda] Failed to load CUDA symbols
[vo/gpu] Loading failed.
[vd] Could not create device.
[vd] Looking at hwdec h264-vaapi...
[vo/gpu] Loading hwdec drivers for format: 'vaapi'
[... 53 lines cut ...]
[ffmpeg/video] h264: Missing reference picture, default is 65757
[ffmpeg/video] h264: reference picture missing during reorder
[ffmpeg/video] h264: Missing reference picture, default is 65764
AO: [alsa] 48000Hz stereo 2ch s16
Using hardware decoding (v4l2request-copy).
[vd] Decoder format: 1920x1080 nv12 bt.709/bt.709/bt.1886/limited/auto CL=mpeg2/4/h264 crop=1920x1080+0+0
VO: [gpu] 1920x1080 nv12
[vo/gpu] reconfig to 1920x1080 nv12 bt.709/bt.709/bt.1886/limited/display CL=mpeg2/4/h264 crop=1920x1080+0+0
[vo/gpu] Resize: 1920x1080
[vo/gpu] Window size: 1920x1080 (Borders: l=0 t=0 r=0 b=0)
[vo/gpu] Video source: 1920x1080 (1:1)
[vo/gpu] Video display: (0, 0) 1920x1080 -> (0, 0) 1920x1080
[vo/gpu] Video scale: 1.000000/1.000000
[vo/gpu] OSD borders: l=0 t=0 r=0 b=0
[vo/gpu] Video borders: l=0 t=0 r=0 b=0
[vo/gpu] Reported display depth: 8
[vo/gpu] Texture for plane 0: 1920x1080
[vo/gpu] Texture for plane 1: 960x540
[vo/gpu] Testing FBO format rgba16f
[vo/gpu] Using FBO format rgba16f.
[ffmpeg/demuxer] mpegts: PES packet size mismatch
[ffmpeg/demuxer] mpegts: Packet corrupt (stream = 1, dts = 6690260374).
[vo/gpu] Dither to 8.
AV: 00:02:04 / 00:02:04 (100%) A-V:  0.000 Cache: 0.0s
```

JFLim1, 2025-01-12:
> Yes - mpv supports v4l2_request
> Here is output from mpv 0.39 playing h264 video
> (i not provided drm_plane id - so only hw decondig is in use; there is no DMA transfers of decoded video frame to framebuffer)

Read with interest.  Just installed mpv and yt-dlp from alarm-repo. Use mpv to play a local h264 video but HTOP is indicating the Opi5-Plus the 4 Big Core CPUs is running at almost >80% most of the time. Doesn't look like there is h

```
[alarm@MiniArch-opi5plus ~]$ mpv --version
mpv v0.39.0 Copyright © 2000-2024 mpv/MPlayer/mplayer2 projects
 built on Nov 30 2024 20:17:13
libplacebo version: v7.349.0
FFmpeg version: n7.1
FFmpeg library versions:
   libavcodec      61.19.100
   libavdevice     61.3.100
   libavfilter     10.4.100
   libavformat     61.7.100
   libavutil       59.39.100
   libswresample   5.3.100
   libswscale      8.3.100
[alarm@MiniArch-opi5plus ~]$ 

[alarm@MiniArch-opi5plus ~]$ mpv --vo=gpu --hwdec=auto '/run/media/alarm/writable/home/jfl/Videos/jellyfish-120-mbps-4k-uhd-h264.mkv' 
● Video  --vid=1  --vlang=eng  (h264 3840x2160 29.97 fps) [default]
[vo/gpu/wayland] Unable to set DRM atomic cap: Operation not supported
[ffmpeg] AVHWDeviceContext: Instance creation failure: VK_ERROR_INCOMPATIBLE_DRIVER
Failed to open VDPAU backend libvdpau_nvidia.so: cannot open shared object file: No such file or directory
[ffmpeg/video] h264_v4l2m2m: Could not find a valid device
[ffmpeg/video] h264_v4l2m2m: can't configure decoder
Could not open codec.
VO: [gpu] 3840x2160 yuv420p
V: 00:00:29 / 00:00:30 (100%)
Exiting... (End of file)
[alarm@MiniArch-opi5plus ~]$
```

warpme, 2025-01-16:
I suggest leave mpv as player of choice. Every time I was testing mpv on my systems - there was some kind of issues...
As KODI and MythTV are working perfectly fine for me - i'm qualifying this as mpv issues - not miniarch....

By above (mpv issue; not miniarch) - may we close this ticket?

closed 2025-01-16 by JFLim1

JFLim1, 2025-05-05:
Hi @warpme,

FYI, managed to find mpv-git and ffmpeg with v4l2request from https://drive.google.com/drive/folders/1hAr_PB52fIBcZOgtx2WoGghv63H1-JzB.

MiniArch-EOS-Gnome with the above mpv and ffmpeg now support vpu hw acceleration albeit only for h264 video up to 1080p/60 resolution.

Clapper video player also support up to 1080p/30 when streaming Youtube h264 videos but can play local h264 1080/60.