warpme/miniarch #71: Failed to Update MiniArch With Start 5

JFLim1, 2025-01-05:
Hi @warpme,

Tried to update MiniArch with "start 5" but it failed.

```
---- MiniArch convienience script v1.3, (c)Piotr Oniszczuk ----
 
Please choose action by pressing key [1..9]
 
  (1) Init pacman keyring and populate packages signing keys
  (2) Full update (ArchLinuxArm codebase)
  (3) Install on eMMC (optinal)
  (4) Resize footfs partition
  (5) Update MiniArch artefacts
  (6) Install EndeavourOS
 
or press Enter to exit...

5
 
    Please confirm updating MiniArch speciffic components by pressing 'y'
    or press Enter (or any other key) to cancel ...
    (action may require root password)
 
y
:: Synchronizing package databases...
 miniarch.db failed to download
error: failed retrieving file 'miniarch.db' from miniarch.mooo.com : Connection timed out after 10002 milliseconds
error: failed to synchronize all databases (download library error)
 
    Requested command ended with error!
    Please fill GitHub ticket with details of this failed action. Thx in advance!
 
[root@opi5plus ~]# 
```

warpme, 2025-01-09:
Well - this tooks to me like temporarily Internet hickup.
I just checked my and now access works ok.
may you try again and report?

JFLim1, 2025-01-10:
> may you try again and report?

Today tried for the last 1 hour or so, connection failed.
```

    Please confirm updating MiniArch speciffic components by pressing 'y'
    or press Enter (or any other key) to cancel ...
    (action may require root password)
 
y
:: Synchronizing package databases...
 miniarch.db failed to download
error: failed retrieving file 'miniarch.db' from miniarch.mooo.com : Connection timed out after 10001 milliseconds
error: failed to synchronize all databases (download library error)
 
    Requested command ended with error!
    Please fill GitHub ticket with details of this failed action. Thx in advance!
 
[alarm@MiniArch-opi5plus ~]$ 
```

JFLim1, 2025-01-16:
Resolved

closed 2025-01-16 by JFLim1