warpme/miniarch #72: MiniArch - Kernel 6.12.7 CPU Load Average Always Greater Than 1 Even On Idling - Results Higher CPU Temperature

JFLim1, 2025-01-05:
Hi @warpme,

Installed MiniArch EndeavourOS with Gnome DE yesterday Orange Pi 5 Plus (OPI5-Plus).  

Noticed that HTOP reports CPU Load Average is always greater than 1 even when the sbc is on idle.  Had similar results on OPI5-Plus with https://github.com/Joshua-Riek/ubuntu-rockchip/issues/606.  The issue was the enabling "hdmirx" is causing the high CPU Load Average > 1.

Is there a way to turn-off or disable "hdmirx" in MiniArch-EndevourOS?

warpme, 2025-01-09:
Well - i done test with 

- installing miniarch,
- updating to latest kernel (6.12.8),
- installing eos
- 2 day usage 

and i can't reproduce this issue

I'm not including rk3588 hdmi rx driver in miniarch - so your issue is not related to hdmi rx driver.
 
I recall sometime ago i had similar issue on mine opi5pro - but it was related to mmc (sd card) and was some time ago (few kernel releases back). 
Since then - using miniarch on few devices constantly - i'm not observing this issue anymore. 

When you have this issue - may you provide me some more info like: what process/thread consumes cpu, kernel interrupts stats, dmesg, etc

JFLim1, 2025-01-09:
After boot up, the CPU Load Average  > 1  as indicated by HTOP even with no other application in use or launch and idle for say 10 minutes or more.

I have just reflash MiniArch-EOS will monitor again.

Is capturing a screen shot of HTOP sufficient?

warpme, 2025-01-09:
may you pls provide here output of:

`cat /proc/interrupts`

`ps aux`

JFLim1, 2025-01-10:
> may you pls provide here output of:
> 
> `cat /proc/interrupts`

```
[alarm@MiniArch-opi5plus ~]$ cat /proc/interrupts
           CPU0       CPU1       CPU2       CPU3       CPU4       CPU5       CPU6       CPU7       
 18:          0          0          0          0          0          0          0          0     GICv3  25 Level     vgic
 19:     500054     312103     615366     290859     131556     106248      83820      80870     GICv3  26 Level     arch_timer
 20:          0          0          0          0          0          0          0          0     GICv3  27 Level     kvm guest vtimer
 23:          0          0          0          0          0          0          0          0     GICv3  30 Level     kvm guest ptimer
 25:      99718     204588     433880     170236      34532      31597      34886      31248     GICv3 321 Level     rk_timer
 31:          0          0          0          0          0          0          0          0     GICv3 425 Level     rockchip_usb2phy
 32:         17          0          0          0          0          0          0          0     GICv3 423 Level     rockchip_usb2phy
 33:          1          0          0          0          0          0          0          0     GICv3 424 Level     rockchip_usb2phy
 34:          0          0          0          0          0          0          0          0     GICv3 426 Level     rockchip_usb2phy
 35:          0          0          0          0          0          0          0          0     GICv3 118 Level     fea10000.dma-controller
 36:          0          0          0          0          0          0          0          0     GICv3 119 Level     fea10000.dma-controller
 37:          0          0          0          0          0          0          0          0     GICv3 120 Level     fea30000.dma-controller
 38:          0          0          0          0          0          0          0          0     GICv3 121 Level     fea30000.dma-controller
 39:        562          0          0          0          0          0          0          0     GICv3 122 Level     fed10000.dma-controller
 40:          0          0          0          0          0          0          0          0     GICv3 123 Level     fed10000.dma-controller
 41:        319          0          0          0          0          0          0          0     GICv3 365 Level     ttyS2
 43:          0          0          0          0          0          0          0          0     GICv3 150 Level     fdb50800.iommu
 44:          0          0          0          0          0          0          0          0     GICv3 153 Level     fdba0800.iommu
 45:          0          0          0          0          0          0          0          0     GICv3 155 Level     fdba4800.iommu
 46:          0          0          0          0          0          0          0          0     GICv3 157 Level     fdba8800.iommu
 47:          0          0          0          0          0          0          0          0     GICv3 159 Level     fdbac800.iommu
 48:          0          0          0          0          0          0          0          0     GICv3 128 Level     fdc38700.iommu, fdc40700.iommu
 49:     169211          0          0          0          0          0          0          0     GICv3 188 Level     fdd97e00.iommu, fdd90000.vop
 50:          2          0          0          0          0          0          0          0     GICv3 392 Level     dw-hdmi-qp-hpd
 51:       1285          0          0          0          0          0          0          0     GICv3 204 Level     fde80000.hdmi
 52:          0          0          0          0          0          0          0          0     GICv3 202 Level     dw-hdmi-qp-cec
 53:          0          0          0          0          0          0          0          0     GICv3 393 Level     dw-hdmi-qp-hpd
 54:          0          0          0          0          0          0          0          0     GICv3 208 Level     fdea0000.hdmi
 55:          0          0          0          0          0          0          0          0     GICv3 206 Level     dw-hdmi-qp-cec
 56:        618          0          0          0          0          0          0          0     GICv3 360 Level     feb20000.spi
 57:          2          0          0          0          0          0          0          0  rockchip_gpio_irq   7 Level     rk806
 58:          0          0          0          0          0          0          0          0     rk806   0 Edge      rk805_pwrkey_fall
 59:          0          0          0          0          0          0          0          0     rk806   1 Edge      rk805_pwrkey_rise
 74:        144          0          0          0          0          0          0          0     GICv3 349 Level     fd880000.i2c
 75:   60632214          0          0          0          0          0          0          0     GICv3 355 Level     fec80000.i2c
 76:        246          0          0          0          0          0          0          0     GICv3 356 Level     fec90000.i2c
 77:        133          0          0          0          0          0          0          0     GICv3 237 Level     mmc1
 78:          0          0          0          0          0          0          0          0  GICv3-23   0 Level     arm-pmu
 79:          0          0          0          0          0          0          0          0  GICv3-23   1 Level     arm-pmu
[... cut ...]
For the last 40 minutes with just one or two terminal open the CPU Load Average is between 0.98 (for less than 5 secs) to 1.25. 
![image](https://github.com/user-attachments/assets/dff3ac3d-dbfc-4aa0-8d6a-6a057780c8bd)
![Screenshot from 2025-01-10 11-00-37](https://github.com/user-attachments/assets/3bc949a6-a091-4127-9d24-5e58ad437f00)

![image](https://github.com/user-attachments/assets/9f1cafa2-0ea6-424b-b2d2-704c710d470c)
![Screenshot from 2025-01-10 11-11-12](https://github.com/user-attachments/assets/d4472d08-3a42-42c0-8aaf-2a79c1a3dcc0)

warpme, 2025-01-10:
And what about `ps aux`

JFLim1, 2025-01-10:
> And what about `ps aux`

Oops did not unstand what "ps aux" meant earlier. 

```
[alarm@MiniArch-opi5plus ~]$ ps aux
USER         PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
root           1  0.0  0.0  23988 13356 ?        Ss   21:31   0:01 /sbin/init
root           2  0.0  0.0      0     0 ?        S    21:31   0:00 [kthreadd]
root           3  0.0  0.0      0     0 ?        S    21:31   0:00 [pool_workqueu
root           4  0.0  0.0      0     0 ?        I<   21:31   0:00 [kworker/R-rcu
root           5  0.0  0.0      0     0 ?        I<   21:31   0:00 [kworker/R-syn
root           6  0.0  0.0      0     0 ?        I<   21:31   0:00 [kworker/R-net
root          11  0.0  0.0      0     0 ?        I<   21:31   0:00 [kworker/R-mm_
root          12  0.0  0.0      0     0 ?        I    21:31   0:00 [rcu_tasks_kth
root          13  0.0  0.0      0     0 ?        I    21:31   0:00 [rcu_tasks_rud
root          14  0.0  0.0      0     0 ?        I    21:31   0:00 [rcu_tasks_tra
root          15  0.0  0.0      0     0 ?        S    21:31   0:00 [ksoftirqd/0]
root          16  0.0  0.0      0     0 ?        I    21:31   0:00 [rcu_preempt]
root          17  0.0  0.0      0     0 ?        S    21:31   0:00 [rcu_exp_par_g
root          18  0.0  0.0      0     0 ?        S    21:31   0:00 [rcu_exp_gp_kt
root          19  0.0  0.0      0     0 ?        S    21:31   0:00 [migration/0]
root          20  0.0  0.0      0     0 ?        S    21:31   0:00 [cpuhp/0]
root          21  0.0  0.0      0     0 ?        S    21:31   0:00 [cpuhp/1]
root          22  0.0  0.0      0     0 ?        S    21:31   0:00 [migration/1]
root          23  0.0  0.0      0     0 ?        S    21:31   0:00 [ksoftirqd/1]
root          26  0.0  0.0      0     0 ?        S    21:31   0:00 [cpuhp/2]
root          27  0.0  0.0      0     0 ?        S    21:31   0:00 [migration/2]
root          28  0.0  0.0      0     0 ?        S    21:31   0:00 [ksoftirqd/2]
root          31  0.0  0.0      0     0 ?        S    21:31   0:00 [cpuhp/3]
root          32  0.0  0.0      0     0 ?        S    21:31   0:00 [migration/3]
root          33  0.0  0.0      0     0 ?        S    21:31   0:00 [ksoftirqd/3]
root          36  0.0  0.0      0     0 ?        S    21:31   0:00 [cpuhp/4]
root          37  0.0  0.0      0     0 ?        S    21:31   0:00 [migration/4]
root          38  0.0  0.0      0     0 ?        S    21:31   0:00 [ksoftirqd/4]
root          41  0.0  0.0      0     0 ?        S    21:31   0:00 [cpuhp/5]
root          42  0.0  0.0      0     0 ?        S    21:31   0:00 [migration/5]
root          43  0.0  0.0      0     0 ?        S    21:31   0:00 [ksoftirqd/5]
root          46  0.0  0.0      0     0 ?        S    21:31   0:00 [cpuhp/6]
root          47  0.0  0.0      0     0 ?        S    21:31   0:00 [migration/6]
root          48  0.0  0.0      0     0 ?        S    21:31   0:00 [ksoftirqd/6]
root          50  0.0  0.0      0     0 ?        I<   21:31   0:00 [kworker/6:0H-
root          51  0.0  0.0      0     0 ?        S    21:31   0:00 [cpuhp/7]
root          52  0.0  0.0      0     0 ?        S    21:31   0:00 [migration/7]
root          53  0.0  0.0      0     0 ?        S    21:31   0:00 [ksoftirqd/7]
root          56  0.0  0.0      0     0 ?        S    21:31   0:00 [kdevtmpfs]
root          57  0.0  0.0      0     0 ?        I<   21:31   0:00 [kworker/R-ine
root          59  0.0  0.0      0     0 ?        S    21:31   0:00 [kauditd]
root          62  0.0  0.0      0     0 ?        S    21:31   0:00 [khungtaskd]
root          63  0.0  0.0      0     0 ?        S    21:31   0:00 [oom_reaper]
root          64  0.0  0.0      0     0 ?        I<   21:31   0:00 [kworker/R-wri
root          65  0.0  0.0      0     0 ?        S    21:31   0:00 [kcompactd0]
root          66  0.0  0.0      0     0 ?        SN   21:31   0:00 [ksmd]
root          67  0.0  0.0      0     0 ?        I<   21:31   0:00 [kworker/R-cry
root          68  0.0  0.0      0     0 ?        I<   21:31   0:00 [kworker/R-kin
root          69  0.0  0.0      0     0 ?        I<   21:31   0:00 [kworker/R-kbl
root          72  0.0  0.0      0     0 ?        I<   21:31   0:00 [kworker/R-tif
root          73  0.0  0.0      0     0 ?        I<   21:31   0:00 [kworker/R-dev
root          74  0.0  0.0      0     0 ?        S    21:31   0:00 [watchdogd]
root          77  0.0  0.0      0     0 ?        I<   21:31   0:00 [kworker/7:1H-
root          78  0.0  0.0      0     0 ?        I<   21:31   0:00 [kworker/R-rpc
root          79  0.0  0.0      0     0 ?        I<   21:31   0:00 [kworker/R-xpr
root          80  0.0  0.0      0     0 ?        S    21:31   0:00 [kswapd0]
root          81  0.0  0.0      0     0 ?        I<   21:31   0:00 [kworker/R-nfs
root          82  0.0  0.0      0     0 ?        I<   21:31   0:00 [kworker/R-kth
root          83  0.0  0.0      0     0 ?        S    21:31   0:00 [irq/31-rockch
root          84  0.0  0.0      0     0 ?        S    21:31   0:00 [irq/34-rockch
root          92  0.0  0.0      0     0 ?        S    21:31   0:00 [irq/50-dw-hdm
root          93  0.0  0.0      0     0 ?        S    21:31   0:00 [cec-dw_hdmi_q
root          94  0.0  0.0      0     0 ?        S    21:31   0:00 [irq/52-dw-hdm
[... 172 lines cut ...]
alarm       1615  3.0  1.0 1211877028 175104 ?   Sl   22:28   0:06 /usr/lib/chrom
root        1627  0.2  0.0      0     0 ?        I    22:28   0:00 [kworker/u32:5
[... cut ...]
root        1733  0.0  0.0      0     0 ?        I<   22:31   0:00 [kworker/5:0H]
root        1734  0.0  0.0      0     0 ?        I    22:31   0:00 [kworker/2:0]
root        1735  0.0  0.0      0     0 ?        I    22:31   0:00 [kworker/4:1-p
root        1736  0.0  0.0      0     0 ?        I    22:32   0:00 [kworker/3:0-m
root        1737  0.0  0.0      0     0 ?        I    22:32   0:00 [kworker/7:2]
alarm       1738  200  0.0   9692  3728 pts/1    R+   22:32   0:00 ps aux
[alarm@MiniArch-opi5plus ~]$ 
```

warpme, 2025-01-10:
well  i looked at %CPU and all thread are almost zero.... (except ps as it comumes cpu to display this info)
So for me all looks ok here!

JFLim1, 2025-01-10:
> well i looked at %CPU and all threads are almost zero of cpu load.... (except ps as it comumes cpu to display this info)
> So for me all looks ok here!

Interesting.  HTOP keeps indicating CPU Load Average > 1 (or around 1)  even when idle with no other app running just HTOP.  Right now HTOP is indicating CPU Load Average of 1.17 while editing this message on Chromium and  all lthe 8 cores of CPU% is very low.

JFLim1, 2025-01-11:
Hi @warpme,
Currently on trying BredOS (Archlinux) with kernel linux-rockchip-rkr3-3%3A6.1.75-20-aarch64.pkg.tar.zst, when system is idle for 15 minutes or so the HTOP indicated CPU Load Average of 0.06 with Chromium running.

[image]

warpme, 2025-01-16:
Well - at this moment it looks to me like potential mainline kernel issue. Without more data i can't move forward...
I don't observing this on my 11 different rk3588[s] systems :-(
May you pls try to investigate what component (task or thread or process or...) consumes cpu?
Without this we don't know where issue is.....

warpme, 2025-01-16:
This is on mine opi5plus: 
```
    0[|||                                                                            2.0%]   4[|                                                                               0.7%]
    1[|                                                                              0.7%]   5[|||                                                                             2.0%]
    2[                                                                               0.0%]   6[||                                                                              1.3%]
    3[                                                                               0.0%]   7[                                                                                0.0%]
  Mem[|||||||||||||||||||                                                     1014M/7.51G] Tasks: 64, 139 thr, 136 kthr; 1 running
  Swp[                                                                              0K/0K] Load average: 0.02 0.34 0.37
                                                                                           Uptime: 00:10:11
```
pls remember: load is relative to all cores.
So on 8-core system (yours opi5plus) fully loaded system will have load=8
So if you have load i.e. =1 then this means avg. each cpu core load is 12.5%
On mine system 5min avg load is 0.02 (2%) - so per core it is 0.25%
For all looks ok here.....

JFLim1, 2025-01-16:
On my Opi5-Plus after 30 minutes of idle with only HTOP running and two open Terminal with NO command running. The CPU Load iAve > 1.

Not sure what settings you have that is different than on my device.  By the way, CPU Governor=performance as default.  Setting CPU Governor=ondemand did not make much difference.

[image]

warpme, 2025-01-16:
I can't reproduce this in my system.
So - if you will be not able to determine what exact sys. component (or ui module or app or whatever) causes this - we can't move forward....

JFLim1, 2025-01-16:
Really no idea why.  Using BTOP also show the similar result as HTOP.  Any other  Turn-Off Wifi does not make a difference either.

Last resort is to reflash MiniArch to see whether it makes a difference on the CPU Load Average.  Will report back in a day or two.

[image]

warpme, 2025-01-16:
I think better migth be turn-off component by component and see.
I would start with turn-off graphical UI

JFLim1, 2025-01-16:
> I would start with turn-off graphical UI

Not sure I understand "turn-off graphical UI" as I am on Gnome.    Pretty green on this.  

I tried to kill --gapplications-service in HTOp using F9 but it just kill HTOP instead.

JFLim1, 2025-01-17:
Hi @warpme,

Might have found the issue that causes the high CPU Load Average > 1 on MiniArch-EOS with kernel-6.12.9-2.

It might be related to rtc_hym8563.  From ps aux, noticed that irq hym8563 CPU uses 3.7% to 6.4% on MiniArch-EOS-6.12.9-2. 

On another Archlinuxarm-Gnome with kernel-6.1.75-joshua-git image which does not have the high CPU Load Ave issue irq hym8563 CPU usage is 0%.

```
ps aux output:
MiniArch-EOS-Gnome Kernel-6.12.9-2
root         284  3.7  0.0      0     0 ?        D    10:37   1:21 [irq/121-hym8563]

root         306  6.4  0.0      0     0 ?        D    11:28   0:12 [irq/122-hym8563]

root         306  5.5  0.0      0     0 ?        D    11:28   4:06 [irq/122-hym8563]

OPIOS-Arch-Gnome Kernel-6.1.75-joshua-git
root         124  0.0  0.0      0     0 ?        S    12:47   0:00 [irq/87-hym8563]

```

Web search hym8563 and found out it might be related to OrangePi 5 Plus - rtc hym8563 - irq issue which might be specific to Orange Pi 5 Plus.
https://forum.armbian.com/topic/33360-orangepi-5-plus-rtc-hym8563-irq-issue/

The suggested solution from Armbian Forum was a kernel patch.

Compiling kernel and adding patches is beyond my skill set.

warpme, 2025-01-24:
Ah - this might be because lack of rtc battery.
Do you have rtc battery connected in your system ?

JFLim1, 2025-01-24:
> Ah - this might be because lack of rtc battery. Do you have rtc battery connected in your system ?

No, did not install a RTC battery.

FYI, the blacklist rtc_hym8563 resolved the high CPU Load Average issue.

warpme, 2025-01-24:
Ok lack rtc battery may explain issue....
I saw in board release notes that some versions had issue with rtc triggering int when no power.

As now we know issue - may we close this ticket?

JFLim1, 2025-01-24:
Thanks for the support and info on NO rtc battery triggering int when no power.  That will explain the high cpu load average.

Yes, issue resolved.

closed 2025-01-24 by JFLim1